raspberrypi/linux #4352: sc16is7xx driver: error when removing and reinserting the module

sonelu, 2021-05-19:
**Describe the bug**
If running the following:

```
sudo modprobe -r sc16is7xx
sudo modprobe sc16is7xx
```

There is an error issued in ``dmesg``:

```
[ 2381.687987] irq: type mismatch, failed to map hwirq-23 for gpio@7e200000!
[ 2381.691676] sc16is7xx: probe of spi0.1 failed with error -22
```

The ``modprobe`` itself does not raise any errors.

**To reproduce**
On a system with the driver ``sc16is7xx`` running, execute the removal and reinsertion of the module as described above.

**Expected behaviour**
It should be possible to remove and add back the module (eventually after a recompile).

**Actual behaviour**
Error in ``dmseg``

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 
> Pi4 (4GB RAM)

* Which OS and version (`cat /etc/rpi-issue`)?
> Raspberry Pi reference 2021-03-04
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2

* Which firmware version (`vcgencmd version`)?
> Apr 30 2021 13:45:52
> Copyright (c) 2012 Broadcom
> version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

* Which kernel version (`uname -a`)?
> Linux MH5-3B76 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

**Logs**
Included above.

**Additional context**
using SC16IS762 connected on SPI0 at 15MHz. Communication at 2Mbs on both UARTs.

acgrama, 2021-07-09:
Hi, did you manage to find a solution to this problem? Thanks!

sonelu, 2021-07-09:
Hi @acgrama. I think this was because of changes in irq handling between 4.19 and 5.10 kernel. See also #4351 that has more discussion on the 5.10 kernel.

If you're using a fresh kernel install you should not see this error as the stock driver is supposed to work in 5.10. There are some performance problems discussed in the issue mentioned above, but no errors at start.

If you have kernel 5.10 and want the 4.19 version updated for the irq there is a driver code and `.ko` compilation [here](https://github.com/SYNTHesse-GIT/mh5_platform/tree/main/sc16is7xx) that I currently use on my robot to avoid the performance issues raised in the #4351.

acgrama, 2021-07-12:
@sonelu Thank you very much for your detailed explanation! It turns out that we can ignore these commands and the error, as we were removing and adding a module just because of some now obsolete issues.