raspberrypi/linux #4357: Audio quality from aux is poor

mgrouch, 2021-05-22:
It sounds like the speaker is put inside 
a carton box or something. 

If I play same song to same speakers from iPhone via Bluetooth the quality is much better. 

When it comes out aux is quite poor no matter which speakers/ headphones I use. 

Thanks

popcornmix, 2021-05-22:
What do you mean by aux? The 3.5mm headphone jack?

mgrouch, 2021-05-22:
Yes

popcornmix, 2021-05-22:
The audio output is basic, using filtered output from PWM, rather than a dedicated DAC.
That may just be how it is. HDMI will give better quality, as will an I2S audio HAT, USB dongle or bluetooth.

P33M, 2021-05-25:
The "aux" (3.5mm jack) audio uses a driver circuit that has a shoestring budget. It does not have good bass reproduction when driving low-impedance loads - I recommend a minimum of 10kohms sink impedance for an acceptable frequency response. 

Absent any further information, I'm closing this as the design works as intended.

closed 2021-05-25 by P33M