raspberrypi/linux #4358: Add CM4 supporting for kernel 4.19.y (back porting)

micronrobot, 2021-05-23:
Hello,
Due to personal reason, I must use kernel 4.19.y for my system with CM4 module. I have tried to add CM4 supporting to kernel 4.19.y by creating the bcm2711-rpi-cm4.dts file from bcm2711-rpi-4-b.dts file. I've patched my bcm2711-rpi-cm4.dts file based on the differences between bcm2711-rpi-cm4.dts and  bcm2711-rpi-4-b.dts in kernel 5.4.y. My CM4 system can boot with new bcm2711-rpi-cm4.dtb file but I pcie and usb doesn't work. 
Does anyone have tip what I should do more to get CM4 fully work in kernel 4.19.y? 
Thank you.

ostrich3, 2025-06-03:
I am trying to do the same. I need to use kernel 4.19y, but need PCIe to work. Did you ever manage to get this to work?

JamesH65, 2025-06-04:
It's likely this will never be possible. You should find out why you need such an ancient kernel (2018 release, although EOL wasn't until Dec 2024) . Fixing that would be a better use of time.

popcornmix, 2025-06-04:
> But perhaps I will look into modifying the patch for a newer kernel

That sounds like a much more useful solution.

I'll close this as we'll obviously not be backporting anything to 4.19 kernel.

pelwell, 2025-06-04:
Although this forum isn't meant for general support, if you do get stuck with the port we may be able to help.

pelwell, 2025-06-04:
If you can do it on 6.12 then you can do it on 4.19, but you may have to back-port so many commits that you end up with a  kernel looking suspiciously like 6.12.

wirelessOJ, 2025-06-04:
hmmmm. I am no expert so I am not sure if it is worth to try that or to modify the kernel patch designed for 4.19 to be adapted for a later kernel

6by9, 2025-06-04:
Post a link to this atk9k patch that you're trying to use.

If https://github.com/zlab-pub/wifi-ptp/blob/main/ath9k.diff, then it looks relatively simple, but will depend on how much churn there has been to the underlying driver.
Sticking on 4.19 forever isn't a viable solution.

wirelessOJ, 2025-06-04:
yes it is that one. I agree, I would love to be able to update it for newer kernel.

6by9, 2025-06-04:
Almost all of that forward ports with minimal intervention.

The only bit that doesn't is the `adjfreq` hook has been merged into a `adjfine`, and been removed with 75ab70ec5cef.
There are a number of other conversions from `adjfreq` to `adjfine` for various cards (d8fae250,  5a55423, ccd3bf9, 3626a69, etc), but I know nothing about that API or hardware to be immediately able to do the conversion.

I've thrown the basic forward port to 6.12 at #6882, but I am NOT proposing that this would be merged into our tree.
The function of `adjfine` has been disabled with `#if 0`, so needs fixing up in order to get PTP to work.

The first test would be to see if that works at all without PTP to confirm that no other parts are broken. If everything appears to be working, then someone with a little more knowledge or time could look into the implementation of that function.
You might ask via an issue on https://github.com/zlab-pub/wifi-ptp/issues in the first case.

wirelessOJ, 2025-06-04:
Thank you so much. I have been in contact with them as well and will follow up on adjfine as you mention.

6by9, 2025-06-04:
> [@6by9](https://github.com/6by9) did you forward port for 6.12? or what kernel version?

6.12, hence the PR is targeting rpi-6.12.y.

Once CI has run (about 40 mins), you can use `sudo rpi-update pulls/6882` to get the test kernel build on a Raspberry Pi OS install.

wirelessOJ, 2025-06-04:
Running sudo rpi-update pulls/6882 resulted in errors.  It said invalid artifact specified. Response: 404.

6by9, 2025-06-04:
CI = Continuous Integration.
All pull requests get automatically built and a few other tests run.

The builds have failed due to a build warning because the diff is using the wrong format specifier in a printk call. I'll try a very quick fixup, otherwise it'll wait for tomorrow.

CALMorACT, 2025-06-09:
Thank you for migrating our project's code to the newer kernel! We truly appreciate this community effort.
As this work was done years ago, we no longer have access to a suitable test platform (RPi running latest kernel) to analyze compatibility or guarantee fixes.

To move forward:
1. Please help test if you have the hardware. Maybe you can try to not change the specific logic of `adjfreq` first, that is, just cancel `if 0` and observe whether there is any error. @wirelessOJ
2. Report any specific issues encountered (errors, logs, steps).

We'll gladly review feedback and provide suggestions based on the concrete problems found.
Thanks again for your contribution and collaboration!

6by9, 2025-06-11:
@CALMorACT is this still an ongoing project from your side? If so then I may be able to arrange to send some Pi hardware to you. We can't help with the specific wifi dongle as it isn't something we sell.

jacob-keller, 2025-06-11:
adjfine is in units of scaled ppm, which is parts per 2^16 million. Adjfreq is parts per billion. You have to adjust the units.

 I recommend using adjust_by_scaled_ppm, or diff_by_scaled_ppm to implement your calculations, depending on how your hardware handles increments. I can take a closer look at code when I get home, and offer a more detailed suggestion

jacob-keller, 2025-06-12:
> > adjfine is in units of scaled ppm, which is parts per 2^16 million. Adjfreq is parts per billion. You have to adjust the units.
> > 
> > I recommend using adjust_by_scaled_ppm, or diff_by_scaled_ppm to implement your calculations, depending on how your hardware handles increments. I can take a closer look at code when I get home, and offer a more detailed suggestion
> 
> Thank you! The difference between the units used for adjfine and adjfreq was the bit of information that I couldn't find or deduce easily from the patches.
> I'll update the draft PR based on your comments later to allow others to test.

That's part of why I added the helper functions. It makes it easier to get right both with the units and with avoiding pitfalls of multiplication if the values get beyond 64 bits in the temporary variables.

wirelessOJ, 2025-06-12:
Is this updated now?

wirelessOJ, 2025-06-16:
@CALMorACT and @6by9 when I run sudo ./ptp4l -I wlan0 -p /dev/ptp1 -m I get the following message:
ptp4l[314.325]: ethtool
ptp4l[314.325]: interface 'wlan0' does not support requested time stamping mode
ptp4l[314.325]: phc_device: /dev/ptp1
ptp4l[314.325]: selected /dev/ptp1 as PTP clock
ptp4l[314.325]: Failed to open /dev/ptp1: No such file or directory
failed to create a clock 

I see a ptp0 device and tried that one as well

wirelessOJ, 2025-06-17:
@jacob-keller I was wondering if it would make sense to also updated the linuxptp.patch to work with the newest linuxptp? This patch was built for this older commit: https://github.com/richardcochran/linuxptp/tree/dd30b3a0d94d1c087066066e5df6bc84e3019b0b 

will that cause any problems? Thank you for your help on this

jacob-keller, 2025-06-17:
It's a good idea to update linuxptp

wirelessOJ, 2025-06-23:
@jacob-keller , @CALMorACT and @6by9 _I went through and modified the necessary files using patch generated reject files. This included clock.c, config.c, config.h, linreg.c, makefile, port.c, port_private.h, sk.c, transport.c, udp.c._ 

**However, I am getting some errors when I try to run make.** 

```
/bin/sh: 1: .//incdefs.sh: Permission denied
make: .//version.sh: Permission denied
gcc -Wall -DVER=  -g    -c -o clock.o clock.c
In file included from clock.c:35:
missing.h:60:9: error: redeclaration of enumerator ‘HWTSTAMP_TX_ONESTEP_SYNC’
   60 |         HWTSTAMP_TX_ONESTEP_SYNC = 2,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from clock.c:21:
/usr/include/linux/net_tstamp.h:122:9: note: previous definition of ‘HWTSTAMP_TX_ONESTEP_SYNC’ with type ‘enum hwtstamp_tx_types’
  122 |         HWTSTAMP_TX_ONESTEP_SYNC,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
missing.h:66:9: error: redeclaration of enumerator ‘HWTSTAMP_TX_ONESTEP_P2P’
   66 |         HWTSTAMP_TX_ONESTEP_P2P = 3,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/linux/net_tstamp.h:130:9: note: previous definition of ‘HWTSTAMP_TX_ONESTEP_P2P’ with type ‘enum hwtstamp_tx_types’
  130 |         HWTSTAMP_TX_ONESTEP_P2P,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
missing.h:72:9: error: redeclaration of enumerator ‘SOF_TIMESTAMPING_BIND_PHC’
   72 |         SOF_TIMESTAMPING_BIND_PHC = (1 << 15),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/linux/net_tstamp.h:33:9: note: previous definition of ‘SOF_TIMESTAMPING_BIND_PHC’ with type ‘enum <anonymous>’
   33 |         SOF_TIMESTAMPING_BIND_PHC = (1 << 15),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
missing.h:75:8: error: redefinition of ‘struct so_timestamping’
   75 | struct so_timestamping {
      |        ^~~~~~~~~~~~~~~
/usr/include/linux/net_tstamp.h:59:8: note: originally defined here
   59 | struct so_timestamping {
      |        ^~~~~~~~~~~~~~~
In file included from clock.c:42:
clock.c: In function ‘clock_create’:
clock.c:1251:47: error: ‘struct interface’ has no member named ‘ts_info’
 1251 |                 pr_err("phc_device: %d", iface->ts_info.phc_index);
      |                                               ^~
print.h:60:26: note: in definition of macro ‘PRINT_CL’
   60 |                 print(l, x);                            \
      |                          ^
clock.c:1251:17: note: in expansion of macro ‘pr_err’
 1251 |                 pr_err("phc_device: %d", iface->ts_info.phc_index);
      |                 ^~~~~~
clock.c: At top level:
clock.c:2059:12: warning: ‘clock_offset_slide_window’ defined but not used [-Wunused-function]
 2059 | static int clock_offset_slide_window(struct clock *c, int64_t *mofft) {
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [<builtin>: clock.o] Error 1

```

I am not sure why there is redundancy between /usr/include/linux/net_tstamp.h and missing.h because nothing should have been changed here. 

Any tips?

jacob-keller, 2025-06-23:
It looks like incdefs.sh isn't running. This script generated an include file based on your Linux headers. Make sure incdefs.sh and version.sh can execute

wirelessOJ, 2025-06-24:
Thank you for that. I am having the most difficulty with the port.c forward porting.

wirelessOJ, 2025-06-24:
This is the current output when I run `make`. 

```
DEPEND port.c
gcc -Wall -DVER=4.4 -D_GNU_SOURCE -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC -DHAVE_ONESTEP_P2P -DHAVE_VCLOCKS -DHAVE_PTP_CAPS_ADJUST_PHASE -DHAVE_IF_TEAM -g    -c -o port.o port.c
port.c: In function ‘port_tx_sync’:
port.c:1809:42: error: ‘CTL_SYNC’ undeclared (first use in this function)
 1809 |         msg->header.control            = CTL_SYNC;
      |                                          ^~~~~~~~
port.c:1809:42: note: each undeclared identifier is reported only once for each function it appears in
port.c: In function ‘process_delay_req’:
port.c:2333:42: error: ‘CTL_DELAY_RESP’ undeclared (first use in this function); did you mean ‘PDELAY_RESP’?
 2333 |         msg->header.control            = CTL_DELAY_RESP;
      |                                          ^~~~~~~~~~~~~~
      |                                          PDELAY_RESP
port.c: In function ‘process_follow_up’:
port.c:2483:38: error: incompatible type for argument 2 of ‘port_synchronize’
 2483 |         port_synchronize(p, syn->hwts.ts, fup->ts.pdu,
      |                             ~~~~~~~~~^~~
      |                                      |
      |                                      tmv_t
port.c:1428:39: note: expected ‘uint16_t’ {aka ‘short unsigned int’} but argument is of type ‘tmv_t’
 1428 |                              uint16_t seqid,
      |                              ~~~~~~~~~^~~~~
port.c:2483:50: error: incompatible type for argument 3 of ‘port_synchronize’
 2483 |         port_synchronize(p, syn->hwts.ts, fup->ts.pdu,
      |                                           ~~~~~~~^~~~
      |                                                  |
      |                                                  struct timestamp
port.c:1429:36: note: expected ‘tmv_t’ but argument is of type ‘struct timestamp’
 1429 |                              tmv_t ingress_ts,
      |                              ~~~~~~^~~~~~~~~~
port.c:2484:29: error: incompatible type for argument 4 of ‘port_synchronize’
 2484 |                  syn->header.correction,
      |                  ~~~~~~~~~~~^~~~~~~~~~~
      |                             |
      |                             Integer64 {aka long int}
port.c:1430:47: note: expected ‘struct timestamp’ but argument is of type ‘Integer64’ {aka ‘long int’}
 1430 |                              struct timestamp origin_ts,
      |                              ~~~~~~~~~~~~~~~~~^~~~~~~~~
port.c:2483:9: error: too few arguments to function ‘port_synchronize’
 2483 |         port_synchronize(p, syn->hwts.ts, fup->ts.pdu,
      |         ^~~~~~~~~~~~~~~~
port.c:1427:13: note: declared here
 1427 | static void port_synchronize(struct port *p,
      |             ^~~~~~~~~~~~~~~~
port.c:2441:14: warning: unused variable ‘i’ [-Wunused-variable]
 2441 |     int len, i;
      |              ^
port.c:2441:9: warning: unused variable ‘len’ [-Wunused-variable]
 2441 |     int len, i;
      |         ^~~
port.c: In function ‘process_pdelay_req’:
port.c:2502:9: warning: unused variable ‘len’ [-Wunused-variable]
 2502 |     int len;
      |         ^~~
port.c:2501:18: warning: unused variable ‘i’ [-Wunused-variable]
 2501 |         int err, i;
      |                  ^
port.c:2500:29: warning: unused variable ‘burst’ [-Wunused-variable]
 2500 |         struct ptp_message *burst[S_BURST_SIZE];
      |                             ^~~~~
port.c: In function ‘process_delay_resp’:
port.c:2620:13: error: invalid storage class for function ‘port_peer_delay’
 2620 | static void port_peer_delay(struct port *p)
      |             ^~~~~~~~~~~~~~~
port.c:2894:13: error: invalid storage class for function ‘port_e2e_transition’
 2894 | static void port_e2e_transition(struct port *p, enum port_state next)
      |             ^~~~~~~~~~~~~~~~~~~
port.c:2941:13: error: invalid storage class for function ‘port_p2p_transition’
[... 120 lines cut ...]
 3412 | int port_number(struct port *p)
      |     ^~~~~~~~~~~
port.c:3366:5: warning: ‘port_forward_to’ defined but not used [-Wunused-function]
 3366 | int port_forward_to(struct port *p, struct ptp_message *msg)
      |     ^~~~~~~~~~~~~~~
port.c:3355:5: warning: ‘port_forward’ defined but not used [-Wunused-function]
 3355 | int port_forward(struct port *p, struct ptp_message *msg)
      |     ^~~~~~~~~~~~
port.c:3105:16: warning: ‘port_event’ defined but not used [-Wunused-function]
 3105 | enum fsm_event port_event(struct port *p, int fd_index)
      |                ^~~~~~~~~~
port.c:2859:23: warning: ‘port_compute_best’ defined but not used [-Wunused-function]
 2859 | struct foreign_clock *port_compute_best(struct port *p)
      |                       ^~~~~~~~~~~~~~~~~
port.c:2837:6: warning: ‘port_close’ defined but not used [-Wunused-function]
 2837 | void port_close(struct port *p)
      |      ^~~~~~~~~~
port.c:1492:13: warning: ‘port_syfufsm_print_mismatch’ defined but not used [-Wunused-function]
 1492 | static void port_syfufsm_print_mismatch(struct port *p, enum syfu_event event,
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
port.c:608:12: warning: ‘fup_sync_ok’ defined but not used [-Wunused-function]
  608 | static int fup_sync_ok(struct ptp_message *fup, struct ptp_message *sync)
      |            ^~~~~~~~~~~
make: *** [<builtin>: port.o] Error 1
```

wirelessOJ, 2025-06-25:
I have since made progress editing msg.h and transport.c

wirelessOJ, 2025-06-25:
I am now getting the error:

```missing.h:79:19: error: static declaration of 'clock_adjtime' follows non-static declaration
79 | static inline int clock_adjtime(clocked_t id, struct timex *tx)```

This is what it looks like in the `missing.h` file:

```
78| #ifndef HAVE_CLOCK_ADJTIME
79| static inline int clock_adjtime(clockid_t id, struct timex *tx)
80| {
81| 	return syscall(__NR_clock_adjtime, id, tx);
82| }
```

jacob-keller, 2025-06-26:
> I am now getting the error:
> 
> `missing.h:79:19: error: static declaration of 'clock_adjtime' follows non-static declaration 79 | static inline int clock_adjtime(clocked_t id, struct timex *tx)`
> 
> This is what it looks like in the `missing.h` file:
> 
> ```
> 79| static inline int clock_adjtime(clockid_t id, struct timex *tx)
> 80| {
> 81| 	return syscall(__NR_clock_adjtime, id, tx);
> 82| }
> ```

My guess is that your system has clock_adjtime available, so missing.h shouldn't be defining it. It shouldn't have generated but maybe the checks for it are wrong?