raspberrypi/linux #4361: apt update today on the 24 May 2021, having VLC issues.

mrusli70, 2021-05-24:
Hi,

After doing the update today on the 24th May 2021, i am having issues with VLC. The vlc cannot seems to work now. I cannot uninstall or install vlc. On my raspberry pi 4 8GB Ram. Using Raspbian OS version 5.10.

Can you guys please fix the issues. Thank you. 

pi@raspberrypi:~ $ sudo apt install vlc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 vlc : Depends: vlc-plugin-base (= 3.0.12-0+deb10u1+rpt2) but it is not going to be installed
       Recommends: vlc-l10n (= 3.0.12-0+deb10u1+rpt2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

XECDesign, 2021-05-24:
Should be resolved now. In the future please use https://github.com/RPi-Distro/repo/issues/

mrusli70, 2021-05-25:
Thanks! I try with add/remove software.

closed 2021-05-26 by mrusli70