raspberrypi/linux #4362: Netplan.io package is outdated

abrilevskiy, 2021-05-26:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
The Netplan.io package is outdated - as a result it is impossible to setup wireguard tunnel with netplan.io
**To reproduce**

1. apt-get install netplan.io
2. Configure setup with wireguard

**Expected behaviour**
Setup is applied successfully 

**Actual behaviour**
Error is shown: wireguard is not supported

**System**

System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2
* Which firmware version (`vcgencmd version`)?
Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

popcornmix, 2021-09-29:
Closing as the issue is not related to the linux kernel.
If you still have an issue then https://github.com/RPi-Distro/repo/issues/ may be a better place for report.

closed 2021-09-29 by popcornmix

abrilevskiy, 2021-09-29:
Could you please help me to find right project to submit the issues?

pelwell, 2021-09-29:
> If you still have an issue then https://github.com/RPi-Distro/repo/issues/ may be a better place for report.