raspberrypi/linux #4363: Questions: Upstream Raspberry Pi 400

lategoodbye, 2021-05-26:
During the last rainy days i had a little bit time to work on Raspberry Pi 400 mainline support.

The initial draft is available [here](https://github.com/lategoodbye/rpi-zero/tree/rpi-400-v1).

Currently i have the following questions:

- Is it correct there is only a Power LED (no ACT LED) which connected to GPIO 42 ?
- What is connected to GPIO 2 of the GPIO expander ?
- Is there a explanation for the reduced MDIO clock ?

pelwell, 2021-05-26:
1. There is a single software-controlled LED on GPIO 42. Originally it was configured as an inverted ACT LED, as on Pi Zero, but it was decided that this left the LED off for too long during boot and might possibly lead people to think the unit wasn't powered correctly. I then wrote a combined activity/power LED trigger that alternated between on and ACT every 500ms, but that wasn't liked. So it is now configured as default-on, but can be overridden with the `act_led_trigger` dtparam.
2. Although the schematic shows it being connected to PWR_LED_OFF, that trace doesn't appear to go anywhere, so it is effectively unconnected.
3. We've seen a few issues around PHY clocking on the 400 and thought it safer to reduce it. There should be essentially no performance impact.

1 & 2 mean that the downstream DTs for 400 is somewhere between misleading and wrong. If you were to combine the two into a PWR LED using GPIO 42 then I would probably use those definitions downstream.

lategoodbye, 2021-05-26:
Okay, in this case everything is correct in the first draft. Btw there is a different [approach](https://github.com/lategoodbye/rpi-zero/commit/0431410599cd14ae2e70d4d5a94e1f37cdf3d67f) to handle the poweroff.

lategoodbye, 2021-06-30:
Since Raspberry Pi 400 support will be in next kernel release, we can close this.

closed 2021-06-30 by lategoodbye