raspberrypi/linux #4368: USB Gadget only mode is broken: struct dwc2_hsotg has no member named bus_suspended

pmunts, 2021-05-28:
One of the recent changes to the dwc2 driver has broken USB Gadget Only mode (`CONFIG_USB_DWC2_PERIPHERAL=y`).

The `bus_suspended` field of `struct dwc2_hsotg` is not included in the structure definition  if `CONFIG_USB_DWC2_PERIPHERAL` is defined.  This causes a compile error:

```
drivers/usb/dwc2/core_intr.c: In function 'dwc_handle_gpwrdn_disc_det':
drivers/usb/dwc2/core_intr.c:710:7: error: 'struct dwc2_hsotg' has no member named 'bus_suspended'
```

pelwell, 2021-05-28:
That's an upstream bug introduced by a recent fix (which is why it was back-ported). I've pushed a patch to rpi-5.10.y that should fix it.

pelwell, 2021-05-28:
https://marc.info/?l=linux-usb&m=162218990901875&w=2