raspberrypi/linux #4370: Slow / extremely lag 5GHz WLAN

tomty89, 2021-05-31:
**Describe the bug**
Seemingly after upgrading kernel from c2f585ae894081445b7f1ab8d0b6334061be9965 to 131a6abaac64b638dbfab44a92f4403fc89ba569, 5GHz WLAN connection seems to struggle. Latency of `ping`ing other LAN hosts ranges from several ms to several seconds (to the extent that I can hardly `ssh`; yet it's not entirely dead; it seems to sort of "ramp up" after I started `ping`ing).

I downgraded back to c2f585ae894081445b7f1ab8d0b6334061be9965 but it doesn't seem to help. It at least feels like the upgrade has done something permanent to the wireless NIC. 2.4GHz WLAN still seem to work perfectly, with LAN ping latency of around 5ms with occasional spikes up to 20ms.

I haven't moved the Pi at all since it last work. 5GHz WLAN still seem to work fine one or two reboot before I upgraded. (I was browsing the Internet and streaming some not-so-high-res video. Now I can't even practically open my router's admin page. Regardless of the fact that my other 5Ghz clients still worked fine, I rebooted the router and it didn't help either.

It's a Raspberry Pi 4.

popcornmix, 2021-05-31:
A problem with wifi is you are dependent on your environment.
The reduced performance could be due to a neighbour using the same wireless channel, or some other interfering equipment.

I don't believe there is any means where updating kernel (I assume through rpi-update) and then reverting it back can have any permanent change after a reboot.

tomty89, 2021-05-31:
Hmm, just now I'm testing again I cannot reproduce it anymore. I still don't think it's just interference. It wasn't just "reduced performance" anyway. (Even with 2.4GHz being so "crowded", I've never experienced "congestion", or maybe more like "constipation", like that. Let alone 5GHz. If it's really interference, then it must be some first-time visiting alien *AND* Broadcom / Pi being the weakest in dealing with it.)

But meh it seems like "ghost problem" like that is for some reason typical among the different components of a Pi. I suppose that's what you can get with the low price.

adminy, 2024-06-22:
having the same issue on the pi 5. It can't be the 2.4Ghz band, because its connected to the 5Ghz band:
```
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr <<MAC-ADDR>>
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr <<MAC-ADDR>>
                ssid <<WIFI-NAME>>
                type managed
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 31.00 dBm
                
wlan0     IEEE 802.11  ESSID:"<<WIFI-NAME>>"  
          Mode:Managed  Frequency:5.18 GHz  Access Point: <<MAC-ADDRESS>>
          Bit Rate=433.3 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=64/70  Signal level=-46 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0 
```