raspberrypi/linux #4373: Write to usb_gadget HID device stuck after lot of traffic to usb_gadget mass_storage

berkutta, 2021-06-03:
**Describe the bug**
I'm using Linux usb_gadget to emulate a USB Stick + USB Touchscreen to an Intel Board. The Intel Board boots of the emulated USB Stick and receives touch inputs over the emulates USB Touchscreen. When I constantly send data into the emulated HID USB Touchscreen + boot the device from the emulated USB Stick the HID device file gets not writeable any more. Write attempts to it will get stuck forever.

**To reproduce**
Download Archlinux ISO to `/data/` https://mirror.puzzle.ch/archlinux/iso/2021.06.01/archlinux-2021.06.01-x86_64.iso

Initialize USB devices:
```
#!/bin/bash

cd /sys/kernel/config/usb_gadget/
mkdir -p isticktoit
cd isticktoit
echo 0x1d6b > idVendor # Linux Foundation
echo 0x0104 > idProduct # Multifunction Composite Gadget
echo 0x0100 > bcdDevice # v1.0.0
echo 0x0200 > bcdUSB # USB2
mkdir -p strings/0x409
echo "1234" > strings/0x409/serialnumber
echo "1234" > strings/0x409/manufacturer
echo "1234" > strings/0x409/product

mkdir -p configs/c.1/strings/0x409
echo "Config 1: ECM network" > configs/c.1/strings/0x409/configuration

echo 250 > configs/c.1/MaxPower

# ELO touch
mkdir -p functions/hid.usb0
echo 1 > functions/hid.usb0/protocol
echo 1 > functions/hid.usb0/subclass
echo 5 > functions/hid.usb0/report_length
echo -ne \\x05\\x01\\x09\\x01\\xa1\\x01\\x05\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x01\\x15\\x00\\x25\\x01\\x35\\x00\\x45\\x01\\x66\\x00\\x00\\x75\\x01\\x95\\x01\\x81\\x62\\x75\\x01\\x95\\x07\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x16\\x00\\x00\\x26\\x10\\x27\\x36\\x00\\x00\\x46\\x10\\x27\\x66\\x00\\x00\\x75\\x10\\x95\\x02\\x81\\x62\\xc0\\xc0 > functions/hid.usb0/report_desc

ln -s functions/hid.usb0 configs/c.1/

# Init mass storage device
ITER=0
FILE="/data/archlinux-2021.06.01-x86_64.iso"

mkdir -p functions/mass_storage.usb$ITER
echo 1 > functions/mass_storage.usb$ITER/stall
echo 0 > functions/mass_storage.usb$ITER/lun.0/cdrom
echo 0 > functions/mass_storage.usb$ITER/lun.0/ro
echo 0 > functions/mass_storage.usb$ITER/lun.0/nofua
echo $FILE > functions/mass_storage.usb$ITER/lun.0/file
echo $(basename $FILE) > functions/mass_storage.usb$ITER/lun.0/inquiry_string

ln -s functions/mass_storage.usb$ITER configs/c.1/

# End functions
ls /sys/class/udc > UDC
```

Connect a PC/Laptop via OTG to the Raspberry Pi. The Raspberry Pi should be powered externally. I use a special USB-C Cable which just connects D+/D- and GND.

Boot the PC/Laptop of the emulated USB Stick. When the Archlinux Bootloader starts to countdown for booting, start the HID write script:
```
#!/bin/bash

while true; do
	if [ -e "/dev/hidg0" ]; then
		echo write0
		echo \\x00\\xff\\xff\\x00\\xff\\xff > /dev/hidg0
		echo write1
		echo \\x00\\xff\\xff\\x00\\xff\\x00 > /dev/hidg0
	fi
done
```

After Archlinux is fully booted the HID write script is stuck as it can't write any data any more to the HID device. It keeps hanging at the `write0` or `write1` statement. Also restarting the script doesn't help.

**Expected behaviour**
HID device still writeable after Archlinux boot

**Actual behaviour**
HID device not writeable/usable any more after Archlinux boot

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 (Tried multiple Boards, can pretty rule out an HW issue)

* Which OS and version (`cat /etc/rpi-issue`)?
```
# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

* Which firmware version (`vcgencmd version`)?
```
# vcgencmd version
Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
```
# uname -a
Linux rkag-chd-4 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```

**Logs**
```
[  169.505286] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  200.567390] dwc2 fe980000.usb: new device is high-speed
[  200.582768] dwc2 fe980000.usb: new address 1
[  201.361192] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  201.475147] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  201.475170] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  201.475226] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  217.539307] dwc2 fe980000.usb: new device is high-speed
[  217.554660] dwc2 fe980000.usb: new address 1
[  218.333070] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  218.445242] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  218.445265] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  218.445318] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  229.913190] dwc2 fe980000.usb: new device is high-speed
[  229.928556] dwc2 fe980000.usb: new address 1
[  230.706876] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  230.825310] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  230.825333] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  230.825389] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  275.543830] configfs-gadget gadget: End Point Request ERROR: -108
[  275.543894] configfs-gadget gadget: in_ep is disabled
[  275.598634] dwc2 fe980000.usb: new device is high-speed
[  275.721413] dwc2 fe980000.usb: new device is high-speed
[  275.789239] dwc2 fe980000.usb: new address 1
[  276.814140] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  276.925716] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  276.925738] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  276.926136] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  276.926606] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.045711] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.045731] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  277.046142] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  277.061272] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.175721] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.175743] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  277.176144] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  277.176942] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.295725] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 1)
[  277.295747] dwc2 fe980000.usb: dwc2_hsotg_start_req: ep2 is stalled
[  277.296150] dwc2 fe980000.usb: dwc2_hsotg_ep_sethalt(ep 28cb4fc7 ep2in, 0)
[  303.411945] configfs-gadget gadget: End Point Request ERROR: -108
[  303.458692] dwc2 fe980000.usb: new device is high-speed
```

**Additional context**
Can I mitigate this issue somehow? Maybe I need to do some checks before writing to the HID device?

I also used different Live USB images like Yocto. Also using different Intel PC doesn't change this issue.

mdevaev, 2021-07-16:
Try to change `echo 1 > functions/mass_storage.usb$ITER/stall` to `echo 0 > functions/mass_storage.usb$ITER/stall`.

berkutta, 2021-09-14:
Thanks! This helped to fix this problem (together with changing the fwrite's to O_NONBLOCK)

closed 2021-09-14 by berkutta