raspberrypi/linux #4379: Authentication with RPI3B in AP-mode with a single network using WPA-EAP fails on client with code 17=IE_DIFFERENT

Animajosser, 2021-06-08:
**Is this the right place for my bug report?**
It might also be a firmware problem. Using Openwrt packages with version number 20190416-1:
brcmfmac-firmware-43430-sdio
brcmfmac-firmware-43455-sdio
brcmfmac-firmware-usb

The upstream repository is here (correct me if I'm wrong): https://github.com/RPi-Distro/firmware-nonfree

**Describe the bug**
When trying to connect to the raspberry pi with a specific config (see configuration below), The client will try to connect, but disconnect before finishing successfully. When using wpa_supplicant 2.9 on a client, it is quite easy to see why when setting logging turned on (wpa_supplicant -dd -u -f ~/wpa.log): In "RSN IE in Beacon/ProbeResp" the last bit in the 20th byte (starting at 0) byte is 1 and not 0 as in "RSN IE in 3/4 msg".

An excerpt of the debugging output: 

```
WPA: decrypted EAPOL-Key key data - hexdump(len=48): [REMOVED]
wlp4s0: State: 4WAY_HANDSHAKE -> 4WAY_HANDSHAKE
wlp4s0: WPA: RX message 3 of 4-Way Handshake from 60:38:e0:8a:3b:60 (ver=2)
WPA: IE KeyData - hexdump(len=48): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 01 00 00 dd 16 00 0f ac 01 01 00 4a 61 6e f8 4c cf b5 98 39 a6 bb 1b d5 1e 10 89 dd 00
WPA: RSN IE in EAPOL-Key - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 01 00 00
WPA: GTK in EAPOL-Key - hexdump(len=24): [REMOVED]
wlp4s0: WPA: IE in 3/4 msg does not match with IE in Beacon/ProbeResp (src=60:38:e0:8a:3b:60)
WPA: RSN IE in Beacon/ProbeResp - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 01 01 00
WPA: RSN IE in 3/4 msg - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 01 00 00
wlp4s0: Request to deauthenticate - bssid=60:38:e0:8a:3b:60 pending_bssid=00:00:00:00:00:00 reason=17 (IE_IN_4WAY_DIFFERS) state=4WAY_HANDSHAKE
TDLS: Tear down peers
wpa_driver_nl80211_deauthenticate(addr=60:38:e0:8a:3b:60 reason_code=17)
```

I don't know if the endianness of the hexdump is converted. Assuming it is not, I deducted that the bit means: pre authentication is enabled. With that in mind I set: option rsn_preauth '1'. under config wifi-iface 'wifinet0' . After that the issue was fixed. It now doesn't complain anymore and connects correctly.

**To reproduce**
Setup a wifi network in /etc/config/wireless:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
	option htmode 'HT20'
	option country '<COUNTRY>'
	option legacy_rates '1'
	option channel '6'

config wifi-iface 'wifinet0'
	option auth_server '<SERVER-IP>'
	option ssid '<SSID>'
	option device 'radio0'
	option acct_port '1813'
	option auth_port '1812'
	option acct_server '<SERVER-IP>'
	option network 'lan'
	option mode 'ap'
	option auth_secret '<PASSWORD>'
	option acct_secret '<PASSWORD>'
	option encryption 'wpa2'
```

**Expected behaviour**
A client should connect correctly to the RPi3B AP.

**Actual behaviour**
The client exits on code 17=IE_DIFFERENT, unless you set: option rsn_preauth='1' 
This could only be tested on linux clients using wpasupplicant, but not a single client was able to connect.

**System**
RaspInfo is not available on the OpenWRT platform.

Model: Pi3B
Operating System: OpenWrt 19.07
Kernel: 4.14.221

**Additional context**

In client mode a connection cannot be made to a WPA-EAP AP. With the same logic as the AP problem I set okc=1 in wpasupplicant.conf. This also fixed the issue.

This does not happen when using WPA-PSK.

Threads with some, maybe, useful information:
The original openwrt forum thread: https://forum.openwrt.org/t/wifi-client-deauthenticated-from-rpi-3b-as-802-1x-ap-reason-ie-different/90607
An OpenWrt forum thread about the same issue: https://forum.openwrt.org/t/clients-abort-4-way-handshake-rsn-ie-different-in-message-3-4/44639

lategoodbye, 2021-07-11:
@Animajosser Could you please reproduce this issue with the recent kernel 5.10 on Raspberry Pi OS?

Animajosser, 2021-07-12:
I'll try, but it'll take a little time to set it up (haven't done it on raspbian before).
However I have little hope it's fixed in 5.10 as I have an archlinux rpi0w with the latest kernel running that probably has this issue with as a client to a working AP with the same setup.

Animajosser, 2021-07-12:
I decided to tackle it today anyway and I learned something about OpenSSL and Hostapd in the process.

I installed the latest RPI OS Lite image and used SSH to get in.
I updated all packages and installed hostapd.
I set the country code in raspi-config.
I didn't setup dhcp and set a static IP in the client.

Kernel: 5.10.17-v7+
Hostapd: 2:2.7+git20190128+0c1e29f-6+deb10u3
firmware-brcm80211: 1:20190114-2+rpt1

I eliminated the Radius server and went with the following config:
```
country_code=<COUNTRY>
interface=wlan0
driver=nl80211
ssid=<SSID>
wpa=2
wpa_key_mgmt=WPA-EAP
rsn_pairwise=CCMP
macaddr_acl=0 
auth_algs=1
ieee8021x=1
hw_mode=g
channel=6
ignore_broadcast_ssid=0
#rsn_preauth=1

eap_server=1

eap_user_file=/etc/hostapd_eap_user.conf
server_cert=/home/pi/keys/certs/ca.cert.pem
private_key=/home/pi/keys/private/ca.key.pem
private_key_passwd=<password>
```

Again the connection cannot be made.

When uncommenting the line: rsn_preauth=1, the connection can be made again.