raspberrypi/linux #4380: Kernel 5.10 - Kodi does not play HEVC videos anymore

skrenes, 2021-06-08:
Playback of many HEVC videos (regardless of resolution/bitrate) no longer seem possible with a fresh install of Raspberry Pi OS on Kodi 18.7 (fully upgraded using apt). This comes with kernel `5.10.17-v7l+`, but I've also tested this with kernel `5.10.42-v7l+`, the latest version available from `rpi-update` at the time of writing this. I've also tried variations on setting `cma-512` and `gpu_mem=320`.

Reverting back to kernel `5.4.83-v7l+` via the command  `sudo BRANCH=stable rpi-update b7c8ef64ea24435519f05c38a2238658908c038e` and setting `config.txt` to use `gpu_mem=320` with `dtoverlay=vc4-fkms-v3d` restores HEVC playback. Anything else crashes with certain files.

I really hope these issues with kernel 5.10 can be remedied soon, especially since it's now the default kernel. This issue seems related to issues #4112 and #4136.

**To reproduce**
1. Install latest version of Raspberry Pi OS
2. Fully update the system: `sudo apt update && sudo apt upgrade -y && sudo apt full-upgrade -y && sudo apt autoremove -y`
3. Add `gpu_mem=320` to `config.txt`
4. Reboot: `sudo reboot`
5. Install Kodi: `sudo apt install -y kodi`
6. Playback HEVC video within Kodi. Another user provided this example file:
`wget user-images.githubusercontent.com/53397439/110206300-27e0fc80-7e7d-11eb-8c46-1e318b80f35a.mp4`

**Expected behaviour**
HEVC video plays back smoothly.

**Actual behaviour**
HEVC video crashes Raspberry Pi on kernel `5.10.42-v7l+`:
```
rpi_ctrl_ffmpeg_init
 id=0x5489ee0
setup_io: 00000000 -> 0x9968d000 (fd:55)
setup_io: 00000000 -> 0x97c0c000 (fd:56)
***** gpu_malloc_uncached_internal, 67108864
A 100000000 apb:0x9968d000 axi.arm:0x8c500000 axi.vc:1d700000
```

Reverting back to kernel `5.4.83-v7l+` via the command  `sudo BRANCH=stable rpi-update b7c8ef64ea24435519f05c38a2238658908c038e` restores functionality with smooth playback but with similar errors in the console (unnoticeable if running Kodi as a systemd service):
```
rpi_ctrl_ffmpeg_init
 id=0xb0ac7528
setup_io: 00000000 -> 0xae701000 (fd:59)
setup_io: 00000000 -> 0xa4821000 (fd:60)
***** gpu_malloc_uncached_internal, 67108864
A 100000000 apb:0xae701000 axi.arm:0x8ab49000 axi.vc:3a383000
```
...and then when stopping playback of the HEVC clip (not possible on kernel 5.10, since it's crashed at this point):
```
rpi_ctrl_ffmpeg_free id=0xb0ac7528
***** gpu_free_internal
rpi_ctrl_ffmpeg_free freed axi mem
rpi_ctrl_ffmpeg_free closed mbox
rpi_ctrl_ffmpeg_free freed rpi
```

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

Linux RPi-Kodi 5.10.42-v7l+ #1422 SMP Tue Jun 8 13:03:05 BST 2021 armv7l GNU/Linux
Revision	: b03111
Serial		: 10000000ebadc0de
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------
[... 654 lines cut ...]
[   10.745520] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   10.874713] brcmfmac: F1 signature read @0x18000000=0x15264345
[   10.928732] Console: switching to colour frame buffer device 240x67
[   10.934307] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k FS
[   10.947537] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.948792] usbcore: registered new interface driver brcmfmac
[   10.952624] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[   10.979470] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[   11.245192] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.261764] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[   11.788278] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   17.701705] Bluetooth: Core ver 2.22
[   17.701827] NET: Registered protocol family 31
[   17.701844] Bluetooth: HCI device and connection manager initialized
[   17.701877] Bluetooth: HCI socket layer initialized
[   17.701900] Bluetooth: L2CAP socket layer initialized
[   17.701935] Bluetooth: SCO socket layer initialized
[   17.714506] Bluetooth: HCI UART driver ver 2.3
[   17.714522] Bluetooth: HCI UART protocol H4 registered
[   17.714590] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.714794] Bluetooth: HCI UART protocol Broadcom registered
[   17.969618] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.969627] Bluetooth: BNEP filters: protocol multicast
[   17.969640] Bluetooth: BNEP socket layer initialized
```

ZedeN1, 2021-07-08:
This doesn't address your issue but I have **all** HEVC videos playing perfectly using 5.10 and Kodi 19.1. As a workaround you could do the same:

For `Linux raspberrypi 5.10.46-v7l+ #1433 SMP Mon Jul 5 15:04:09 BST 2021 armv7l GNU/Linux` try the following `/boot/config.txt`, based on [LibreELEC](https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/devices/RPi4/config/distroconfig.txt).

```
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
disable_overscan=1
disable_fw_kms_setup=1
```

It works for me using `19.1 (19.1.0) Git:20210701-a12eccbe35 Media Center Kodi` from @popcornmix [github](https://github.com/popcornmix/xbmc/tree/gbm_matrix). 

Follow build [instructions](https://github.com/popcornmix/xbmc/blob/gbm_matrix/docs/README.Linux.md). You will probably need the most of the following dependencies on Raspberry Pi OS 32bit:
```
sudo apt update
sudo apt install autoconf automake autopoint gettext autotools-dev cmake curl default-jre gawk gcc gcc-4.9 g++ g++-4.9 cpp cpp-4.9 gdc gperf libasound2-dev libasound-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec-dev libp8-platform-dev libcrossguid-dev libcwiid-dev libdbus-1-dev libegl1-mesa-dev libenca-dev libflac-dev libfontconfig-dev  libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libgl1-mesa-dev libgl-dev libglew-dev libglu1-mesa-dev libglu-dev libgnutls28-dev libgpg-error-dev libgtest-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libnfs-dev libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev  libpulse-dev libshairplay-dev libsmbclient-dev libspdlog-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtiff-dev libtinyxml-dev libtool libudev-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxslt-dev libxt-dev netcat wayland-protocols wipe lsb-release meson nasm ninja-build python3-dev python3-pil  python-support python3-minimal rapidjson-dev swig unzip uuid-dev yasm zip zlib1g-dev libcap-dev libclang-dev libcdio-dev libiso9660-dev libudf-dev libspdlog-dev autoconf automake autopoint autotools-dev cmake cpp curl default-jre g++ gawk gcc gdc gettext gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec4 libcec-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libenca-dev libflac-dev libfmt3-dev libfontconfig-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgbm-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libglew-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libgtest-dev libinput-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmariadb-dev libmicrohttpd-dev libnfs-dev  libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libshairplay-dev libsmbclient-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtool libudev-dev libunistring-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev lsb-release meson nasm ninja-build python3-dev python3-pil python-support rapidjson-dev swig unzip uuid-dev wayland-protocols yasm zip zlib1g-dev raspberrypi-kernel-headers
```

Just change the configure command to be something like this: 
```
cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DAPP_RENDER_SYSTEM=gles -DENABLE_VAAPI=OFF -DENABLE_VDPAU=OFF -DENABLE_DEBUGFISSION=OFF -DENABLE_TESTING=OFF -DENABLE_INTERNAL_FLATBUFFERS=ON -DENABLE_INTERNAL_FFMPEG=ON -DENABLE_INTERNAL_SPDLOG=ON -DENABLE_INTERNAL_FMT=ON
```

Harvie, 2021-07-11:
Oh cr*p. Just upgraded and TV does not work anymore... Is there easy way to downgrade?

skrenes, 2021-07-25:
Thanks @ZedeN1 . It's nice to know 19.1 is working with it. I may do that in the future, but right now, the downgraded kernel and otherwise vanilla install is serving me well.

gsjv2, 2021-07-26:
> This doesn't address your issue directly but I have had the exact same HEVC issues with: RPi4, v5.10 and Kodi v18.7 from the official repo. I managed to get **all** HEVC videos playing perfectly using 5.10 and Kodi 19.1. As a workaround you could do the same:
> 
> For `Linux raspberrypi 5.10.46-v7l+ #1433 SMP Mon Jul 5 15:04:09 BST 2021 armv7l GNU/Linux` try the following `/boot/config.txt`, based on [LibreELEC](https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/devices/RPi4/config/distroconfig.txt).
> 
> ```
> dtoverlay=vc4-kms-v3d,cma-512
> dtoverlay=rpivid-v4l2
> disable_overscan=1
> disable_fw_kms_setup=1
> ```
> 
> It works for me using `19.1 (19.1.0) Git:20210701-a12eccbe35 Media Center Kodi` from @popcornmix [github](https://github.com/popcornmix/xbmc/tree/gbm_matrix).
> 
> Follow build [instructions](https://github.com/popcornmix/xbmc/blob/gbm_matrix/docs/README.Linux.md). You will probably need most of the following dependencies on Raspberry Pi OS (buster) 32bit:
> 
> ```
> sudo apt update
> sudo apt install autoconf automake autopoint gettext autotools-dev cmake curl default-jre gawk gcc gcc-4.9 g++ g++-4.9 cpp cpp-4.9 gdc gperf libasound2-dev libasound-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec-dev libp8-platform-dev libcrossguid-dev libcwiid-dev libdbus-1-dev libegl1-mesa-dev libenca-dev libflac-dev libfontconfig-dev  libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libgl1-mesa-dev libgl-dev libglew-dev libglu1-mesa-dev libglu-dev libgnutls28-dev libgpg-error-dev libgtest-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libnfs-dev libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev  libpulse-dev libshairplay-dev libsmbclient-dev libspdlog-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtiff-dev libtinyxml-dev libtool libudev-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxslt-dev libxt-dev netcat wayland-protocols wipe lsb-release meson nasm ninja-build python3-dev python3-pil  python-support python3-minimal rapidjson-dev swig unzip uuid-dev yasm zip zlib1g-dev libcap-dev libclang-dev libcdio-dev libiso9660-dev libudf-dev libspdlog-dev autoconf automake autopoint autotools-dev cmake cpp curl default-jre g++ gawk gcc gdc gettext gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec4 libcec-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libenca-dev libflac-dev libfmt3-dev libfontconfig-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgbm-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libglew-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libgtest-dev libinput-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmariadb-dev libmicrohttpd-dev libnfs-dev  libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libshairplay-dev libsmbclient-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtool libudev-dev libunistring-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev lsb-release meson nasm ninja-build python3-dev python3-pil python-support rapidjson-dev swig unzip uuid-dev wayland-protocols yasm zip zlib1g-dev raspberrypi-kernel-headers
> ```
> 
> Just change the configure command to be something like this:
> 
> ```
> cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DAPP_RENDER_SYSTEM=gles -DENABLE_VAAPI=OFF -DENABLE_VDPAU=OFF -DENABLE_DEBUGFISSION=OFF -DENABLE_TESTING=OFF -DENABLE_INTERNAL_FLATBUFFERS=ON -DENABLE_INTERNAL_FFMPEG=ON -DENABLE_INTERNAL_SPDLOG=ON -DENABLE_INTERNAL_FMT=ON
> ```

This totally works.  I just tested and HVEC HW decoding is working!

andrebrait, 2022-02-02:
This is still needed as of today on Buster (running Kodi 19.3 on top of Raspbian on a Raspberry Pi 4).

At least the part below. 
I can't tell if the order of those lines matter.

```
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
```

johanbluecreek, 2022-03-08:
Looking at this issue and #4112 , using a rpi4, Linux raspberrypi 5.15.26-v7l+ kernel, Kodi 19.3, I managed to get a HEVC playing flawlessly once after a reboot (not been able to reproduce since) by having the following in my `config.txt`

```
[pi4]
dtoverlay=vc4-fkms-v3d,cma-384
dtoverlay=rpivid-v4l2
disable_fw_kms_setup=1
max_framebuffers=2
[all]
enable_uart=1
gpu_mem=384
hdmi_enable_4kp60=1
```

The most common thing that happens is that the video plays for a few seconds and then stops (except for that one time). If I do not have the `rpivid-v4l2` line I don't even get the video to play those initial seconds, and changing the memory reserved for cma or gpu either causes a kernel panic (if the sum of them are higher or equal to 384+512), or the pi stops sending a HDMI signal mid boot.

Is the downgrade option and/or building Kodi from source the only work-arounds available?

andrebrait, 2022-03-08:
@johanbluecreek 

I have an Raspberry Pi 4B 4GB and I can play any HEVC (tested up to 4k@10bpc) just fine for any amount of time.

I'm on kernel `5.10.92-v7l+` on 32-bit Raspbian Buster, though. All I have in my config is:

```
[pi4]                                                                           dtoverlay=rpivid-v4l2                                                           dtoverlay=vc4-kms-v3d,cma-512
```

Akrai, 2022-03-08:
Has HEVC stopped working again on 5.15? No issues on 5.10.63-v8+ so far, not gonna upgrade any time soon it seems

popcornmix, 2022-03-08:
HEVC is fine on 5.15 for me (on bullseye).

andrebrait, 2022-03-08:
@johanbluecreek 

Could you check if all your firmware is up to date?

Maybe run `sudo rpi-eeprom-update` and see if there's anything to update there. If so, you can update it with `sudo rpi-eeprom-update -a` and then `systemctl reboot` it.

Keep the Pi powered on the entire time, of course.

Also, can you enable debug logging on Kodi and try again? These error logs don't have enough information.

popcornmix, 2022-03-08:
@johanbluecreek your issue isn't related to this bug report.

> CFileCache::Open - </home/pi/storage/videos/video.mkv> failed to open

the video file couldn't be opened at all - it never gets near the hevc decoder.
Now, I don't know if /home/pi/storage/videos/ is on the sdcard, or a mounted USB or network drive,
but that should be where you are looking to fix it.

johanbluecreek, 2022-03-09:
@popcornmix Sorry, I messed up the permissions on that file. I fixed that and moved the file to the SD card to not allow the mount to play a role. The log file is now
```
2022-03-09 10:12:01.110 T:1114     INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-03-09 10:12:01.111 T:1114     INFO <general>: CAESinkALSA::Initialize - Attempting to open device "default"
2022-03-09 10:12:01.119 T:1114     INFO <general>: CAESinkALSA::Initialize - Opened device "default"
2022-03-09 10:12:01.120 T:1114     INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2022-03-09 10:12:01.120 T:1114     INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S16NE
2022-03-09 10:12:01.145 T:1109     INFO <general>: VideoPlayer::OpenFile: /home/pi/video_sd/vid.mkv
2022-03-09 10:12:01.147 T:1222     INFO <general>: Creating InputStream
2022-03-09 10:12:01.173 T:1222     INFO <general>: Creating Demuxer
2022-03-09 10:12:01.332 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]: Input #0, matroska,webm, from '/home/pi/video_sd/vid.mkv':
2022-03-09 10:12:01.332 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:   Metadata:
2022-03-09 10:12:01.332 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:     encoder         : libebml v1.3.4 + libmatroska v1.5.0
2022-03-09 10:12:01.332 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:     creation_time   : 2019-10-24T02:02:28.000000Z
2022-03-09 10:12:01.332 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:   Duration: 00:31:23.12, start: 0.000000, bitrate: 1384 kb/s
2022-03-09 10:12:01.333 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:     Stream #0:0: Video: hevc (Main 10), yuv420p10le(tv, bt709/unknown/unknown), 1920x800, SAR 1:1 DAR 12:5, 23.98 fps, 23.98 tbr, 1k tbn, 23.98 tbc (default)
2022-03-09 10:12:01.333 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:     Stream #0:1(eng): Audio: aac (HE-AAC), 48000 Hz, 5.1, fltp
2022-03-09 10:12:01.333 T:1222     INFO <general>: ffmpeg[0x4c5edb8X]:     Stream #0:2(eng): Subtitle: subrip
2022-03-09 10:12:01.337 T:1222     INFO <general>: Opening stream: 0 source: 256
2022-03-09 10:12:01.337 T:1222     INFO <general>: Creating video codec with codec id: 173
2022-03-09 10:12:01.337 T:1222     INFO <general>: CDVDVideoCodecDRMPRIME::Open - using decoder HEVC (High Efficiency Video Coding)
2022-03-09 10:12:01.340 T:1222     INFO <general>: Creating video thread
2022-03-09 10:12:01.340 T:1230     INFO <general>: running thread: video_thread
2022-03-09 10:12:01.340 T:1222     INFO <general>: Opening stream: 1 source: 256
2022-03-09 10:12:01.341 T:1222     INFO <general>: Finding audio codec for: 86018
2022-03-09 10:12:01.350 T:1222     INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2022-03-09 10:12:01.350 T:1222     INFO <general>: Creating audio thread
2022-03-09 10:12:01.350 T:1231     INFO <general>: running thread: CVideoPlayerAudio::Process()
2022-03-09 10:12:01.351 T:1222     INFO <general>: Opening stream: 2 source: 256
2022-03-09 10:12:01.364 T:1231     INFO <general>: Skipped 1 duplicate messages..
2022-03-09 10:12:01.364 T:1231     INFO <general>: Creating audio stream (codec id: 86018, channels: 6, sample rate: 48000, no pass-through)
2022-03-09 10:12:01.381 T:1114     INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-03-09 10:12:01.581 T:1114     INFO <general>: CAESinkALSA::Initialize - Attempting to open device "default"
2022-03-09 10:12:01.584 T:1114     INFO <general>: CAESinkALSA::Initialize - Opened device "default"
2022-03-09 10:12:01.585 T:1114     INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2022-03-09 10:12:01.585 T:1114     INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S16NE
2022-03-09 10:12:11.381 T:1222    ERROR <general>: Skipped 1 duplicate messages..
2022-03-09 10:12:11.381 T:1222    ERROR <general>: ffmpeg[0x4c5edb8X]: [matroska,webm] File ended prematurely
2022-03-09 10:12:21.515 T:1109     INFO <general>: Loading skin file: VideoFullScreen.xml, load type: KEEP_IN_MEMORY
2022-03-09 10:12:24.817 T:1222     INFO <general>: Process - eof reading from demuxer
2022-03-09 10:12:24.817 T:1222     INFO <general>: CVideoPlayer::OnExit()
2022-03-09 10:12:24.817 T:1222     INFO <general>: VideoPlayer: eof, waiting for queues to empty
2022-03-09 10:12:24.817 T:1222     INFO <general>: Closing stream player 1
2022-03-09 10:12:24.817 T:1222     INFO <general>: CDVDMessageQueue(audio)::WaitUntilEmpty
2022-03-09 10:12:24.818 T:1222     INFO <general>: Waiting for audio thread to exit
2022-03-09 10:12:24.818 T:1231     INFO <general>: thread end: CVideoPlayerAudio::OnExit()
2022-03-09 10:12:24.818 T:1222     INFO <general>: Closing audio device
2022-03-09 10:12:25.201 T:1222     INFO <general>: Skipped 2 duplicate messages..
2022-03-09 10:12:25.201 T:1222     INFO <general>: Deleting audio codec
2022-03-09 10:12:25.201 T:1222     INFO <general>: Closing stream player 2
2022-03-09 10:12:25.201 T:1222     INFO <general>: CDVDMessageQueue(video)::WaitUntilEmpty
2022-03-09 10:12:25.202 T:1222     INFO <general>: waiting for video thread to exit
2022-03-09 10:12:25.202 T:1230    ERROR <general>: Got MSGQ_ABORT or MSGO_IS_ERROR return true
2022-03-09 10:12:25.202 T:1230     INFO <general>: thread end: video_thread
2022-03-09 10:12:25.202 T:1222     INFO <general>: deleting video codec
[... cut ...]
hdmi_enable_4kp60=1
``` 

@andrebrait 

> Maybe run `sudo rpi-eeprom-update` and see if there's anything to update there. If so, you can update it with `sudo rpi-eeprom-update -a` and then `systemctl reboot` it.

There was only a bootloader update available. I performed the update anyway, and it had no effect.

> Also, can you enable debug logging on Kodi and try again? These error logs don't have enough information.

The log file above is fixed, but if debug is still needed, see: https://pastebin.com/k24eszTd

andrebrait, 2022-03-09:
@johanbluecreek 

According to the logs, it seems your file is broken. It says it ended prematurely. I think it may have reached EOF sooner than it expected.

Are you able to play the file on your computer or some other device?

The only difference between our config.txt files os that I still have the `dtparam=audio=on` line and I got `dtparam=i2c_arm=on` commented out.

popcornmix, 2022-03-09:
The log did contain
```
2022-03-09 10:12:01.182 T:1222    DEBUG <general>: Open - probing detected format [matroska,webm]
2022-03-09 10:12:01.187 T:1225    DEBUG <general>: Thread Terminating with Exception: std::bad_alloc
```
which suggests an allocation failure (this is a normal malloc style alloc, rather than cma or gpu_mem)
Your debug log file was cut down so didn't show if there were any harmful advancedsettings. Have you changed default caching behaviour? Too large a file cache could fail with allocation error.

johanbluecreek, 2022-03-09:
> The log did contain
>
> ```
> 2022-03-09 10:12:01.182 T:1222    DEBUG <general>: Open - probing detected format [matroska,webm]
> 2022-03-09 10:12:01.187 T:1225    DEBUG <general>: Thread Terminating with [...]

I had changed the default. Changing to `<memorysize>1073741824</memorysize>` there seems to be no issue left. Thank you very much.

closed 2023-02-23 by popcornmix