raspberrypi/linux #4383: Kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:1093!

Marietto2008, 2021-06-11:
Hello,

I'm trying to emulate the raspi3 on my jetson nano and at the same time I want to enable this network interface :

`--netdev user,id=net0,hostfwd=tcp::5022-:22 -device usb-net,netdev=net0 \  `

but to do this,I need to recompile the kernel used by the image :

2019-09-26-raspbian-buster-lite-kernel.img  

because by default the CONFIG_USB_NET_CDCETHER=m ,instead it should be set to y.

Anyway,this is the qemu command line I run :

```
qemu-system-aarch64 \
  --machine raspi3 \
  --cpu arm1176 \
  --m 1024 \
  --drive file=/var/www/html/2019-09-26-raspbian-buster-lite-kernel.img,format=raw \
  --netdev user,id=net0,hostfwd=tcp::5022-:22 -device usb-net,netdev=net0 \
  --dtb bcm2710-rpi-3-b-plus.dtb \
  --kernel kernel8.img \
  --append "rw earlyprintk loglevel=8 console=ttyAMA0,115200 dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootwait panic=1 dwc_otg.fiq_fsm_enable=0" \
  --no-reboot \
  --display none \
  --serial mon:stdio

```

Let's try to recompile this kernel version :

```
root@raspberrypi:/home/pi# uname -a  

Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux 
```

on my PC with the toolchain :

```
root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# ARCH=arm64

CROSS_COMPILE=/home/ziomario/Scaricati/gcc-arm-10.2-2020.11-x86_64-aarch64-none-linux-gnu/bin/aarch64-none-linux-gnu-

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/gcc-arm-10.2-2020.11-x86_64-aarch64-none-linux-gnu# cd /home/ziomario/Scaricati/linux-rpi-5.4.y/

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# ls

arch   COPYING  Documentation  include  Kbuild   lib          Makefile  README   security  usr
block  CREDITS  drivers        init     Kconfig  LICENSES     mm        samples  sound     virt
certs  crypto   fs             ipc      kernel   MAINTAINERS  net       scripts  tools

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# cd arch

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch# ls

alpha  arm    c6x   h8300    ia64     m68k        mips   nios2     parisc   riscv  sh     um         x86
arc    arm64  csky  hexagon  Kconfig  microblaze  nds32  openrisc  powerpc  s390   sparc  unicore32  xtensa

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch# cd arm64

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64# ls

boot     crypto   Kbuild   Kconfig.debug      kernel  lib       mm   xen
configs  include  Kconfig  Kconfig.platforms  kvm     Makefile  net

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64# cd configs

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64/configs# ls

bcm2711_defconfig  bcmrpi3_defconfig  defconfig

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64/configs# cp bcmrpi3_defconfig /home/ziomario/Scaricati/linux-rpi-5.4.y

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64/configs# cd ..

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch/arm64# cd ..

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y/arch# cd ..

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# ls

arch               certs    crypto         fs       ipc      kernel    MAINTAINERS  net      scripts   tools
bcmrpi3_defconfig  COPYING  Documentation  include  Kbuild   lib       Makefile     README   security  usr
block              CREDITS  drivers        init     Kconfig  LICENSES  mm           samples  sound     virt

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# cp bcmrpi3_defconfig .config

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# nano .config

root@Z390-AORUS-PRO:/home/ziomario/Scaricati/linux-rpi-5.4.y# make -j12 ARCH=arm64 CROSS_COMPILE=/home/ziomario/Scaricati/gcc-arm-10.2-2020.11-x86_64-aarch64-none-linux-gnu
/bin/aarch64-none-linux-gnu-
```

At this point I've passed the new recompiled kernel called "Image" and the new recompiled dbt file called "bcm2710-rpi-3-b-plus.dtb" to qemu command line.

Unfortunately,while I was upgrading the packages I found this bug :

```
Preparing to unpack .../82-raspberrypi-sys-mods_20210310_armhf.deb ...
Unpacking raspberrypi-sys-mods (20210310) over (20190730) ...
[18346.660822] kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:1093!
[18346.671786] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[18346.675684] Modules linked in:
[18346.679583] CPU: 0 PID: 9 Comm: ksoftirqd/0 Not tainted 5.4.83-v8 #1
[18346.681680] Hardware name: Raspberry Pi 3 Model B+ (DT)
[18346.684453] pstate: 00000085 (nzcv daIf -PAN -UAO)
[18346.688672] pc : bcm2835_sdhost_finish_data+0xe8/0xf0
[18346.690112] lr : bcm2835_sdhost_dma_complete+0x1fc/0x254
[18346.691560] sp : ffffffc010063be0
[18346.692509] x29: ffffffc010063be0 x28: ffffffc010b5a0d0
[18346.693872] x27: ffffff8035ce70e0 x26: 0000000000000001
[18346.695670] x25: ffffffc010063c98 x24: dead000000000100
[18346.697234] x23: dead000000000122 x22: ffffff8036590000
[18346.698771] x21: 0000000000000000 x20: 0000000000000000
[18346.700369] x19: ffffff8035680500 x18: 0000000000000000
[18346.702308] x17: 0000000000000000 x16: 0000000000000000
[18346.704187] x15: 0000000000c35961 x14: 000000000000ba7e
[18346.706119] x13: 00bb112531ec5894 x12: 00bb112531ec5894
[18346.708810] x11: 0000000000bb1125 x10: 0000000000001980
[18346.710666] x9 : 0000000000000000 x8 : ffffff8016876040
[18346.711880] x7 : 0000000000000000 x6 : 0000000000000001
[18346.713026] x5 : 0000000000000000 x4 : 0000000000000002
[18346.714306] x3 : ffffff800c039000 x2 : 0000000000000000
[18346.716544] x1 : ffffff8036590000 x0 : ffffffc01079a3fc
[18346.719165] Call trace:
[18346.726455]  bcm2835_sdhost_finish_data+0xe8/0xf0
[18346.728547]  bcm2835_sdhost_dma_complete+0x1fc/0x254
[18346.731524]  vchan_complete+0x1d8/0x210
[18346.733036]  tasklet_action_common.constprop.0+0x124/0x17c
[18346.738480]  tasklet_action+0x40/0x50
[18346.742001]  __do_softirq+0x1a0/0x488
[18346.747955]  run_ksoftirqd+0x5c/0x74
[18346.751779]  smpboot_thread_fn+0x200/0x234
[18346.755742]  kthread+0x144/0x150
[18346.757748]  ret_from_fork+0x10/0x18
[18346.762560] Code: aa1303e0 97ffff09 f9408662 17ffffeb (d4210000)
[18346.782730] ---[ end trace 69fac190f1cdaf5b ]---
[18346.790292] Kernel panic - not syncing: Fatal exception in interrupt
[18346.795872] SMP: stopping secondary CPUs
[18357.566916] SMP: failed to stop secondary CPUs 0,2
[18357.569218] Kernel Offset: disabled
[18357.570636] CPU features: 0x0002,24002004
[18357.571740] Memory Limit: none
[18357.579261] Rebooting in 1 seconds..
[18358.811241] SMP: stopping secondary CPUs

```

pelwell, 2021-06-11:
Are you seriously expecting support for emulating a Raspberry Pi on a competitor's board from Raspberry Pi?

closed 2021-06-11 by pelwell

Marietto2008, 2021-06-11:
I am a hobbyist and I follow the utilitarian principle. I ask for support for the product I use and which has some defects and I use this product because it is useful for me to perform a task. But I'm not a fan of this or that product. I'm as pragmatic as Linus Torvalds. I remind you that he did not say no a priori to the entry of closed software into the open source ecosystem. In the world of open source that I know, there are no ideological battles and no barriers.