raspberrypi/linux #4384: Audio capture noisy with WM8960 SoC codec

tewarid, 2021-06-11:
I'm using the WM8960 SoC codec as part of a custom kernel build at version 5.10.31.

Audio playback works with Seeed's [ReSpeaker 2-Mics Pi HAT](http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/) after enabling a few switches that are off, and adjusting the volume

```bash
amixer -c1 sset 'Left Input Mixer Boost' toggle
amixer -c1 sset 'Left Output Mixer PCM' toggle
amixer -c1 sset 'Right Input Mixer Boost' toggle
amixer -c1 sset 'Right Output Mixer PCM' toggle
amixer -c1 sset 'Headphone',0 100%,100%
```

Sound capture however is very noisy, see [test.wav.zip](https://github.com/raspberrypi/linux/files/6257795/test.wav.zip) produced with

```bash
arecord -f cd -Dhw:1 -d5 > test.wav
```

[Seeed's soundcard driver](https://github.com/respeaker/seeed-voicecard) works without issue.

Found no meaningful differences in alsa settings. See  [asound.state.txt](https://github.com/raspberrypi/linux/files/6257874/asound.state.txt) vs [seeed-asound.state.txt](https://github.com/raspberrypi/linux/files/6257875/seeed-asound.state.txt).

tewarid, 2022-02-25:
Cannot reproduce issue with kernel version 5.15.24.

closed 2022-02-25 by tewarid

strix-technica, 2025-03-28:
I know this is a rather old Issue and my hardware is a little different but I noticed, when closely examining the [schematic for the Adafruit Voice Bonnet](https://learn.adafruit.com/adafruit-voice-bonnet/downloads) and the WM8960's datasheet, that the codec has three pairs of inputs two of which are floating and will be prone to noise pick-up, depending on ambient conditions.

As of this writing, the overlays in the Seeed and Waveshare repositories reference these floating inputs in routing:

```
                simple-audio-card,routing =
                        "Headphone Jack", "HP_L",
                        "Headphone Jack", "HP_R",
                        "Speaker", "SPK_LP",
                        "Speaker", "SPK_LN",
                        "LINPUT1", "Mic Jack",
                        "LINPUT3", "Mic Jack",
                        "RINPUT1", "Mic Jack",
                        "RINPUT2", "Mic Jack";
```

If this has the effect of enabling those inputs, this could be one possible cause of the noise mentioned in this Issue.  I haven't messed with device trees before and exactly how that line will be interpreted will anyway depend on the respective version of the WM8960 driver, but it seems worth flagging anyway.

In due course, I will need to modify the DT I'm using for my own purposes.  I will update if I find anything relevant in the course of doing that.

meiser79, 2026-09-21:
Hi,

this overlay needs an update, see https://github.com/jxlarrea/kiosk-satellite/issues/563#issuecomment-5750401495

I was able to improve silence-to-speech SNR from ~1.5dB (original, unpatched) to ~8.7dB.

Regards,
meiser79

pelwell, 2026-09-21:
Why not submit a Pull Request?

meiser79, 2026-09-21:
@pelwell I'd like to get one more confirmation before merging it.

pelwell, 2026-09-21:
Of course, in your own time.