raspberrypi/linux #4387: Making RTS5261 work on the CM4

benadler, 2021-06-11:
Hello,

I am trying to use the RTS5261 on the CM4 for writing to SD Express cards, using any version of linux, 32 or 64 bit. I have the CM4, an IOboard, the chip and some UHS and SD express cards. I saw that realtek recently added linux support for this chip, and kernel 5.11.1 should be the first kernel that fully supports using both SD's UHS-I and express modes.

The setup works on amd/intel based systems and recent kernels, but it doesn't work on the CM4 even when using https://github.com/raspberrypi/linux/tree/rpi-5.12.y

Using brute force, I figured out the chip doesn't work at all when using the 1b43d5b6fe5b71c300563afc0548122752a0618b rpi-eeprom/bootloader, but it works somewhat using the older bfe1c7ddc094b735134e96f2338c0ab2c9b10a5b. Somewhat meaning that if I plug in an SDexpress card and power up, it works, but there is no hotplugging and no UHS-I functionality.

So there is hope that the rpi-eeprom can help. I believe that the CM4's PCI bus may have some intricate/uncodumented differences that I can't make sense of.

What I've tried:

1. Added dtoverlay=pcie-32bit-dma to /boot/config.txt (mentioned here <https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/71> and other places)
2. Applied a DMA to physical to virtual address conversion to dma_alloc_coherent() as in this fix <https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/blob/5ae3a3e877e5b04da6e1cfdcca2d2f124857ddb0/xtrx.c#L176>
3. Switched to using pci_alloc_consistent() and applied the same DMA to physical to virtual address conversion as in this fix  <https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/blob/5ae3a3e877e5b04da6e1cfdcca2d2f124857ddb0/xtrx.c#L176>

I tried (1) with (2) then (1) with (3) but none of the above helped. I still think the most likely explanation is related to memory mapping or memory access configuration of the CM4's PCIe bus.

Here's a PCIe trace. The host asks the chip to move 2DWORD from memory address 0x1ed44000. The chip issues a read request to host, but it fails. CA is Completer Abort. Completer (host) doesn’t accept the chip’s request. So when the chip can’t access the memory, the SD card can’t be initialized.

[image]

I would like to get some hints on how to debug this properly.

- Is there a PCIe-on-CM4 expert that can help with this?
- Is there proper documentation on the PCIe part of the SoC of the CM4?
- Is there something to be learned from different rpi-eeprom versions influencing this chip's functionality?

Thank you!

timg236, 2021-06-14:
AFAIK all the public Broadcom documentation for this controller is in the Linux PCIe driver itself. The only significant gotcha that I'm aware of is that the PCIe / infrastructure doesn't support 64bit writes to PCIe IOMEM.

Re: rpi-eeprom
Different versions may or may not attempt to probe PCIe and reset the PCIe controller afterwards. Since CM4 can have arbitrary PCIe devices attached we are looking at changing it to avoid touching PCIe at all unless USB-MSD or NVMe boot options are enabled (BOOT_ORDER 0x4 or 0x6). I've attached something from the dev branch which does that so Linux should see the PCIe RC in the default hardware reset state.

[cm4-avoid-pcie-pieeprom.zip](https://github.com/raspberrypi/linux/files/6648565/cm4-avoid-pcie-pieeprom.zip)

gavingao666, 2021-06-23:
Hello, 
I think the problem is why the completer abort. 
Is that address "0x1ed44000" illegal?  I think this address is allocate from CPU.
But is there any problem in it? 

Thanks

gavingao666, 2021-06-30:
Hello @timg236 

I tried cm4-avoid-pcie-pieeprom.zip, but it seems not gonna help.
Any other sugestion?
Do you have any idea about completer  abort?

Thanks

timg236, 2021-06-30:
Sorry,  no idea.

ffainelli, 2021-07-28:
The only way for the 2711 RC to generate a completion abort is if the EP generated a PCIe address that has *NOT* been programmed into the RC window. This would mean that either the Device Tree or firmware that configured the RC did not correctly provide a 'dma-ranges' property or did an incorrect configuration of these registers:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pci/controller/pcie-brcmstb.c#n916

Assuming that the VPU correctly edited the 'dma-ranges' based upon the amount of DRAM populated, this could leave us with the RSTX driver incorrectly mangling the dma_addr_t that are being provided by the kernel somehow? Could you attach the following to ensure everything looks good:

- enabling CONFIG_DMA_API_DEBUG in the kernel to see if the driver is somehow misbehaving with respect to DMA addresses
- DTB converted back to DTS as viewed by the kernel: dtc -I dtb -O dts /sys/firmware/fdt
- register dumps for the following physical addresses: 0x7d504050, 0x7d504054 and 0x7d504058.

0x7d504050 captures the "low" address with bit 0 indicating whether the capture is valid or not and bits 31:02 containing the lower 32-bit of the address. 0x7d504054 captures the upper 32-bits and finally 0x7d504058 captures the bad data. These should also provide us with a clue.

Thank you!