raspberrypi/linux #4388: Errors in include/linux/seqlock.h while building the kernel

Mr-JoE1, 2021-06-11:
**Describe the bug**
while building the kernel v 5.10.y patched with RT official patch, ./include/linux/seqlock.h

**generate errors :**
`./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
`

**Cross Compiler *
arm-linux-gnueabihf-
from rasbian tools official repo 

**Target Machine**
Rasspberry Pi B+ board 

**Host Machine** 
Linux  5.8.0-55-generic 
20.04.1-Ubuntu 
x86_64 x86_64 x86_64 GNU/Linux

**Logs**
`In file included from ./include/linux/mmzone.h:16,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/mm.h:10,
                 from ./include/linux/nfs_fs.h:23,
                 from fs/nfs/nfstrace.c:5:
./include/net/sch_generic.h: In function ‘qdisc_run_begin’:
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:312:40: note: in expansion of macro ‘__seqprop’
  312 | #define __seqcount_lock_preemptible(s) __seqprop(s, preemptible)
      |                                        ^~~~~~~~~
./include/linux/seqlock.h:462:6: note: in expansion of macro ‘__seqcount_lock_preemptible’
  462 |  if (__seqcount_lock_preemptible(s))    \
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h:202:2: note: in expansion of macro ‘raw_write_seqcount_begin’
  202 |  raw_write_seqcount_begin(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:310:28: note: in expansion of macro ‘__seqprop’
  310 | #define __seqcount_ptr(s)  __seqprop(s, ptr)
      |                            ^~~~~~~~~
./include/linux/seqlock.h:465:29: note: in expansion of macro ‘__seqcount_ptr’
  465 |  raw_write_seqcount_t_begin(__seqcount_ptr(s));   \
      |                             ^~~~~~~~~~~~~~
./include/net/sch_generic.h:202:2: note: in expansion of macro ‘raw_write_seqcount_begin’
  202 |  raw_write_seqcount_begin(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h: In function ‘qdisc_run_end’:
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:310:28: note: in expansion of macro ‘__seqprop’
  310 | #define __seqcount_ptr(s)  __seqprop(s, ptr)
      |                            ^~~~~~~~~
./include/linux/seqlock.h:552:23: note: in expansion of macro ‘__seqcount_ptr’
  552 |  write_seqcount_t_end(__seqcount_ptr(s));   \
      |                       ^~~~~~~~~~~~~~
./include/net/sch_generic.h:209:2: note: in expansion of macro ‘write_seqcount_end’
  209 |  write_seqcount_end(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:312:40: note: in expansion of macro ‘__seqprop’
  312 | #define __seqcount_lock_preemptible(s) __seqprop(s, preemptible)
      |                                        ^~~~~~~~~
./include/linux/seqlock.h:554:6: note: in expansion of macro ‘__seqcount_lock_preemptible’
  554 |  if (__seqcount_lock_preemptible(s))    \
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h:209:2: note: in expansion of macro ‘write_seqcount_end’
  209 |  write_seqcount_end(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~
In file included from ./include/linux/dcache.h:10,
                 from fs/nfs/export.c:7:
./include/net/sch_generic.h: In function ‘qdisc_run_begin’:
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:312:40: note: in expansion of macro ‘__seqprop’
  312 | #define __seqcount_lock_preemptible(s) __seqprop(s, preemptible)
      |                                        ^~~~~~~~~
./include/linux/seqlock.h:462:6: note: in expansion of macro ‘__seqcount_lock_preemptible’
  462 |  if (__seqcount_lock_preemptible(s))    \
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h:202:2: note: in expansion of macro ‘raw_write_seqcount_begin’
  202 |  raw_write_seqcount_begin(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:310:28: note: in expansion of macro ‘__seqprop’
  310 | #define __seqcount_ptr(s)  __seqprop(s, ptr)
      |                            ^~~~~~~~~
./include/linux/seqlock.h:465:29: note: in expansion of macro ‘__seqcount_ptr’
  465 |  raw_write_seqcount_t_begin(__seqcount_ptr(s));   \
      |                             ^~~~~~~~~~~~~~
./include/net/sch_generic.h:202:2: note: in expansion of macro ‘raw_write_seqcount_begin’
  202 |  raw_write_seqcount_begin(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h: In function ‘qdisc_run_end’:
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:310:28: note: in expansion of macro ‘__seqprop’
  310 | #define __seqcount_ptr(s)  __seqprop(s, ptr)
      |                            ^~~~~~~~~
./include/linux/seqlock.h:552:23: note: in expansion of macro ‘__seqcount_ptr’
  552 |  write_seqcount_t_end(__seqcount_ptr(s));   \
      |                       ^~~~~~~~~~~~~~
./include/net/sch_generic.h:209:2: note: in expansion of macro ‘write_seqcount_end’
  209 |  write_seqcount_end(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~
./include/linux/seqlock.h:302:37: error: ‘_Generic’ selector of type ‘struct <anonymous>’ is not compatible with any association
  302 | #define __seqprop(s, prop) _Generic(*(s),    \
      |                                     ^
./include/linux/seqlock.h:312:40: note: in expansion of macro ‘__seqprop’
  312 | #define __seqcount_lock_preemptible(s) __seqprop(s, preemptible)
      |                                        ^~~~~~~~~
./include/linux/seqlock.h:554:6: note: in expansion of macro ‘__seqcount_lock_preemptible’
  554 |  if (__seqcount_lock_preemptible(s))    \
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/sch_generic.h:209:2: note: in expansion of macro ‘write_seqcount_end’
  209 |  write_seqcount_end(&qdisc->running);
      |  ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:279: fs/nfs/nfstrace.o] Error 1
  CC      kernel/user.o
  CC      kernel/signal.o
make[2]: *** [scripts/Makefile.build:279: fs/nfs/export.o] Error 1
  CC      fs/nfs/sysfs.o
  CC      fs/debugfs/file.o
  CC      kernel/sys.o
^Cmake[2]: *** [scripts/Makefile.build:279: fs/nfs/sysfs.o] Interrupt
make[1]: *** [scripts/Makefile.build:279: kernel/signal.o] Interrupt
make[1]: *** [scripts/Makefile.build:279: kernel/sys.o] Interrupt
make[2]: *** [scripts/Makefile.build:279: fs/debugfs/file.o] Interrupt
make[1]: *** [scripts/Makefile.build:496: fs/nfs] Interrupt
make[1]: *** [scripts/Makefile.build:496: fs/debugfs] Interrupt
make: *** [Makefile:1824: kernel] Interrupt
make: *** [Makefile:1824: fs] Interrupt

`

pelwell, 2021-06-14:
The -rt patches are neither created or supported by us, but I'll leave this issue open for a while in case somebody has some suggestions.

JinShil, 2021-06-23:
I'm running into this error also when building branch 5.10.y with RT patch 5.10.41-rt42.  Branch 5.10.y appears to be at 5.10.44, but the patches are at 5.10.41.  I'm not sure if that accounts for the errors.

The latest version I've been able to build successfully is branch 5.6.y with rt path 5.6.19-rt12.  All others later than 5.6 either give me patch errors or compiler errors.

Some advise on how to build later version would be most welcome.

JinShil, 2021-06-23:
I believe I've figured this out.

In order to get a successful build you must match the *3 digit* version number of the kernel at the top of https://github.com/raspberrypi/linux/blob/rpi-5.10.y/Makefile with the 3 digit version number of the RT patches at https://cdn.kernel.org/pub/linux/kernel/projects/rt/5.10/older/

So the key is to browse the git history of the kernel until you locate a "Linux 5.10.xx" commit.  The 3 digit version number must match a 3 digit RT patch version number.  Then find precisely the next "Merge remote tracking..." commit.  You need to download the code at the "Merge remote tracking..." commit to get a successful build.

So, the most recent match I've been able to find is this one:
[image]

So with that information, if you obtain the source code at commit 53a5ac4935c500d32bfc465551cc5107e091c09c and RT patch 5.10.35-rt39, you can get a successful build.  Here's an Ubuntu cross-compile script I threw together to build a real-time 5.10.35 kernel for the Raspberry Pi Compute Module 3+:

```Bash
#!/bin/bash

KERNEL_VERSION=5.10
RT_PATCH_VERSION=$KERNEL_VERSION.35-rt39
COMMIT_HASH=53a5ac4935c500d32bfc465551cc5107e091c09c
DEFCONFIG=bcm2709_defconfig
ARCH=arm
TARGET=arm-linux-gnueabihf

KERNEL_BRANCH=rpi-$KERNEL_VERSION.y
COMPILER=$TARGET-

# All work will be done in $PWD/rtkernel.  The final result will be packaged as $PWD/rtkernel/rtkernel.tgz
PROJECT_DIR=`pwd`/rtkernel
rm -rf $PROJECT_DIR
mkdir $PROJECT_DIR
cd $PROJECT_DIR

# install necessary packages
sudo apt update
sudo apt install -y git bc bison flex libssl-dev make libc6-dev libncurses5-dev crossbuild-essential-armhf

# download the kernel and RT patch
git clone --branch $KERNEL_BRANCH https://github.com/raspberrypi/linux

wget http://cdn.kernel.org/pub/linux/kernel/projects/rt/$KERNEL_VERSION/older/patch-$RT_PATCH_VERSION.patch.gz
gunzip xf patch-$RT_PATCH_VERSION.patch.gz

cd linux

# Need this to get precisely the correct kernel version
git checkout $COMMIT_HASH

# Patch the kernel
patch -p1 < ../patch-$RT_PATCH_VERSION.patch

# Prepare configuration
make ARCH=$ARCH CROSS_COMPILE=$COMPILER mrproper
make ARCH=$ARCH CROSS_COMPILE=$COMPILER $DEFCONFIG

# Configure the kernel
# At this step, select "General setup" --> "Preemption Model" --> "Fully Preemptible Kernel (Real-Time)"
make ARCH=$ARCH CROSS_COMPILE=$COMPILER menuconfig

# Compile
make -j16 ARCH=$ARCH CROSS_COMPILE=$COMPILER zImage modules dtbs

# copy assets to the $PROJECT_DIR/result directory
RESULT_DIR=$PROJECT_DIR/result
mkdir $RESULT_DIR
EXT4_DIR=$RESULT_DIR/ext4
FAT32_DIR=$RESULT_DIR/fat32
mkdir $FAT32_DIR
mkdir $EXT4_DIR
make ARCH=$ARCH CROSS_COMPILE=$COMPILER INSTALL_MOD_PATH=$EXT4_DIR modules_install
cp arch/$ARCH/boot/zImage $FAT32_DIR/
cp arch/$ARCH/boot/dts/*.dtb $FAT32_DIR/
mkdir $FAT32_DIR/overlays
cp arch/$ARCH/boot/dts/overlays/*.dtb* $FAT32_DIR/overlays/
cp arch/$ARCH/boot/dts/overlays/README $FAT32_DIR/overlays/
```

mkreisl, 2021-06-24:
I recently experimented with the RT patches, both for kernel 5.4 and 5.10.
Building the kernel works flawlessly and even on a Pi4 it runs without errors. But on a Pi3B+ the whole thing is extremely unstable, practically unusable, at least when Kodi comes into play.

JinShil, 2021-06-25:
@mkreisl The instability you are encountering with the Pi3 may be due to this bug:  https://www.osadl.org/Single-View.111+M5c03315dc57.0.html

mkreisl, 2021-06-25:
> @mkreisl The instability you are encountering with the Pi3 may be due to this bug: https://www.osadl.org/Single-View.111+M5c03315dc57.0.html

Very interesting. I will definitely test this. The patch can even be applied to the 5.10 kernel.
Many thanks for this

mkreisl, 2021-06-25:
I have now tested this, it still doesn't work properly, after a very short time the Kodi desktop freezes again. 
However, the system remains operable (ssh), so that one can say that the dispmanx functionality is unfortunately not RT capable.