raspberrypi/linux #4391: Documentation for power LED inputs for sysfs & dtparam

seamusdemora, 2021-06-13:
I'd like to better understand the options/variables available for controlling the (red) power led on the Raspberry Pi 4B. I've read /boot/overlays/README, but it doesn't seem to cover all the possible values for controlling the power led. Are these values defined in the documentation somewhere? Are the same values available for sysfs and dtparam? 

As an example, I'd like to be able to dim the power led - perhaps using `/sys/devices/platform/leds/leds/led1/brightness`. The default value seems to be 255, but all my efforts to change it (e.g. to 128) have failed to have any apparent effect. So it seems that some values can be changed, but others cannot. And some that can be changed (e.g. `trigger`) have values listed I've been unable to find definitions for.  

I'm able to enable & disable the power led using `dtparam=pwr_led_trigger=none` in `/boot/config.txt`, and using sysfs with `echo none | sudo tee /sys/class/leds/led1/trigger`. However, `dtparam=pwr_led_activelow=on` seems to have no effect (as it does for `act_led`). 

Finally, is there any way to make the changes to the power led behavior persistent after the RPi 4B enters "low power mode" (i.e. following a halt/shutdown with `WAKE_ON_GPIO=0` and `POWER_OFF_ON_HALT=1` in bootloader config)?  If so - I'd much appreciate a link to any documentation or examples on how to do that.

timg236, 2021-06-14:
There's now PWM control of the power LED so you can't control the brightness. It's either on or off.

The is no option to disable the power led from the bootloader during HALT with power off
https://github.com/raspberrypi/rpi-eeprom/issues/344

seamusdemora, 2021-06-15:
@timg236 , @pelwell : 
OK - so the brightness does not need to be controlled necessarily. Turning it OFF completely would be much better than leaving it FULL ON when the RPi is in a non-operational state. 

Wrt the 2nd comment: I obviously do not have the details to know what options are available, but [this comment](https://github.com/raspberrypi/rpi-eeprom/issues/344#issuecomment-860477762), `we aren't going to add the extra code` indicates the author believes that it can be controlled with **"code"**. It would be unreasonable of me to ask for something that is physically impossible, but that doesn't seem to be the case here - at least not based on the referenced comment. 

I get that you guys don't want to do this - I guess that's because you have much to do. But I've not heard you say that you disagree with the premise that it isn't entirely logical that the pwr led be illuminated in both the non-operational "low power mode", and in the fully operational mode. And so I will say again, would you be willing to **outline** the approach to do this - to write this code?

timg236, 2021-06-15:
It would have to be in the bootloader and that's not going to change

closed 2021-06-15 by pelwell