raspberrypi/linux #4392: Upstream kernel v5.12.10 unable to probe HDMI for vc4 driver, CM4

adam900710, 2021-06-14:
**Describe the bug**
When booting upstream kernel v5.12.10, even with `vc4-kms-v3d-pi4` overlay applied, vc4 still fails to probe the HDMI.

The board is CM4, thus not sure if it's related to the dual HDMI. (Only HDMI0 has been utilized).

The FDT and overlays are from the official firmware repo, as upstream don't yet have CM4 fdt.

The involved dmesg are:
```
[    2.312122] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    2.682089] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    3.620522] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    3.641296] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    3.720201] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    4.052063] systemd[1]: Starting Load Kernel Module drm...
[    4.153762] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    4.154861] systemd[1]: Finished Load Kernel Module drm.
[    5.119174] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    5.215041] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    5.242834] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    5.406496] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
[    5.473185] [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get pixel bvb clock
[    5.490880] vc4-drm gpu: failed to bind fef00700.hdmi (ops vc4_hdmi_ops [vc4]): -2
[    5.505295] vc4-drm gpu: master bind failed: -2
[    5.505359] vc4-drm: probe of gpu failed with error -2
```

**To reproduce**
Boot upstream kernel, with official firmware fdt and overlay, with `vc4-kms-v3d-pi4` overlay applied.

**Expected behaviour**
VC4 is expected to probe the HDMI without problem.

**Actual behaviour**
VC4 is unable to probe the HDMI, thus gpu is not initialized at all.

**System**
 Not using RPI specific distro.

* Which model of Raspberry Pi?
  CM4, 8G ram, 32G eMMC, with wifi

* Which OS and version
  Manjaro ARM, with upstream kernel and uboot.

* Which firmware version
  Latest, and no vc specific user tool installed

* Which kernel version
  5.12.10
  Upstream kernel without any RPI specific patch

lategoodbye, 2021-06-14:
I case you refer to the mainline kernel, there is no support for the CM4 yet:
https://github.com/lategoodbye/rpi-zero/issues/48

Also the downstream DTB won't help you.

adam900710, 2021-06-14:
No wonder.

But I tried the raspberry pi kernel, commit 94535fc8931c43d7106693a180113f7394fe1d48 of `https://github.com/raspberrypi/linux`.

With the same down stream device tree, still no help.

So I guess I have to go the 5.10.x branch to get VC4 working?

adam900710, 2021-06-14:
BTW, although no mainline kernel support for CM4, it works exceptionally well, except the VC4 part.

Everything from Uboot to kernel wifi/bluetooth to PCIE host controller all work fine, using the down stream device tree.

Thus it gives me some illusion that VC4 should work without problem too.

adam900710, 2021-06-18:
@lategoodbye BTW, is there any working kernel branch which enabled VC4/V3D?

lategoodbye, 2021-06-18:
@vianpl was working on this topic

His last branch against mainline seems to be here:
https://github.com/vianpl/linux/commits/v3d-enable-v2

vianpl, 2021-06-18:
I just uploaded https://github.com/vianpl/linux/tree/v3d-enable-v3 which is more or less the same code as v2 rebased to yesterday's linux-next. I tested it against mesa 21.1.2 and manages to start X11 (although it's easy to make it crash).

adam900710, 2021-06-18:
Do I need any special device tree or can just reuse the existing down stream device tree?

vianpl, 2021-06-18:
I use the upstream one.

adam900710, 2021-06-18:
But I didn't see the upstream kernel has device tree for CM4, at least not in torvalds/master branch.

vianpl, 2021-06-18:
@adam900710 Oh! sorry for that. I missed you were talking about CM4 specifically. Then my suggestion would be to adapt the downstream DT. It should match the changes available on my branch.

adam900710, 2021-06-18:
Awesome, would try that branch and try to give some feedback in recent days.