raspberrypi/linux #4393: no ethernet hotplug on Raspberry Pi 3B with current raspios

jrfromfl, 2021-06-14:
**Describe the bug**
after booting the newest version of raspios (May 7th,2021) with a Pi3B without ethernet connection I can't connect to a local network via ETH anymore. Plugging in the ethernet cable will leave the LEDs on the plug dark, /sys/class/net/$interface/carrier=0.
Booting the same sd-card with a Pi4B works fine.

**To reproduce**
copy iso to sd card and boot with a Pi3B without plugged ethernet cable, than connect cable (switch/dhcp srv)

**Expected behaviour**
LEDs on eth plug start blinking, ip addr show lists a ip addr on ethernet interface, link is UP

**Actual behaviour**
LEDs stay dark, ip link show lists ethernet interface as DOWN, no ip addr

**System**
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
Revision	: a02082
Serial		: 000000008bd985a3
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Apr 30 2021 13:47:07 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        1507280 1289304    123360  92% /
devtmpfs          439916       0    439916   0% /dev
tmpfs             473196       0    473196   0% /dev/shm
tmpfs             473196   12324    460872   3% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             473196       0    473196   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48781    209314  19% /boot
tmpfs              94636       0     94636   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
  Installed: (none)
lxpanel:
  Installed: (none)
pcmanfm:
[... 501 lines cut ...]
[   11.207364] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   11.215721] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   13.275498] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
[   13.275556] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.478007] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   14.596727] Bluetooth: Core ver 2.22
[   14.596843] NET: Registered protocol family 31
[   14.596855] Bluetooth: HCI device and connection manager initialized
[   14.596884] Bluetooth: HCI socket layer initialized
[   14.596899] Bluetooth: L2CAP socket layer initialized
[   14.596928] Bluetooth: SCO socket layer initialized
[   14.609626] Bluetooth: HCI UART driver ver 2.3
[   14.609645] Bluetooth: HCI UART protocol H4 registered
[   14.609745] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.609954] Bluetooth: HCI UART protocol Broadcom registered
[   14.880139] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.880160] Bluetooth: BNEP filters: protocol multicast
[   14.880188] Bluetooth: BNEP socket layer initialized
root@raspberrypi:/home/pi# 

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
I made the same test with the raspios version of Dec 10,2020 and it worked just fine.

morrisrob, 2021-09-13:
I am experiencing the same issue on our custom 3b boards after updating to the 5.10 kernel and have been looking for a solution. A couple of observations that I've made while troubleshooting this:

- I only experience the issue when plugged into some network switches. When connected to one my switches, it consistently works fine but when connected to a couple other switches, I experience this issue. With the switches where I am experiencing the issue, if I downgrade back to our previous 4.19 kernel that we were using, everything works fine, so this rules out an issue with the switches / cabling (I have tried multiple other ethernet cables). I have replicated this same test a number of times.

- When it is in this state, if I run 'ifconfig eth0 down', and then 'ifconfig eth0 up', I get back to a working state.

jrfromfl, 2021-09-14:
I can confirm, that it works with some switches and with others it does not.
I did tests with a Netgear switch and a FritzBox router and plugging in the ethernet cable after boot didn't work.
I also tested with a D-Link switch and it worked fine.

I can also confirm, that 'ip link set eth0 down' followed by 'ip link set eth0 up' reactivates the network conection to the Netgear switch and also to the FritzBox router.

For us this is a critical issue, because headless systems, that are remotely controlled via network may not be connected to the network after a powerfail, when the switch or router needs more time to activate the network ports than the rpi needs to boot.

pelwell, 2021-09-22:
I think I've got at least a workaround for this. It seems that link detection can take more than the allowed 640ms, but if the PHY driver is allowed to wait 1500ms then the link comes back up.

Can anyone here test this patch? I've had success on a 3B and a 2B with it (and not without it):
```diff
diff --git a/drivers/net/phy/smsc.c b/drivers/net/phy/smsc.c
index caf7291ffaf8..9983eeded624 100644
--- a/drivers/net/phy/smsc.c
+++ b/drivers/net/phy/smsc.c
@@ -195,12 +195,12 @@ static int lan87xx_read_status(struct phy_device *phydev)
                if (rc < 0)
                        return rc;
 
-               /* Wait max 640 ms to detect energy and the timeout is not
+               /* Wait max 1500 ms to detect energy and the timeout is not
                 * an actual error.
                 */
                read_poll_timeout(phy_read, rc,
                                  rc & MII_LAN83C185_ENERGYON || rc < 0,
-                                 10000, 640000, true, phydev,
+                                 10000, 1500000, true, phydev,
                                  MII_LAN83C185_CTRL_STATUS);
                if (rc < 0)
                        return rc;
```

pelwell, 2021-09-22:
See https://github.com/raspberrypi/linux/pull/4598.

morrisrob, 2021-09-23:
> Can anyone here test this patch? I've had success on a 3B and a 2B with it (and not without it):

@pelwell Yes, this does look to resolve the issue on the custom 3b that I'm using.  I have not been able to reproduce the issue after applying this patch.  Thank you for your help with this.

pelwell, 2021-09-23:
Thanks for the feedback. The patch will be in future releases.

pelwell, 2021-09-24:
Yesterday's firmware release (available via `rpi-update`) includes this fix.

ThingUroboros, 2021-12-08:
I have the same problem, is it possible to replace only the firmware for ethernet?
I tested `rpi-update`
I don't know if I need all the changes.

JamesH65, 2021-12-08:
I would expect that the changes mentioned above have already been incorporated in to apt, so the following should be enough.

sudo apt update 
sudo apt full-upgrade