raspberrypi/linux #4394: Hardware acceleration doesnt't work in Elvenar game

hvdkamer, 2021-06-15:
**Is this the right place for my bug report?**
I was pointed her in [this forum post](url=https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275372&p=1877251#p1877827) about feedback on the 64 bit version. If this is wrong, I have an excuse :-)

**Describe the bug**
The web based [Elvenar game](url=https://en.elvenar.com/) depends on functional hardware acceleration in modern browsers. Tested the lasted May 7th version, both 32 and 64 bit. In those both Chromium and Firefox (ESR) is used to start the game. In all four combinations the images of the interface are show, but the game area itself remains a dark brown.

**To reproduce**
Create or use an existing account on the Elvenar game, choose Winyandor and wait until the game is loaded.

**Expected behaviour**
A game area with all the buildings and other created stuff.

**Actual behaviour**
Only a dark brown background is showing. You can click in this area and see the reponses of the buildings etc. But it isn't shown. If wanted I can create a screenshot of a working game on Debian in Chromium or Firefox. The game is working on a variety of x86_64 hardware with both Intel and AMD graphic cards.

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4

Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Revision	: d03114
Serial		: 10000000250194ad
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 27 2021 14:02:15 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root      122603860 3341340 114227924   3% /
devtmpfs         3900044       0   3900044   0% /dev
tmpfs            4032140       0   4032140   0% /dev/shm
tmpfs            4032140    8768   4023372   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            4032140       0   4032140   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48788    209307  19% /boot
tmpfs             806428       4    806424   1% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
[... 537 lines cut ...]
[    8.138890] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.555914] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.556424] bcmgenet fd580000.ethernet eth0: Link is Down
[   10.195163] broken atomic modeset userspace detected, disabling atomic
[   12.713841] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.713885] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.909288] fuse: init (API version 7.32)
[   14.315818] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   14.740600] Bluetooth: Core ver 2.22
[   14.740672] NET: Registered protocol family 31
[   14.740678] Bluetooth: HCI device and connection manager initialized
[   14.740695] Bluetooth: HCI socket layer initialized
[   14.740706] Bluetooth: L2CAP socket layer initialized
[   14.740724] Bluetooth: SCO socket layer initialized
[   14.751840] Bluetooth: HCI UART driver ver 2.3
[   14.751854] Bluetooth: HCI UART protocol H4 registered
[   14.751924] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.752091] Bluetooth: HCI UART protocol Broadcom registered
[   14.999336] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.999347] Bluetooth: BNEP filters: protocol multicast
[   14.999363] Bluetooth: BNEP socket layer initialized
[   15.058570] Bluetooth: RFCOMM TTY layer initialized
[   15.058590] Bluetooth: RFCOMM socket layer initialized
[   15.058613] Bluetooth: RFCOMM ver 1.11
[  101.806545] v3d fec00000.v3d: MMU error from client L2T (0) at 0x10e1000, pte invalid

XECDesign, 2021-06-15:
I think it's just that we don't have a chromium build for arm64 yet.

XECDesign, 2021-06-15:
My mistake, it looks like GL should work in Debian's "chromium" package that we currently use on the 64bit image.

Does it work any differently whether you launch chromium with `--use-gl=desktop` or `--use-gl=egl`?

You could also try going to `chrome://flags`, enabling `Override software rendering list` and then launching chromium with:
`LIBGL_ALWAYS_SOFTWARE=true chromium ` (and another time with `--use-gl=egl`, just in case) to see if the problem is our driver. If you still see the problem with software rendering, the cause might be something else.

hvdkamer, 2021-06-15:
Most arm64 packages in the Raspberry Pi OS are coming from Debian. I'm pretty sure I tested both Firefox -- default in both Raspberry Pi OS 32 and 64 bit version -- and Chromium (available in Debian Buster for arm64). But I did a lot of tests, so may be I screwed this one up. That however doesn't matter! Firefox is the default and I definitely tested both images:

`hvdkamer@obelix:~/werk/downloads$ ls -l 2021-05-07-raspios-buster-arm*
-rw-r--r-- 1 hvdkamer hvdkamer 3737124864 mei  7 18:14 2021-05-07-raspios-buster-arm64.img
-rw-r--r-- 1 hvdkamer hvdkamer 1055240912 jun 13 17:30 2021-05-07-raspios-buster-arm64.zip
-rw-r--r-- 1 hvdkamer hvdkamer        102 jun 13 17:29 2021-05-07-raspios-buster-arm64.zip.sha256
-rw-r--r-- 1 hvdkamer hvdkamer 3980394496 mei  7 17:07 2021-05-07-raspios-buster-armhf.img
-rw-r--r-- 1 hvdkamer hvdkamer 1237837307 jun 13 18:57 2021-05-07-raspios-buster-armhf.zip`

Because it isn't working in the 32 bits version as well, you may ignore the 64 bit version. I know that one is still in beta. If it was only in that version I wouldn't have made the forum post in the first place...

XECDesign, 2021-06-15:
Sorry, I missed that the issue was also in 32-bit. On the 32-bit image you'd need to launch it with `chromium-browser` instead of `chromium` for the tests.

We don't pre-install firefox in any of our images, and certainly don't support it.

But yes, it does sound like one for the people working on gl support, I just wanted to make sure it wasn't a packaging issue on my end.

hvdkamer, 2021-06-15:
> If you still see the problem with software rendering, the cause might be something else.

Just to be sure, I said that this game depends on hardware acceleration. If you disable that it shows a popup that the game couldn't be started. So testing with software rendering can't be done :-).

> We don't pre-install firefox in any of our images, and certainly don't support it.

Hmm, I used the 2021-05-07-raspios-buster-arm(hf|64).zip and I'm almost certain that Firefox was installed. And that I had to install Chromium. But again it doesn't matter because the default browser didn't work also :-). I just installed the other to be thorough. Again all four combinations had the same problem.

popcornmix, 2021-06-15:
@naushir possibly a bug or missing feature in 3d mes support - one for Igalia?

hvdkamer, 2021-06-15:
Ok, got access to the hardware. Tests done with the 32 bit version. Indeed Chromium is the default browser, my bad. Debian uses Firefox as default browser in Gnome. As said, I tested a lot.

The `--use-gl=desktop` doesn't resolve the issue. Same for the `--use-gl=egl`. There is a difference in messages in the console though. Attached the output of both. As said, I won't test software rendering options because this game depends on hardware rendering. It says so when you disable the hardware acceleration.

[chromium_20210615.txt](https://github.com/raspberrypi/linux/files/6656952/chromium_20210615.txt)

naushir, 2021-06-16:
@itoral and @txenoo, any thoughts?

txenoo, 2021-06-16:
I could reproduce it with the buster Raspbery OS packages. But we can confirm that is working with a recent mesa version. 

[image]

It seems to work with desktop or egl chromium renderer.

I'll try to bisect it to check what fixed the rendering.

popcornmix, 2021-06-16:
Ping @XECDesign we might want a mesa bump to fix webgl rendering errors (once it's been identified).

XECDesign, 2021-06-16:
Hope it's something that can be backported to 19.3.2

txenoo, 2021-06-16:
@XECDesign it seems that can be easily backported to 19.3.2, it was directly backported to 20.0 20.1

This commit fixes the rendering:
```
commit e3ecf48dda2ddabfbabdad83e19d280d0edb8246
Author: Jose Maria Casanova Crespo <jmcasanova@igalia.com>
Date:   Thu May 7 03:37:22 2020 +0200

    v3d: Fix swizzle in DXT3 and DXT5 formats
    
    Swizzles were ignoring the W component of the format DXT3_RGBA and
    DXT5_RGBA.
    
    This fixes 15 piglit tests:
    
    spec/!opengl 1.1/copyteximage 2d
    spec/!opengl 1.2/copyteximage 3d
    spec/arb_texture_compression/fbo-generatemipmap-formats/gl_compressed_rgba
    spec/arb_texture_compression/fbo-generatemipmap-formats/gl_compressed_rgba npot
    spec/arb_texture_compression/texwrap formats bordercolor-swizzled/gl_compressed_rgba, swizzled, border color only
    spec/arb_texture_compression/texwrap formats bordercolor/gl_compressed_rgba, border color only
    spec/arb_texture_cube_map/copyteximage cube
    spec/arb_texture_cube_map/copyteximage cube samples=2
    spec/arb_texture_cube_map/copyteximage cube samples=4
    spec/arb_texture_rectangle/copyteximage rect
    spec/arb_texture_rectangle/copyteximage rect samples=2
    spec/arb_texture_rectangle/copyteximage rect samples=4
    spec/ext_texture_array/copyteximage 2d_array
    spec/ext_texture_array/copyteximage 2d_array samples=2
    spec/ext_texture_array/copyteximage 2d_array samples=4
    
    Fixes: 469bbd8387d1 "broadcom/vc5: Move the formats table to per-V3D-version compile."
    Reviewed-by: Iago Toral Quiroga <itoral@igalia.com>
    Part-of: <https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/4934>

```
But I would cherry-pick the complete series including also: 
```
commit 905edc376dd1ace6ac2af0fc351606210a0141a1
Author: Jose Maria Casanova Crespo <jmcasanova@igalia.com>
Date:   Thu May 7 03:50:16 2020 +0200

    v3d: Include supported DXT formats to enable s3tc/dxt extensions
    
    DXT1_RGBA and sRGB variants of DXT[135] formats are enabled as
    valid format on V3D.
    
    Once all S3TC formats supported by V3C are enabled the following
    extensions become exposed by gallium.
    
        * GL_ANGLE_texture_compression_dxt3
        * GL_ANGLE_texture_compression_dxt5,
        * GL_EXT_texture_compression_dxt1
        * GL_EXT_texture_compression_s3tc
        * GL_S3_s3tc
        * GL_EXT_texture_compression_s3tc_srgb
    
    This enables 206 passing piglit test related to gl_compressed.*s3tc_dxt
    
    Cc: 20.0 20.1 <mesa-stable@lists.freedesktop.org>
    Reviewed-by: Iago Toral Quiroga <itoral@igalia.com>
    Part-of: <https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/4934>
```
Many applications assume the support of s3tc is basic enough to assume that the driver support it. So it would be better to have it enabled. Many games started to work after landing both patches like Quake4 https://youtu.be/zZaNC2WlYeA

txenoo, 2021-06-16:
> @XECDesign it seems that can be easily backported to 19.3.2, it was directly backported to 20.0 20.1

I've just confirmed that both commits apply perfectly over 19.3.2: And the render is fixed.

XECDesign, 2021-06-22:
I've heard no complaints from internal testing and it seems to work for me, so the fix should be in the repo now. Many thanks, @txenoo.

popcornmix, 2021-06-25:
@hvdkamer can you upgrade through apt and test if this issue is resolved? Please close if it is.