raspberrypi/linux #4395: Raspberry Pi CM4 can't poweroff with command or Desktop operation

bigbearishappy, 2021-06-16:
**Describe the bug**
I use my CM4 with the Raspberry Pi Compute Module 4 IO Board.When I use the command to poweroff the CM4,the Ethernet leds and the D2 led turn off after a few seconds.But the D1(power led) led is still on. 

**To reproduce**
List the steps required to reproduce the issue.
Hardware:CM4+Raspberry Pi Compute Module 4 IO Board
Operation:Power on the CM4,then turn off it with command.
I tried some other command to poweroff the CM4:
halt -p
poweroff
reboot -p
poweroff -p
shuwdown now
shutdown -h now
shutdown -P now
The D1(power led) led is still on. 
Of course. I have tried to turn off the CM4 by the VNC control to it, it doesn't work too.

**System**
The OS I used is download from https://www.raspberrypi.org/software/operating-systems/#third-party-software

* Which model of Raspberry Pi? 
Raspberry Pi Compute Module 4

* Which OS and version (`cat /etc/rpi-issue`)?
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4

* Which firmware version (`vcgencmd version`)?
Apr 30 2021 13:45:52
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

**Additional context**
Add any other relevant context for the problem.
When I use another CM4.The poweroff command works fine to me.It makes me confused.
I am trying to find out the different between the two CM4 I have.After a detail check,I find that when I poweroff the abnormal CM4, the 3.3V(Pin 84 and Pin86) voltage won't turn to 0V but the nomal CM4 does.

I would be appreciate that if someone can help me out~

JamesH65, 2021-06-16:
@dp111 For you!

timg236, 2021-06-16:
Run 'rpi-eeprom-config' on the board which isn't powering off and check that POWER_OFF_ON_HALT=1 and WAKE_ON_GPIO=0

If not, use rpiboot to update the bootloader with the new configuration settings.
https://www.raspberrypi.org/documentation/hardware/computemodule/cm-emmc-flashing.md#bootloader

It's possible that this CM4 has an older manufacturing image.

bigbearishappy, 2021-06-17:
Yes.The CM4 which isn't powering off has the different bootloader configuration.
pi@raspberrypi:/ $ rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try  SD- > USB PCIe MSD -> USB 2.0 BCM XHCI -> Network ->  Loop
BOOT_ORDER=0xf2541

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

I'm trying to update the bootloader with the new configurtation.
I'll give a reply after the update.

bigbearishappy, 2021-06-17:
Thank you so much.The board which isn't powering off works fine after the bootloader configuration update.
It's not easy for to get this result.Because I first use the rpiboot on the windows10 host PC.But it can't exec the update-pieeprom.sh.
So I can not  update the bootloader success.
Maybe I think is necessary to add the tips on the website(https://www.raspberrypi.org/documentation/hardware/computemodule/cm-emmc-flashing.md#bootloader).

timg236, 2021-06-17:
I use Cygwin32 on Windows 10 for this. Alternatively, if you have a Raspberry Pi 3 / 4B then that's the best platform for usbboot.

bigbearishappy, 2021-06-18:
Finally, I use my raspberry pi 4B as the HOST.Thanks for you suggestion.

timg236, 2021-06-18:
Thanks for confirming. Closing issue.

closed 2021-06-18 by timg236