raspberrypi/linux #4401: Non-working graphical environment:  invalid kernel paging request in drivers/gpu/drm/drm_modeset_lock.c on kernel 5.10.44

SteveFosdick, 2021-06-21:
**Describe the bug**
This bug causes both X11 and Wayland graphical environments to fail to start and logs a message to the system journal giving a "Unable to handle kernel paging request" as the probable cause.  This does not occur on linux-raspberrypi4-5.10.43 but does occur on linux-raspberrypi4-5.10.44

**To reproduce**
Install linux-raspberrypi4-5.10.44, boot the Raspberry Pi and attempt to start a graphical environment.  The issue exists with SSDM/LxQT: see https://archlinuxarm.org/forum/viewtopic.php?f=65&t=15398&p=66785#p66785 and with Gnome/gdm see https://archlinuxarm.org/forum/viewtopic.php?f=65&t=15394&start=10

**Expected behaviour**
Graphical environment starts normally.

**Actual behaviour**
On GDM: the login screen is draw but the mouse and keyboard are unresponsive.  It is not possible to switch to another virtual console.

With LxQT/SSDM the follow error message appears on the virtual console:
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: could not load the Qt platform plugin “xcb”  in “” even though it was found
this application failed to start because no Qt platform plugin could be initialized

In each case the following or similar appears in the system journal:
```
Jun 21 12:17:57 alarm kernel: ------------[ cut here ]------------
Jun 21 12:17:57 alarm kernel: WARNING: CPU: 2 PID: 205 at drivers/gpu/drm/drm_modeset_lock.c:246 drm_modeset_lock+0xdc/0x104 [drm]
Jun 21 12:17:57 alarm kernel: Modules linked in: snd_soc_hdmi_codec hid_elecom btsdio bluetooth joydev ecdh_generic vc4 ecc bcm2835_codec(C) bcm2835_v4l2(C) rpivid>
Jun 21 12:17:57 alarm kernel: CPU: 2 PID: 205 Comm: kworker/u8:4 Tainted: G         C        5.10.44-1-ARCH #1
Jun 21 12:17:57 alarm kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
Jun 21 12:17:57 alarm kernel: Workqueue: events_unbound commit_work [vc4]
Jun 21 12:17:57 alarm kernel: pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
Jun 21 12:17:57 alarm kernel: pc : drm_modeset_lock+0xdc/0x104 [drm]
Jun 21 12:17:57 alarm kernel: lr : drm_atomic_get_new_or_current_crtc_state+0x6c/0x114 [drm]
...
Jun 21 12:17:57 alarm kernel: Unable to handle kernel paging request at virtual address fffffffffffffff9
Jun 21 12:17:57 alarm kernel: Mem abort info:
Jun 21 12:17:57 alarm kernel:   ESR = 0x96000005
Jun 21 12:17:57 alarm kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Jun 21 12:17:58 alarm kernel:   SET = 0, FnV = 0
Jun 21 12:17:58 alarm kernel:   EA = 0, S1PTW = 0
Jun 21 12:17:58 alarm kernel: Data abort info:
Jun 21 12:17:58 alarm kernel:   ISV = 0, ISS = 0x00000005
Jun 21 12:17:58 alarm kernel:   CM = 0, WnR = 0
Jun 21 12:17:58 alarm kernel: swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001157000
Jun 21 12:17:58 alarm kernel: [fffffffffffffff9] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Jun 21 12:17:58 alarm kernel: Internal error: Oops: 96000005 [#1] PREEMPT SMP
```
Full version of this attached.

**System**
* Which model of Raspberry Pi? 4B 8Gb.
* Which OS and version (`cat /etc/rpi-issue`)? Arch Linux ARM (rolling release) but with Raspberry Pi (not Arch) kernel.
* Which firmware version (`vcgencmd version`)? version 03df0e450c356822f03454c27edea347fd8d211b (clean) (release) (start)
* Which kernel version (`uname -a`)? 5.10.44

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
[journal.txt](https://github.com/raspberrypi/linux/files/6686406/journal.txt)

**Additional context**
Last known good commit (with this bug) https://github.com/raspberrypi/linux/co ... 06a9016cef   See https://archlinuxarm.org/forum/viewtopic.php?f=65&t=15398&p=66785#p66785

6by9, 2021-06-21:
@mripard. This looks like the same error that Laurent has just seen in trying to bring up a DSI display - https://lists.freedesktop.org/archives/dri-devel/2021-June/311819.html

Somethings clipped the text for the last known good commit - 06a9016cef is not a valid hash in the rpi-5.10.y branch.
Copy/paste from the archlinux forum thread, https://github.com/raspberrypi/linux/commit/86b1bed34352428d0826dff892b3ac06a9016cef. That's the merge commit from 5.10.42 to 5.10.43.

I'd suspect it's part of https://github.com/raspberrypi/linux/pull/4365 that has triggered this.

mripard, 2021-06-21:
I think I found a fix for it, feel free to test #4402

satmandu, 2021-06-21:
Reverting to [5.10.43-v8+](https://github.com/Hexxeh/rpi-firmware/commit/85bda3d8fd6a8b70d189aefd9234440ca99cb81c) via `sudo rpi-update 85bda3d8fd6a8b70d189aefd9234440ca99cb81c` resolves this for me, so there is at least an issue after 5.10.43 was merged which is causing this.

pelwell, 2021-06-21:
Have you tested #4402 ?

satmandu, 2021-06-21:
I have not tested #4402. I've been sticking to the kernel builds on the hexxeh repo for a while, and haven't built a RPI kernel in some time. 😅

satmandu, 2021-06-23:
#4402 fixes this for me.

pelwell, 2021-06-23:
Thanks for taking the time to check this. As the responses say, #4402 introduces some problems with Kodi, but I'm sure @mripard is working on a patch that solves both problems.

pelwell, 2021-06-23:
There's an update to #4402 that may also fix the Kodi problem.

pelwell, 2021-06-23:
The improved patch has been merged after receiving positive feedback. Expect a kernel build in our firmware repo (available via `rpi-update`) in the next few days.

popcornmix, 2021-06-23:
The fix is now in rpi-update kernel.

satmandu, 2021-06-23:
The rpi-update kernel also has the issue resolved for me.

pelwell, 2021-06-23:
@SteveFosdick Please try the rpi-update kernel and close the issue if appropriate.

SteveFosdick, 2021-07-06:
Yes, this is now fixed.

closed 2021-07-06 by SteveFosdick