raspberrypi/linux #4422: DS1307 driver question

seamusdemora, 2021-06-29:
Not sure I'm in the *right place*, but I have a question/request wrt the [DS1307 driver](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/rtc/rtc-ds1307.c) - specifically wrt the DS3231: 

I'm using the DS3231 with my RPi to manage some hardware. The `INT/SQW` pin on the DS3231 asserts `LOW` when the alarm time I have set is reached. I "handle" this interrupt in my peripheral hardware, and it is "serviced" immediately. 

I must eventually "clear" the interrupt in software so that the `INT/SQW` pin returns to a `HIGH` state. This cannot be done in hardware, and so - as it works now - I do this by writing a `0` to the `sysfs` at `/sys/class/rtc/rtc0/wakealarm`. It would be better for me if the alarm interrupt could be cleared automatically by the driver - asap in other words. Ideally, this "auto clearing" of the alarm interrupt would be an option specified in the `dtoverlay` for the RTC. 

Is this something that could be considered as an enhancement - or should I forget about driver mods & find another way?

pelwell, 2021-06-29:
As the bug report template says:
> If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

closed 2021-06-29 by pelwell

seamusdemora, 2021-06-29:
@pelwell : I cannot use the forums

pelwell, 2021-06-30:
Of course you can - they're not that awful.

seamusdemora, 2021-07-01:
Ummm - no... I was "banned for life" - [for this](https://www.raspberrypi.org/forums/viewtopic.php?p=1300747#p1300747).

pelwell, 2021-07-01:
I've checked, and I see no record of a banning. It takes quite a lot to get to that stage, and would normally be preceded by some kind of clear warning.

Try signing in and posting.

seamusdemora, 2024-01-19:
I don't know where or how you checked, but I did try to sign in (a few days after your post & again recently), and it still says that I am "permanently banned".