raspberrypi/linux #4427: Raspberry PI 400 crashes

randomMesh, 2021-07-02:
**Describe the bug**
Raspberry PI 400 crashes and i am not able to recover it other than to power off and on again.

**To reproduce**
n/a

**Expected behaviour**
Shoud not crash.

**Actual behaviour**
My Raspberry PI 400 crashes randomly (after 10 minutes, or after 3 hours of uptime) and i can't do recover it other than to power off.

**System**
Output of `raspinfo`
```
System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage5

Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Revision	: c03130
Serial		: 1000000000b1dcd0
Model		: Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 27 2021 14:02:15 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Dateisystem    1K-Bl\u00f6cke Benutzt Verf\u00fcgbar Verw% Eingeh\u00e4ngt auf
/dev/root       14987608 8460880   5818476   60% /
devtmpfs         1827468       0   1827468    0% /dev
tmpfs            1959564   20744   1938820    2% /dev/shm
tmpfs            1959564   16940   1942624    1% /run
tmpfs               5120       4      5116    1% /run/lock
tmpfs            1959564       0   1959564    0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48788    209307   19% /boot
tmpfs             391912       4    391908    1% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installiert:           1.20201210+nmu1
raspberrypi-sys-mods:
  Installiert:           20210310
openbox:
  Installiert:           3.6.1-8+rpt5
lxpanel:
  Installiert:           0.10.0-2+rpt15
[... 600 lines cut ...]
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538783] [<c07dd700>] (clk_core_set_rate_nolock) from [<c07dd8e4>] (clk_set_rate+0x40/0x98)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538787]  r8:00000040 r7:eff29900 r6:eff26640 r5:00002710 r4:c3912480
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538801] [<c07dd8a4>] (clk_set_rate) from [<bf3bcea0>] (v3d_clock_down_work+0x24/0x68 [v3d])
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538804]  r5:c5567280 r4:c31d83b4
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538818] [<bf3bce7c>] (v3d_clock_down_work [v3d]) from [<c023dc18>] (process_one_work+0x250/0x5a0)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538822]  r5:c5567280 r4:c31d83b4
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538828] [<c023d9c8>] (process_one_work) from [<c023dfc8>] (worker_thread+0x60/0x5c4)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538833]  r10:eff26640 r9:c1203d00 r8:eff26658 r7:00000008 r6:eff26640 r5:c5567294
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538836]  r4:c5567280
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538843] [<c023df68>] (worker_thread) from [<c0245a50>] (kthread+0x170/0x174)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538848]  r10:c1a5de74 r9:c5567280 r8:c023df68 r7:c2b8a000 r6:00000000 r5:c5e2c180
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538851]  r4:c5025780
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538857] [<c02458e0>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538860] Exception stack(0xc2b8bfb0 to 0xc2b8bff8)
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538864] bfa0:                                     00000000 00000000 00000000 00000000
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538869] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538873] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538878]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02458e0
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538881]  r4:c5e2c180
Jul  2 22:06:45 raspberrypi kernel: [ 2053.538886] ---[ end trace f62a40a8c106d916 ]---
Jul  2 22:06:47 raspberrypi kernel: [ 2055.378324] ------------[ cut here ]------------
```

**Additional context**
Add any other relevant context for the problem.

pelwell, 2021-07-05:
You have trimmed the start of the kernel crash/warning log, which makes it harder to interpret. In normal use there shouldn't be much output sent to the kernel log, so the entire output of `dmesg` would be useful (or all the preceding "kernel:" entries in /var/log/messages back to the start of the boot).

randomMesh, 2021-07-05:
> so the entire output of `dmesg` would be useful

Hi,
thank you for your answer. Here's the complete output of `dmesg`:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1421 SMP Thu May 27 14:00:13 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 400 Rev 1.0
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50700 r8192 d23028 u81920
[    0.000000] pcpu-alloc: s50700 r8192 d23028 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:23:2B:FC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=9981d7c3-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000017cc0000-0x000000001bcc0000] (64MB)
[    0.000000] Memory: 3654940K/4050944K available (10240K kernel code, 1358K rwdata, 3160K rodata, 2048K init, 890K bss, 133860K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33942 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=0
[    0.000009] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000028] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000093] bcm2835: system timer (irq = 25)
[    0.000744] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000762] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000780] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000797] Switching to timer-based delay loop, resolution 18ns
[    0.001056] Console: colour dummy device 80x30
[    0.001124] printk: console [tty1] enabled
[    0.001180] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001214] pid_max: default: 32768 minimum: 301
[    0.001391] LSM: Security Framework initializing
[    0.001584] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001611] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003251] Disabling memory control group subsystem
[    0.003376] CPU: Testing write buffer coherency: ok
[    0.003858] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005088] Setting up static identity map for 0x200000 - 0x20003c
[    0.005294] rcu: Hierarchical SRCU implementation.
[    0.006253] smp: Bringing up secondary CPUs ...
[... 355 lines cut ...]
[ 1073.807166] [<bf2f9eec>] (cfg80211_roamed [cfg80211]) from [<bf69edf0>] (brcmf_notify_roaming_status+0x194/0x25c [brcmfmac])
[ 1073.807179]  r8:c18d4000 r7:c75b0000 r6:c1814800 r5:c1815000 r4:c1205048
[ 1073.807394] [<bf69ec5c>] (brcmf_notify_roaming_status [brcmfmac]) from [<bf6ac110>] (brcmf_fweh_call_event_handler+0x44/0x94 [brcmfmac])
[... cut ...]
[ 1073.807882] [<c02458e0>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[ 1073.807892] Exception stack(0xc391ffb0 to 0xc391fff8)
[ 1073.807904] ffa0:                                     00000000 00000000 00000000 00000000
[ 1073.807918] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1073.807931] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1073.807945]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02458e0
[ 1073.807955]  r4:c3742440
[ 1073.807968] ---[ end trace 6df7d2f91d4a6f12 ]---
```

jc-kynesim, 2021-07-06:
I don't think I've ever seen that particular crash but then I don't use bluetooth. The VPU is involved in H264 decode - when V4L2 (or mmal) gets confused then VPU lockup is common enough. Otherwise I can't see why Chrome should be a special app.  Chrome does do GPU compositing and I don't know how much that uses the VPU - this also uses CMA so if a lockup could be caused by CMA exhaustion that might be it too, though I think I'd expect greater disaster if that was the case.

randomMesh, 2021-07-06:
Thank you guys for your answer.

> To what extent is the VPU involved when browsing in Chrome on the Pi 4(00)?

I don't know if it makes a difference but i meant to write i was using Chromium, not Chrome. I corrected my mistake in the first post.

> The VPU is involved in H264 decode - when V4L2 (or mmal) gets confused then VPU lockup is common enough.

I remember that i was browsing Reddit and watching videos when the crash / freeze occured.

> I don't think I've ever seen that particular crash but then I don't use bluetooth.

Do you think the Bluetooth stack is related to this problem? I don't need it so i would remove it completely if relevant.

pelwell, 2021-07-06:
I think Bluetooth is a red herring (false clue) here - I should have said so. I suspect that video playback is the common link - it would help a bit if you can either confirm that seems to be the case or say you have a clear counter-example.

randomMesh, 2021-07-06:
> I suspect that video playback is the common link - it would help a bit if you can either confirm that seems to be the case or say you have a clear counter-example.

As far as i remember all the crashes happened when i was using Chromium to watch videos.

jc-kynesim, 2021-07-06:
Yeah my mention of Bluetooth was due to a misread of the trace.  Either way I don't think I've seen that precise trace in the past, but firmware lockup produce a fine variety of arm-side crashes.  My use of "chrome" was just shorthand and shouldn't be taken as referring to any explicit config.

pelwell, 2021-07-06:
OK - so the next question is whether there are specific "killer" videos, whether the failure happens after some number of videos (resource leak/fragmentation) or whether it's just random (race condition).

We probably shouldn't rule out a power problem. Even with no obvious under-voltage throttling there is the possibility of some critical part of the SoC being under-volted for a brief window. Try putting `over_voltage=2` in config.txt and seeing if that affects reliability. If you do have a different power supply (especially if the current one isn't one of ours) then it's worth switching that as well.

jc-kynesim, 2021-07-06:
There are known to be occasional (probably race) issues with the current V4L2 H264 driver mostly to do with startup / video resize.  I haven't been able to bag Dave's time to debug/sort them...

randomMesh, 2021-07-06:
> Try putting over_voltage=2 in config.txt and seeing if that affects reliability.

I will try this.

> If you do have a different power supply (especially if the current one isn't one of ours) then it's worth switching that as well.

I bought the kit which comes with the official power supply included..

pelwell, 2021-07-06:
If the supply is a genuine Pi supply then I'm less concerned about it. Especially if you don't have power-hungry peripherals attached.

randomMesh, 2021-07-06:
> Especially if you don't have power-hungry peripherals attached.

Other than the mouse and monitor I don't have any peripherals attached.

randomMesh, 2021-07-10:
The crash has not ocured anymore, so i think this issue can be closed.
Thanks all for your help!

closed 2021-07-10 by randomMesh

pelwell, 2021-07-10:
Thanks for the feedback. I can only assume that the small over-voltage has made a difference.

Every die (the active part of an integrated circuit) cut from a wafer has slightly different properties due to small variations in the way the layers are built up. As a result they require slightly different amounts of power to do the same tasks. At boot time the firmware takes some measurements to determine the voltages at key areas of the chip, but they can never give a complete picture. It sounds like your SoC has an element that requires a slightly higher input voltage than the measurement would suggest, and that is what the small "over-voltage" (above what is expected, not excessive) achieves.

mattmcneilgit, 2021-07-26:
I found a similar website that was crashing my PI every single time I landed on it. A "killer" site/video as you mention..

https://ca.store.ui.com/

Would be interested to see if this also affected your Pi. If so, may want to reopen this bug.

randomMesh, 2021-07-26:
Hi Matt,

i browsed that site for a few minutes and suddenly Chromium froze/crashed. Otherwise i was able to use the system as normal. (Using Openbox as a windowmanager instead of the default desktop environment.)

I had a terminal open at the time with the command `dmesg -w` running. Here are the messages:

```
[16757.962617] bcm2835_mmal_vchiq: timed out waiting for sync completion
[16757.962632] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling o/p port, ret -62
[16757.962638] ------------[ cut here ]------------
[16757.962661] WARNING: CPU: 3 PID: 11127 at drivers/media/common/videobuf2/videobuf2-core.c:1548 vb2_start_streaming+0x108/0x178 [videobuf2_common]
[16757.962666] Modules linked in: fuse 8021q garp stp llc brcmfmac brcmutil vc4 sha256_generic v3d cec cfg80211 gpu_sched raspberrypi_hwmon rfkill drm_kms_helper bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) drm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks videodev snd_soc_core mc snd_bcm2835(C) snd_compress vc_sm_cma(C) snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[16757.962925] CPU: 3 PID: 11127 Comm: V4L2DecoderThre Tainted: G         C        5.10.17-v7l+ #1421
[16757.962929] Hardware name: BCM2711
[16757.962933] Backtrace: 
[16757.962951] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
[16757.962957]  r7:ffffffff r6:00000000 r5:68000013 r4:c12e69fc
[16757.962964] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
[16757.962971] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[16757.962976]  r10:bf1398dc r9:00000009 r8:bf1091ec r7:0000060c r6:00000009 r5:bf1091ec
[16757.962980]  r4:bf10df90 r3:c1205094
[16757.962986] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
[16757.962991]  r7:0000060c r6:bf10df90 r5:c1205048 r4:00000000
[16757.963005] [<c0b63ae8>] (warn_slowpath_fmt) from [<bf1091ec>] (vb2_start_streaming+0x108/0x178 [videobuf2_common])
[16757.963010]  r9:bf1cd068 r8:c3730be0 r7:00000001 r6:c35ba814 r5:c35ba914 r4:ffffffc2
[16757.963028] [<bf1090e4>] (vb2_start_streaming [videobuf2_common]) from [<bf1098c8>] (vb2_core_streamon+0x90/0x180 [videobuf2_common])
[16757.963032]  r7:00000001 r6:c1205048 r5:c35ba800 r4:c35ba814
[16757.963048] [<bf109838>] (vb2_core_streamon [videobuf2_common]) from [<bf233044>] (vb2_streamon+0x24/0x64 [videobuf2_v4l2])
[16757.963052]  r5:c35ba800 r4:00000009
[16757.963067] [<bf233020>] (vb2_streamon [videobuf2_v4l2]) from [<bf1220dc>] (v4l2_m2m_streamon+0x30/0x60 [v4l2_mem2mem])
[16757.963082] [<bf1220ac>] (v4l2_m2m_streamon [v4l2_mem2mem]) from [<bf12212c>] (v4l2_m2m_ioctl_streamon+0x20/0x24 [v4l2_mem2mem])
[16757.963086]  r7:00000001 r6:c1205048 r5:40045612 r4:bf12210c
[16757.963127] [<bf12210c>] (v4l2_m2m_ioctl_streamon [v4l2_mem2mem]) from [<bf1cd094>] (v4l_streamon+0x2c/0x30 [videodev])
[16757.963180] [<bf1cd068>] (v4l_streamon [videodev]) from [<bf1d2114>] (__video_do_ioctl+0x234/0x460 [videodev])
[16757.963183]  r5:40045612 r4:c37308a0
[16757.963234] [<bf1d1ee0>] (__video_do_ioctl [videodev]) from [<bf1d2b28>] (video_usercopy+0x128/0x5ac [videodev])
[16757.963240]  r10:c7997e3c r9:9cc754dc r8:00000000 r7:ca437600 r6:c1205048 r5:40045612
[16757.963243]  r4:40045612
[16757.963292] [<bf1d2a00>] (video_usercopy [videodev]) from [<bf1d2fcc>] (video_ioctl2+0x20/0x24 [videodev])
[16757.963297]  r10:00000066 r9:c7996000 r8:00000000 r7:ca437600 r6:ca437601 r5:c1205048
[16757.963301]  r4:bf1d2fac
[16757.963350] [<bf1d2fac>] (video_ioctl2 [videodev]) from [<bf1cb1f4>] (v4l2_ioctl+0x4c/0x60 [videodev])
[16757.963380] [<bf1cb1a8>] (v4l2_ioctl [videodev]) from [<c045178c>] (sys_ioctl+0x1d4/0x8ec)
[16757.963384]  r5:c1205048 r4:40045612
[16757.963391] [<c04515b8>] (sys_ioctl) from [<c02001e4>] (__sys_trace_return+0x0/0x1c)
[16757.963395] Exception stack(0xc7997fa8 to 0xc7997ff0)
[16757.963399] 7fa0:                   9cc754dc 40045612 00000066 40045612 9cc754dc 037d02ec
[16757.963404] 7fc0: 9cc754dc 40045612 0adac940 00000036 9cc75508 00000000 9cc75504 9cc75428
[16757.963408] 7fe0: 08fd8a48 9cc75414 037d0310 b54cd51c
[16757.963413]  r10:00000036 r9:c7996000 r8:c0200204 r7:00000036 r6:0adac940 r5:40045612
[16757.963416]  r4:9cc754dc
[16757.963421] ---[ end trace 5f170f0ff5f2d8e1 ]---
[16761.002695] bcm2835_mmal_vchiq: timed out waiting for sync completion
[16761.002725] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling o/p port, ret -62
[16761.002748] ------------[ cut here ]------------
[16761.002801] WARNING: CPU: 2 PID: 11117 at drivers/media/common/videobuf2/videobuf2-core.c:1548 vb2_start_streaming+0x108/0x178 [videobuf2_common]
[16761.002815] Modules linked in: fuse 8021q garp stp llc brcmfmac brcmutil vc4 sha256_generic v3d cec cfg80211 gpu_sched raspberrypi_hwmon rfkill drm_kms_helper bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) drm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks videodev snd_soc_core mc snd_bcm2835(C) snd_compress vc_sm_cma(C) snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[... cut ...]
[16950.686766] ---[ end trace 5f170f0ff5f2d8ea ]---
[16950.686786] videobuf2_common: driver bug: stop_streaming operation is leaving buf c23bf5a7 in active state
[16950.686812] videobuf2_common: driver bug: stop_streaming operation is leaving buf 5ffa1e8b in active state
[16953.724898] bcm2835_mmal_vchiq: timed out waiting for sync completion
[16956.764999] bcm2835_mmal_vchiq: timed out waiting for sync completion
```

I don't know if i should reopen this issue since it's a bit different of an error.

pelwell, 2021-07-27:
Although the Uniquiti Canada homepage loads OK, I get a kernel WARN caused by a timeout on the first product page: https://ca.store.ui.com/products/udm-pro
Increasing the amount of memory reserved for the GPU/VPU seems to help (put "gpu_mem=128" in config.txt) - can you try that?

@jc-kynesim The relocatable heap is showing signs of corruption, but that may just be because allocations stall when it becomes full. Is this to be expected on some sites with the default gpu_mem setting?

popcornmix, 2021-07-27:
@pelwell were you checking from arm side or gpu side?
You need a `vcgencmd cache_flush` before `vcdbg reloc` as the arm doesn't see the L1 cache (and even that can be racy if allocations are frequently occurring).

pelwell, 2021-07-27:
Understood. It was from the ARM side, but `vcgencmd cache_flush` needs VPU cooperation which was lacking at the time. I don't think the problem is caused by reloc heap corruption, but by running out of reloc heap space.

randomMesh, 2021-07-27:
> Increasing the amount of memory reserved for the GPU/VPU seems to help (put "gpu_mem=128" in config.txt) - can you try that?

The time the crash happpened, GPU memory already has been increased to 128MB. I now increased it further to 256MB.

pelwell, 2021-07-27:
I've started seeing those bcm2835-codec message recently. I'm opening a separate issue for them.