raspberrypi/linux #4428: pi4/fkms: boot fails due to flip_done timed out (now 1080p and lower)

alanbork, 2021-07-03:
under f29ab05611eef385d17675c77190df0b87a0d456, I can't get the pi to fully boot with fkms enabled. that is to say it boots but the screen shuts down mid-boot and the machine becomes so unresponsive that I frequently can't ssh in.

 it's failing just like it used to for 4k resolutions (https://github.com/raspberrypi/linux/issues/4271), with a

 [  798.560610] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out 

but now:

* overclocking doesn't help
* using a lower resolution doesn't help (1080p, 480p; I haven't tried that many since it's so hard to recover from this with SSH unresponsive)

revert to 039088f7d929b0d1cc49e8d2012df0652f908518 and it boots with no problems.

config.txt:

```
dtoverlay=vc4-fkms-v3d
max_framebuffers=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
hdmi_enable_4kp60=1
```

WRT overclocking, and unlike the 4k issue, adding these lines don't help:
```
#overclcking option can be toggled on of off, no change 
core_freq=600
core_freq_min=600
```

popcornmix, 2021-07-05:
Can you try with latest rpi-update kernel?

alanbork, 2021-07-07:
did rpi-update 9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2

working now at 1080p, also 3840x2160-60.00hz(60).

closed 2021-07-07 by alanbork