raspberrypi/linux #4434: v3d fec00000.v3d : MMU error from client L2T (0) at 0x3xxxxxx, pte invalid

Ash-S, 2021-07-06:
I am trying to run my MIPI interfaced camera driver on RPi. while running this command to check the 1 frame, its gives me error, no idea how to resolve it. Please help. Thanks.

`v4l2-ctl --stream-mmap --stream-count=1 -d /dev/video0 --stream-to=/home/pi/test.raw`

**Error:** ```
v3d fec00000.v3d: MMU error from client L2T (0) at 0x3481000, pte invalid
          unicam fe801000.csi: stream off failed in subdev
```

6by9, 2021-07-06:
Also this is a Linux kernel issue and not userland, so transferred appropriately.

Ash-S, 2021-07-08:
@6by9 Thanks for your suggestion. I tried to use it after creating buffers to it. and then i tested like....

When execute v4l2-ctl streaming command:
```
[ 9701.854083] videobuf2_common: [cap-(ptrval)] vb2_core_streamon: successful
[ 9701.854136] videobuf2_common: [cap-(ptrval)] __vb2_wait_for_done_vb: will sleep waiting for buffers
```

When used ctrl-c to fore stop
```
[ 9704.162105] videobuf2_common: [cap-(ptrval)] __vb2_wait_for_done_vb: sleep was interrupted
[ 9704.163556] unicam fe801000.csi: stream off failed in subdev
```

Please suggest~ Thanks.

qrp73, 2023-08-25:
any news? I don't use any camera, but error is still there. It appears in the log just after system boot.