raspberrypi/linux #4436: 5.12.y drm breakage bisected to 888d3a839211 "drm/vc4: Leverage the load tracker on the BCM2711"

yetanotheraccount, 2021-07-07:
5.12.y branch at head (490c31351493) boots/works fine, albeit with a black HDMI monitor when modeset kicks in .

I bisected it to 888d3a839211 "drm/vc4: Leverage the load tracker on the BCM2711".  However, reverting ONLY that commit leaves cute as button little rpi4 box in the even worse state of not finishing boot.  To restore box to fully functional, I have to revert the next commit, 490c31351493 "drm/vc4: Increase the core clock based on HVS load", as well.

pelwell, 2021-07-07:
rpi-5.12.y and rpi-5.11.y are lag behind the rpi-5.10.y branch - they are updated periodically when somebody (usually me) gets round to it.

yetanotheraccount, 2021-07-08:
> rpi-5.12.y and rpi-5.11.y are lag behind the rpi-5.10.y branch - they are updated periodically when somebody (usually me) gets round to it.

My round toit regularly goes missing :)   FWIW, the shiny new 5.13.y branch has the same issue.  Attached, please find the dmesg output I (oopsie) forgot to provide originally.

[dmesg-rpi4.txt](https://github.com/raspberrypi/linux/files/6781226/dmesg-rpi4.txt)

yetanotheraccount, 2021-07-08:
By way of due diligence, I went back and recreated the 5.12.y branch oops.

Correction wrt the 5.13.y issue being the same: well, almost, but not quite.  Reverting the same two commits that resulted in a happy camper 5.12.y kernel did NOT do the same for 5.13.y, it instead ended in a boot failure.  I don't have serial console capability ATM, and kdump is busted, so sorry, no data on whatever's going on with that.

[dmesg-5.12.10-v8.txt](https://github.com/raspberrypi/linux/files/6782017/dmesg-5.12.10-v8.txt)

6by9, 2021-07-08:
Saying which platform you're on would help. 64bit kernel by the looks of it, 

Looking at the backtrace it appears to be the load tracker being called when using FKMS, and it shouldn't.
See https://github.com/raspberrypi/linux/commit/242edb3b4f1cd1225d43f2c5b5a5ba8beefdf216 for the fix.

yetanotheraccount, 2021-07-08:
> Saying which platform you're on would help. 64bit kernel by the looks of it,

Yup, rpi4b running 64bit kernel(s) and 64bit Pi-OS.

> Looking at the backtrace it appears to be the load tracker being called when using FKMS, and it shouldn't.
> See [242edb3](https://github.com/raspberrypi/linux/commit/242edb3b4f1cd1225d43f2c5b5a5ba8beefdf216) for the fix.

That produced mixed results.  Adding 242edb3b4f1c cured the 5.12.y branch kernel, but killed the 5.13.y kernel, so there's a part 2 to that branch's woes.

pelwell, 2021-07-08:
I've pushed a set of commits to rpi-5.12.y that should bring it in line with rpi-5.10.y. Please test and close if the problem is solved.

yetanotheraccount, 2021-07-08:
> I've pushed a set of commits to rpi-5.12.y that should bring it in line with rpi-5.10.y. Please test and close if the problem is solved.

Yup, 5.12.y branch is all better.  Closing.   (off to stare at updated 5.13.y changes.. or maybe just build/boot and see what happens... more anticipation;)

closed 2021-07-08 by yetanotheraccount