raspberrypi/linux #4439: Kernel 5.13 aarch64 issue

Dark-Sky, 2021-07-08:
Raspberrypi 8G
gcc version 10.2.0
Compiled with "make bcm2711_defconfig" then compiled kernel

I am unable to boot into lightdm greeter. I also tried sddm greeter using "dtoverlay=vc4-fkms-v3d". I am able to boot to the desktop with "dtoverlay=vc4-kms-v3d". I have no issue booting with kernel's 5.10 or 5.12. I get an Internal oops wihen using "dtoverlay=vc4-fkms-v3d":

```
[    6.814817] vc4-drm gpu: bound fe600000.firmwarekms (ops vc4_fkms_ops [vc4])
[    6.838211] checking generic (3e2fe000 7f8000) vs hw (0 ffffffffffffffff)
[    6.838247] fb0: switching to vc4drmfb from simple
[    6.844506] Console: switching to colour dummy device 80x25
[    6.845388] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[    6.845435] Mem abort info:
[    6.845453]   ESR = 0x96000005
[    6.845473]   EC = 0x25: DABT (current EL), IL = 32 bits
[    6.845496]   SET = 0, FnV = 0
[    6.845515]   EA = 0, S1PTW = 0
[    6.845533] Data abort info:
[    6.845549]   ISV = 0, ISS = 0x00000005
[    6.845568]   CM = 0, WnR = 0
[    6.845586] user pgtable: 4k pages, 39-bit VAs, pgdp=000000010a975000
[    6.845611] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    6.845668] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[    6.845689] Modules linked in: brcmfmac brcmutil bcm2835_v4l2(C) vc4(+) bcm2835_codec(C) cec v3d sha256_generic bcm2835_isp(C) drm_kms_helper bcm2835_mmal_vchiq(C) gpu_sched videobuf2_vmalloc cfg80211 v4l2_mem2mem rfkill videobuf2_dma_contig videobuf2_memops snd_bcm2835(C) videobuf2_v4l2 videobuf2_common snd_soc_core raspberrypi_hwmon snd_compress snd_pcm_dmaengine snd_pcm vc_sm_cma(C) snd_timer videodev snd mc syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem nvmem_rmem uio_pdrv_genirq uio sch_fq_codel uinput crypto_user drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    6.846003] CPU: 2 PID: 197 Comm: systemd-udevd Tainted: G         C        5.13.0-1-v8 #1
[    6.846029] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[    6.846046] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[    6.846068] pc : vc4_debugfs_init+0x34/0xac [vc4]
[    6.846180] lr : drm_debugfs_init+0x10c/0x140 [drm]
[    6.846425] sp : ffffffc0128d3690
[    6.846438] x29: ffffffc0128d3690 x28: ffffffc010039840 x27: 0000000000000000
[    6.846473] x26: ffffffc01143c728 x25: ffffff810dca0680 x24: ffffffc0092df148
[    6.846506] x23: 0000000000000001 x22: ffffff8100e60cc0 x21: ffffff810dcd4000
[    6.846537] x20: ffffff810dca0780 x19: ffffff810dca0780 x18: 0000000000000020
[    6.846568] x17: 0000000000000001 x16: 0000000000000001 x15: ffffff81034ba378
[    6.846597] x14: ffffffffffffffff x13: ffffffc0928d3707 x12: 0000000000000000
[    6.846627] x11: 0101010101010101 x10: 0000000000000000 x9 : ffffffc008c7dd3c
[    6.846658] x8 : 7f7f7f7f7f7f7f7f x7 : ffffff81034b9e40 x6 : 0000000000000000
[    6.846687] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffff810dca07b0
[    6.846716] x2 : 0000000000000000 x1 : ffffffc0092ebb38 x0 : 0000000000000000
[    6.846747] Call trace:
[    6.846759]  vc4_debugfs_init+0x34/0xac [vc4]
[    6.846853]  drm_debugfs_init+0x10c/0x140 [drm]
[    6.847075]  drm_minor_register+0x64/0x120 [drm]
[    6.847292]  drm_dev_register+0xb4/0x220 [drm]
[    6.847505]  vc4_drm_bind+0x1d0/0x210 [vc4]
[    6.847591]  try_to_bring_up_master+0x16c/0x1d4
[    6.847619]  component_master_add_with_match+0xbc/0x104
[    6.847640]  vc4_platform_drm_probe+0xd4/0x110 [vc4]
[    6.847726]  platform_probe+0x74/0xe4
[    6.847747]  really_probe+0xec/0x50c
[    6.847765]  driver_probe_device+0x6c/0xd0
[    6.847782]  device_driver_attach+0xcc/0xd4
[    6.847800]  __driver_attach+0x9c/0x13c
[    6.847817]  bus_for_each_dev+0x7c/0xd0
[    6.847841]  driver_attach+0x30/0x40
[    6.847857]  bus_add_driver+0x114/0x200
[    6.847878]  driver_register+0x84/0x140
[    6.847895]  __platform_driver_register+0x34/0x40
[    6.847915]  vc4_drm_register+0x54/0x1000 [vc4]
[    6.848003]  do_one_initcall+0x54/0x2c0
[    6.848025]  do_init_module+0x60/0x29c
[    6.848048]  load_module+0x1fd4/0x23e0
[    6.848070]  __do_sys_finit_module+0xb8/0x100
[    6.848092]  __arm64_sys_finit_module+0x2c/0x40
[    6.848115]  invoke_syscall+0x50/0x120
[    6.848138]  el0_svc_common.constprop.0+0xf4/0x104
[    6.848160]  do_el0_svc+0x30/0x9c
[    6.848181]  el0_svc+0x2c/0x54
[    6.848204]  el0_sync_handler+0x1a4/0x1b0
[    6.848226]  el0_sync+0x198/0x1c0
[    6.848250] Code: 912ce021 f90013f5 f9400815 f9431ea0 (f9400000) 
[    6.848270] ---[ end trace 7bf4541b6842b390 ]---
```

popcornmix, 2021-07-08:
Be ware that 5.13 kernel is very new and has a number of conflicts with kms drivers (as more patches appear upstream in slightly different forms), so issues are not unexpected.

I do have a local tree that replaces more downstream patches with upstream or pending (i.e. in mailing lists discussion), and that is booting with fkms for me. I'll push it later and you can see if it helps.

Dark-Sky, 2021-07-08:
> Be ware that 5.13 kernel is very new and has a number of conflicts with kms drivers (as more patches appear upstream in slightly different forms), so issues are not unexpected.
>
> I do have a local [...]

Thanks! I will be glad to test it.

pelwell, 2021-07-08:
I've just pushed some updates to rpi-5.13.y that should bring it in line with rpi-5.10.y.

Dark-Sky, 2021-07-08:
> I've just pushed some updates to rpi-5.13.y that should bring it in line with rpi-5.10.y.

I do not know if you meant this as a FYI but I compiled it:
 
The  internal oops is now gone
I lost the ability to boot to the Desktop with "dtoverlay=vc4-fkms-v3d"
The only way I can now boot to the Desktop is disable all dtoverlay=vc4-* and install fbturbo
 
[image]

pelwell, 2021-07-09:
`Linux raspberrypi 5.13.0-v8+ #312 SMP PREEMPT Fri Jul 9 09:11:43 BST 2021 aarch64 GNU/Linux` works for me, with either `vc4-kms-v3d` or `vc4-fkms-v3d` (32-bit userland).

Dark-Sky, 2021-07-09:
Will not work here with vc4-kms-v3d or vc4-fkms-v3d without fbtubo installed to get V3D rendering in glxgears. The boot process when it tries to load a Greeter to log in stops. There is no issue when I load either the latest 5.10 or 5.12 kernels.

Although it is a manjaro-arm distro our mirrors sync with arch-arm mirrors so the DE packages are the same as arch-arm's. The current 5.13 kernel I am testing the .config was generated with your own "bcm2711_defconfig".

If you need an alternate 64bit OS image one can be downloaded here. (I have not tested the Pi OS 64bit image):

https://github-releases.githubusercontent.com/339391081/8ba3f680-dcb2-11eb-9dd7-e7aaa43c3d60?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210709%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210709T131836Z&X-Amz-Expires=300&X-Amz-Signature=7adb03e9832f1dfa9b79880979d1c510b2e9ebfffcf1c3ce2e715115c20d758e&X-Amz-SignedHeaders=host&actor_id=22453495&key_id=0&repo_id=339391081&response-content-disposition=attachment%3B%20filename%3DManjaro-ARM-xfce-rpi4-21.07.img.xz&response-content-type=application%2Foctet-stream

Snapshot of where I replaced the failing 5.13 kernel with the latest 5.10 kernel and rebooted using the same config's and it works:

[image]

Dark-Sky, 2021-07-10:
I installed sddm today and got more than messages of it failing other than lightdm just saying "failed" in journalctl with kernel 5.13 trying to boot with glamor X acceleration enabled on V3D 4.2:

```
Jul 10 08:48:47 pi4 sddm[392]: Initializing...
Jul 10 08:48:47 pi4 sddm[392]: Starting...
Jul 10 08:48:47 pi4 sddm[392]: Logind interface found
Jul 10 08:48:47 pi4 sddm[392]: Adding new display on vt 1 ...
Jul 10 08:48:47 pi4 sddm[392]: Loading theme configuration from ""
Jul 10 08:48:47 pi4 sddm[392]: Display server starting...
Jul 10 08:48:47 pi4 sddm[392]: Adding cookie to "/var/run/sddm/{ece949ab-2b97-46ea-9469-0b281b1111f6}"
Jul 10 08:48:48 pi4 sddm[392]: Running: /usr/bin/X -nolisten tcp -background none -seat seat0 vt1 -auth /var/run/sddm/{ece949ab-2b97-46ea-9469-0b281b1111f6} -noreset -displayfd 18
Jul 10 08:48:48 pi4 sddm[392]: Failed to read display number from pipe
Jul 10 08:48:48 pi4 sddm[392]: Display server stopping...
Jul 10 08:48:48 pi4 sddm[392]: Attempt 1 starting the Display server on vt 1 failed
```
Full xorg log with it failing:

```[    15.093] (--) Log file renamed from "/var/log/Xorg.pid-466.log" to "/var/log/Xorg.0.log"
[    15.096] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    15.096] Build Operating System: Linux Archlinux
[    15.096] Current Operating System: Linux pi4 5.13.1-1-v8 #1 SMP PREEMPT Fri Jul 9 21:21:53 CDT 2021 aarch64
[    15.096] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1360x768M@60 smsc95xx.macaddr=DC:A6:32:AF:B8:7B vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  root=LABEL=ROOT_MNJRO rw rootwait console=ttyS0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyS0,115200 usbhid.mousepoll=8 snd-bcm2835.enable_compat_alsa=0 audit=0
[    15.097] Build Date: 13 April 2021  09:16:13PM
[    15.097]  
[    15.097] Current version of pixman: 0.40.0
[    15.097] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    15.097] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    15.098] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Jul 10 10:03:58 2021
[    15.099] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    15.099] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    15.099] (==) No Layout section.  Using the first Screen section.
[    15.100] (==) No screen section available. Using defaults.
[    15.100] (**) |-->Screen "Default Screen Section" (0)
[    15.100] (**) |   |-->Monitor "<default monitor>"
[    15.101] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    15.101] (==) Automatically adding devices
[    15.101] (==) Automatically enabling devices
[    15.101] (==) Automatically adding GPU devices
[    15.101] (==) Automatically binding GPU devices
[    15.101] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    15.102] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/misc".
[    15.102] 	Entry deleted from font path.
[    15.102] 	(Run 'mkfontdir' on "/usr/share/fonts/misc").
[    15.102] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/TTF".
[    15.102] 	Entry deleted from font path.
[    15.102] 	(Run 'mkfontdir' on "/usr/share/fonts/TTF").
[    15.102] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/OTF".
[    15.102] 	Entry deleted from font path.
[    15.102] 	(Run 'mkfontdir' on "/usr/share/fonts/OTF").
[    15.102] (WW) The directory "/usr/share/fonts/Type1" does not exist.
[    15.102] 	Entry deleted from font path.
[    15.102] (WW) The directory "/usr/share/fonts/100dpi" does not exist.
[    15.103] 	Entry deleted from font path.
[    15.103] (WW) The directory "/usr/share/fonts/75dpi" does not exist.
[    15.103] 	Entry deleted from font path.
[    15.103] (==) FontPath set to:
	
[    15.103] (==) ModulePath set to "/usr/lib/xorg/modules"
[    15.103] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    15.103] (II) Module ABI versions:
[    15.103] 	X.Org ANSI C Emulation: 0.4
[    15.103] 	X.Org Video Driver: 24.1
[    15.103] 	X.Org XInput driver : 24.1
[... 104 lines cut ...]
[     7.597] (II) LoadModule: "modesetting"
[     7.597] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[     7.605] (II) Module modesetting: vendor="X.Org Foundation"
[     7.605] 	compiled for 1.20.11, module version = 1.20.11
[     7.605] 	Module class: X.Org Video Driver
[     7.605] 	ABI class: X.Org Video Driver, version 24.1
[     7.605] (II) LoadModule: "fbdev"
[     7.606] (WW) Warning, couldn't open module fbdev
[     7.606] (EE) Failed to load module "fbdev" (module does not exist, 0)
[     7.606] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[     7.608] (II) modeset(0): using drv /dev/dri/card0
[     7.609] (II) modeset(0): Creating default Display subsection in Screen section
	"Default Screen Section" for depth/fbbpp 24/32
[     7.609] (==) modeset(0): Depth 24, (==) framebuffer bpp 32
[     7.609] (==) modeset(0): RGB weight 888
[     7.609] (==) modeset(0): Default visual is TrueColor
[     7.609] (II) Loading sub module "glamoregl"
[     7.609] (II) LoadModule: "glamoregl"
```
I noticed with the 5.10 kernel when loading properly in the xorg log it adds /dev/dri/card0 & /dev/dri/card1 and uses drv /dev/dri/card0 but with the failed 5.13 kernel /dev/dri/card1 does not exist:

[ray@pi4 ~]$ ls -lah /dev/dri/card0 
crw-rw---- 1 root video 226, 0 May 19 09:11 /dev/dri/card0
[ray@pi4 ~]$ ls -lah /dev/dri/card1
ls: cannot access '/dev/dri/card1': No such file or directory

Dark-Sky, 2021-07-16:
There seems to be some progress today I am able to boot with V3D with kms with commit here: 

https://github.com/raspberrypi/linux/commit/0956b5e2a4bc81d8f3f903b797b34fcf2a699882

With fkms It will boot up to a certain point then the screen goes solid black. I can not get to the pi4 via ssh to see the logs.

Booted with kms:

[image]

jameshilliard, 2021-07-23:
So while trying to debug a wpewebkit crash on 2697f7403187bb2bb61cc716f33ee9f6cfb9af7c I ran into this which looks to be potentially related:
```
[  126.355869] ------------[ cut here ]------------
[  126.360557] WARNING: CPU: 0 PID: 207 at kernel/rcu/tree_plugin.h:297 rcu_note_context_switch+0x64/0x540
[  126.369951] Modules linked in: 8021q garp stp llc iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle brcmfmac brcmutil cfg80211 smsc95xx usbnet mii rfkill bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_codec(C) raspberrypi_hwmon bcm2835_mmal_vchiq(C) i2c_bcm2835 v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig sdhci_iproc videobuf2_memops vc_sm_cma(C) videobuf2_v4l2 videobuf2_common uio_pdrv_genirq uio v4l2loopback(O) videodev fuse
[  126.410276] CPU: 0 PID: 207 Comm: python Tainted: G         C O      5.10.52-v8 #1
[  126.417848] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[  126.423686] pstate: 20000085 (nzCv daIf -PAN -UAO -TCO BTYPE=--)
[  126.429702] pc : rcu_note_context_switch+0x64/0x540
[  126.434586] lr : rcu_note_context_switch+0x38/0x540
[  126.439462] sp : ffffffc0109fba80
[  126.442778] x29: ffffffc0109fba80 x28: ffffffde9afa7000 
[  126.448104] x27: ffffffb045d99e00 x26: 0000000000000000 
[  126.453426] x25: ffffffde9a5ff0e0 x24: ffffffb045d99e00 
[  126.458748] x23: ffffffb045d99e00 x22: 0000000000000000 
[  126.464070] x21: ffffffb045d99e00 x20: ffffffde9ab1d040 
[  126.469392] x19: ffffffb0735d6040 x18: 0000000000000000 
[  126.474715] x17: 000001165db008ff x16: 000000000000000f 
[  126.480039] x15: 031efa151093b124 x14: 00bcfb929b4cfbd0 
[  126.485360] x13: ffffffde9a61b4b0 x12: 000000000000037d 
[  126.490682] x11: 00000000b8fbaf46 x10: 00000000031efa15 
[  126.496003] x9 : ffffffde9a5fe87c x8 : 000000000000000f 
[  126.501326] x7 : ffffffde9add9000 x6 : 0000000000000001 
[  126.506647] x5 : 0000000000000001 x4 : ffffffc0109fbb60 
[  126.511969] x3 : ffffffde9ab08000 x2 : 0000000000000001 
[  126.517292] x1 : ffffffde9add9720 x0 : 0000000000000001 
[  126.522615] Call trace:
[  126.525069]  rcu_note_context_switch+0x64/0x540
[  126.529608]  __schedule+0xcc/0x8e0
[  126.533015]  schedule+0x50/0x110
[  126.536245]  io_schedule+0x48/0x70
[  126.539653]  wait_on_page_bit_common+0x120/0x3b0
[  126.544274]  __lock_page_or_retry+0x194/0x210
[  126.548635]  do_swap_page+0x204/0x710
[  126.552298]  handle_mm_fault+0x538/0xc80
[  126.556239]  do_page_fault+0x14c/0x3e0
[  126.559993]  do_translation_fault+0xbc/0xd8
[  126.564181]  do_mem_abort+0x4c/0xb0
[  126.567672]  el0_da+0x40/0x60
[  126.570641]  el0_sync_handler+0x168/0x1b0
[  126.574656]  el0_sync+0x180/0x1c0
[  126.577971] ---[ end trace 15c8775af0691031 ]---
```

Dark-Sky, 2021-07-23:
I think so also. My issues started when 5.13 came out and saw similar issues with 5,14. Today I compiled the new 5.10.52 and looks like the issues has crept into it now trying to boot with v3d/kms.

popcornmix, 2021-07-23:
Looks like this issue: https://github.com/raspberrypi/linux/issues/4465#issuecomment-885480751
We switched to upstream forms of drm commits which seems to have introduced this issue to 5.10.

mkreisl, 2021-07-26:
I have only created a 32bit kernel here yesterday, but have similar problems, so I report here
```
[   48.487848] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   48.491800] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   48.491824] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   48.498084] init: xbian-failaccess main process (957) killed by TERM signal
[   48.504409] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   48.504455] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   48.519347] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   48.519383] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   48.535272] Console: switching to colour frame buffer device 240x67
[   48.535389] 8<--- cut here ---
[   48.535395] Unable to handle kernel NULL pointer dereference at virtual address 00000010
[   48.535401] pgd = 6a9a54fd
[   48.535409] [00000010] *pgd=02e13003, *pmd=00000000
[   48.535427] Internal error: Oops: 206 [#1] PREEMPT SMP ARM
[   48.535432] Modules linked in: vc4(+) cec drm_kms_helper bcm2835_codec(C) bcm2835_isp(C) rpivid_hevc(C) bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_mem2mem v3d snd_soc_core videobuf2_vmalloc videobuf2_memops gpu_sched videobuf2_v4l2 snd_compress videobuf2_common snd_pcm_dmaengine drm raspberrypi_hwmon syscopyarea videodev sysfillrect vc_sm_cma(C) ir_nec_decoder sysimgblt drm_panel_orientation_quirks mc fb_sys_fops sg gpio_ir_recv uio_pdrv_genirq uio frandom snd_bcm2835(C) snd_pcm snd_timer snd iscsi_tcp libiscsi_tcp libiscsi zfs(PO) zunicode(PO) zzstd(O) zlua(O) zcommon(PO) znvpair(PO) zavl(PO) icp(PO) spl(O) bonding lan78xx smsc95xx rpcsec_gss_krb5 evdev brcmfmac tun cfg80211 uinput rfkill brcmutil genet
[   48.535614] CPU: 2 PID: 2208 Comm: systemd-udevd Tainted: P         C O      5.13.4+ #1
[   48.535624] Hardware name: BCM2711
[   48.535629] PC is at vc4_plane_atomic_check+0x164/0x528 [vc4]
[   48.535697] LR is at vc4_plane_atomic_check+0x140/0x528 [vc4]
[   48.535752] pc : [<bf804398>]    lr : [<bf804374>]    psr: 60000013
[   48.535756] sp : c3e95580  ip : c3e95580  fp : c3e95614
[   48.535760] r10: eeb75800  r9 : bf825b50  r8 : c3b6c61c
[   48.535763] r7 : bf7ec87c  r6 : c3f8b000  r5 : c2761800  r4 : c3b6c440
[   48.535766] r3 : 00000000  r2 : c3b6f440  r1 : 00000031  r0 : 00000001
[   48.535770] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   48.535776] Control: 30c5383d  Table: 2ea00a80  DAC: 55555555
[   48.535779] Register r0 information: non-paged memory
[   48.535789] Register r1 information: non-paged memory
[   48.535795] Register r2 information: slab kmalloc-1k start c3b6f400 pointer offset 64 size 1024
[   48.535817] Register r3 information: NULL pointer
[   48.535826] Register r4 information: slab kmalloc-1k start c3b6c400 pointer offset 64 size 1024
[   48.535840] Register r5 information: slab kmalloc-512 start c2761800 pointer offset 0 size 512
[   48.535854] Register r6 information: slab kmalloc-192 start c3f8b000 pointer offset 0 size 192
[   48.535867] Register r7 information: 48-page vmalloc region starting at 0xbf7ca000 allocated at load_module+0xb30/0x2a90
[   48.535884] Register r8 information: slab kmalloc-1k start c3b6c400 pointer offset 540 size 1024
[   48.535901] Register r9 information: 64-page vmalloc region starting at 0xbf7fb000 allocated at load_module+0xb30/0x2a90
[   48.535910] Register r10 information: slab kmalloc-2k start eeb75800 pointer offset 0 size 2048
[   48.535924] Register r11 information: non-slab/vmalloc memory
[   48.535930] Register r12 information: non-slab/vmalloc memory
[   48.535935] Process systemd-udevd (pid: 2208, stack limit = 0x65aca5a9)
```
Needless to say that it works fine up to the 5.12 kernel

when I remove these two lines 
```
dtoverlay=vc4-fkms-v3d,cma-512
dtoverlay=rpivid-v4l2
```
the kernel OOPs does not come. But then of course Kodi does not start

Somehow it gets worse from kernel version to kernel version. Under progress I understand something else than if the things run from version to version worse and worse.

Sorry, but I had to say that. It seems to me that everything is systematically broken here (and of course seen from Kodi, but this is different thing)

pelwell, 2021-07-26:
The vc4 driver is undergoing rapid change at the moment. We're heading in the right direction but there are bumps along the way, and users choosing to take the path less travelled are more likely to encounter potholes.

mkreisl, 2021-07-26:
> The vc4 driver is undergoing rapid change at the moment. We're heading in the right direction but there are bumps along the way, and users choosing to take the path less travelled are more likely to encounter potholes.

These are not just bumps, but huge potholes, as it seems to me. And quite a few will probably only be closed again in the never-never-ending da, see for example  https://github.com/raspberrypi/linux/issues/4112

That's what I meant, everything I use (basically just a well working Kodi) is getting successively worse. and the path I'm on seems to be going nowhere.

Dark-Sky, 2021-09-22:
@popcornmix I now have  fkms with V3d rendering on the latest 5.14 and 5.15 kernels. Thanks....
I have not tested this 5.13 kernel lately as it is not much interest to me now since it is EOL.
I will leave it up to you to close this issue.

popcornmix, 2021-09-22:
Good to hear.

closed 2021-09-22 by popcornmix

mkreisl, 2021-09-23:
I read this with pleasure. however, the frustration came back promptly after I did a quick test with kernel 5.14.4. 
Kodi starts normally and surface also looks in order
But when playing HEVC videos there is always a crash. AVC videos (h.264) on the other hand are played without errors. 
At least a progress

Dark-Sky, 2021-09-23:
> I read this with pleasure. however, the frustration came back promptly after I did a quick test with kernel 5.14.4.
> Kodi starts normally and surface also looks in order
> But when playing HEVC videos [...]

Although a little off topic of this thread and do not stray away from the 5.10 kernel for kodi I tested it and can comfirm HEVC video crashes big time with the 5.14.4 kernel. I am guessing they probably know about it but here are the logs:

journalctl:

https://drive.google.com/file/d/15F7uEgVN5K0VDlTY3Zsl5E_p0PF1NwLR/view?usp=sharing

kodi crash log:

https://drive.google.com/file/d/1mp1KQdtcTu4AIjTJqvy5rT_zZRYMwchN/view?usp=sharing

mkreisl, 2021-10-09:
Just wanted to report that with the kernel 5.14.9 built today the playback of HEVC videos works again (Kodi v20a1 with current rpi-ffmpeg branch from jc-kynesim)