raspberrypi/linux #4443: USB HUB problems

appleimperio, 2021-07-09:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
When connect a hub to the pi the devices disconnect after a few minutes. I try with a known good pi and a brand new hub and the problem persist. The only way it works is if I connect the devices directly to the PI. Both Pi are 4B one 8gb version and the other 4Gb

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:
Hardware	: BCM2835
Revision	: d03114
Model		: Raspberry Pi 4 Model B Rev 1.4
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?
Linux mastermind 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

[  210.205870] usb 1-1.1: clear tt 2 (9093) error -71
[  210.206484] usb 1-1.1: clear tt 2 (9093) error -71
[  210.207109] usb 1-1.1: clear tt 2 (9093) error -71
[  210.207712] usb 1-1.1: clear tt 2 (9093) error -71
[  210.208363] usb 1-1.1: clear tt 2 (9093) error -71
[  210.208996] usb 1-1.1: clear tt 2 (9093) error -71
[  210.209609] usb 1-1.1: clear tt 2 (9093) error -71
[  210.210197] usb 1-1.1: clear tt 2 (9093) error -71
[  210.210810] usb 1-1.1: clear tt 2 (9093) error -71
[  210.211483] usb 1-1.1: clear tt 2 (9093) error -71
[  210.212112] usb 1-1.1: clear tt 2 (9093) error -71
[  210.212725] usb 1-1.1: clear tt 2 (9093) error -71
[  210.213312] usb 1-1.1: clear tt 2 (9093) error -71
[  210.213968] usb 1-1.1: clear tt 2 (9093) error -71
[  210.214588] usb 1-1.1: clear tt 2 (9093) error -71
[  210.215217] usb 1-1.1: clear tt 2 (9093) error -71
[  210.300465] usb 1-1.1: USB disconnect, device number 3
[  210.300478] usb 1-1.1.1: USB disconnect, device number 5
[  210.300485] usb 1-1.1.1.4: USB disconnect, device number 7
[  210.301883] uvcvideo: Failed to resubmit video URB (-19).
[  210.306005] uvcvideo: Failed to resubmit video URB (-19).
[  210.309989] uvcvideo: Failed to resubmit video URB (-19).
[  210.313980] uvcvideo: Failed to resubmit video URB (-19).
[  210.317982] uvcvideo: Failed to resubmit video URB (-19).
[  210.341010] xhci_hcd 0000:01:00.0: ERROR Transfer event for disabled endpoint slot 10 ep 2
[  210.341023] xhci_hcd 0000:01:00.0: @0000000440003b20 00000000 00000000 0f000000 0a038000
[  210.342817] usb 1-1.1.2: USB disconnect, device number 6
[  210.343206] cdc_acm 1-1.1.2:1.0: failed to set dtr/rts
[  210.583586] usb 1-1.1: new high-speed USB device number 8 using xhci_hcd
[  210.671789] usb 1-1.1: device descriptor read/64, error -71
[  210.859754] usb 1-1.1: device descriptor read/64, error -71
[  211.047567] usb 1-1.1: new high-speed USB device number 9 using xhci_hcd
[  211.127781] usb 1-1.1: device descriptor read/64, error -71
[  211.315754] usb 1-1.1: device descriptor read/64, error -71
[  211.423902] usb 1-1-port1: attempt power cycle
[  212.027591] usb 1-1.1: new high-speed USB device number 10 using xhci_hcd
[  212.027805] usb 1-1.1: Device not responding to setup address.
[  212.235757] usb 1-1.1: Device not responding to setup address.
[  212.443594] usb 1-1.1: device not accepting address 10, error -71
[  212.527613] usb 1-1.1: new high-speed USB device number 11 using xhci_hcd
[  212.527875] usb 1-1.1: Device not responding to setup address.
[  212.735800] usb 1-1.1: Device not responding to setup address.
[  212.947606] usb 1-1.1: device not accepting address 11, error -71
[  212.947930] usb 1-1-port1: unable to enumerate USB device

**Additional context**
Add any other relevant context for the problem.

torndar, 2024-04-24:
Did you ever figure this out?

appleimperio, 2024-04-24:
No, I just stop using the hub