raspberrypi/linux #4446: HDMI out is broken when /boot/config.txt contains hdmi_enable_4kp60=1

graysky2, 2021-07-09:
**Describe the bug**
Booting with `hdmi_enable_4kp60=1` in `/boot/config.txt` results in no video output.  Removing this line restores HDMI output.

**To reproduce**
Use the following for `/boot/config.txt` and boot.
```dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
disable_overscan=1
disable_fw_kms_setup=1
hdmi_enable_4kp60=1
```

**Expected behaviour**
The machine should boot displaying video output (boot messages, console login etc.)

**Actual behaviour**
The machine boots but there is not HDMI output available.  The monitor does not detect any signal.

**System**
> Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

This is ArchARM which does not ship with `raspinfo`

* Which model of Raspberry Pi? RPi4B
* Which firmware version (`vcgencmd version`): https://github.com/raspberrypi/userland/commit/97bc8180ad682b004ea224d1db7b8e108eda4397
* Which kernel version (`uname -a`): `Linux workbench 5.10.48-1-ARCH #1 SMP PREEMPT Fri Jul 9 14:19:24 EDT 2021 aarch64 GNU/Linux`

**Logs**
I see the following in `dmesg`:
```
[  +0.007207] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[  +0.001395] Registered IR keymap rc-cec
[  +0.000085] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[  +0.000074] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input1
[  +0.000423] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
[  +0.008635] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[  +0.000026] cfg80211: failed to load regulatory.db
[  +0.018276] brcmfmac: F1 signature read @0x18000000=0x15264345
[  +0.004858] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  +0.010943] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  +0.000802] usbcore: registered new interface driver brcmfmac
[  +0.006627] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[  +0.001633] Registered IR keymap rc-cec
[  +0.000092] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[  +0.000072] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input2
[  +0.000326] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
[  +0.045470] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[  +0.000529] Registered IR keymap rc-cec
[  +0.000110] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[  +0.000067] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input3
[  +0.001343] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[  +0.000779] Registered IR keymap rc-cec
[  +0.000086] rc rc1: vc4 as /devices/platform/soc/fef05700.hdmi/rc/rc1
[  +0.000067] input: vc4 as /devices/platform/soc/fef05700.hdmi/rc/rc1/input4
[  +0.001551] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[  +0.000129] vc4-drm gpu: bound fe004000.txp (ops vc4_txp_ops [vc4])
[  +0.000075] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[  +0.000061] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[  +0.000055] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[  +0.000045] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[  +0.000058] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[  +0.001265] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[  +0.151747] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  +0.005376] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Sep 18 2020 02:27:58 version 7.45.221 (3a6d3a0 CY) FWID 01-bbd9282b
[  +0.422594] random: crng init done
[  +0.000003] random: 7 urandom warning(s) missed due to ratelimiting
[  +3.192255] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  +0.000037] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  +6.463800] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  +0.000104] Console: switching to colour frame buffer device 480x135
[ +10.239902] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ +10.239987] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[ +10.239998] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:70:plane-3] flip_done timed out
[Jul 9 16:44] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  +0.050926] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[ +10.445038] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ +10.240067] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[ +10.240009] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:70:plane-3] flip_done timed out
[ +10.239990] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
```

Complete dmesg.log from boot.
[dmesg.log](https://github.com/raspberrypi/linux/files/6793872/dmesg.log)

**Additional context**
As stated above, if I comment or remove `hdmi_enable_4kp60=1` the RPi boots with video as expected.

popcornmix, 2021-07-10:
Are you using `dtoverlay=vc4-kms-v3d` ?
Does the monitor prefer 4096x2160@60?
I think that still has a clocking issue compared to 3840x2160@60.
If so try adding `core_freq=600`.

graysky2, 2021-07-10:
> Are you using dtoverlay=vc4-kms-v3d ?

Yes: 
```
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
disable_overscan=1
disable_fw_kms_setup=1
hdmi_enable_4kp60=1
```

> Does the monitor prefer 4096x2160@60?

I don't think so, according to the manufacture's website, "Optimum resolution: 3840 x 2160 @ 60 Hz"

> If so try adding core_freq=600.

I added that line and rebooted but still no video output.

popcornmix, 2021-07-10:
You say it doesn't work with 5.10.48. Was it working with a previous version?

graysky2, 2021-07-10:
I don't know when it broke as I only recently tried using it.  I did step back 5 kernels one-at-a-time going back as far as 5.10.38 and none of them boot with functional video when I have that settings enabled.

graysky2, 2021-07-18:
I verified this bug on another monitor (different brand) and it is consistent.

popcornmix, 2021-07-19:
Is it okay with fkms driver? What hdmi mode do you end up with?

graysky2, 2021-07-19:
No, `dtoverlay=vc4-fkms-v3d` together with `hdmi_enable_4kp60=1` does not work either.  If I remove `hdmi_enable_4kp60=1` it does.

How can I determine which HDMI mode I am in?

graysky2, 2021-08-04:
Just to rule out something in the Arch ARM kernel config, I built the latest commit from rpi-5.10.y using `make bcm2711_defconfig` and found the same bug is triggered.  I think that rules out any kernel config diffs between the two.

popcornmix, 2021-08-04:
> How can I determine which HDMI mode I am in?

```
sudo cat /sys/kernel/debug/dri/0/state
```
(if that doesn't exist then use /sys/kernel/debug/dri/1/state)

graysky2, 2021-08-04:
When I boot with `hdmi_enable_4kp60=1` (when the bug is manifested), the output is below.

```
# cat /sys/kernel/debug/dri/1/state
plane[49]: plane-0
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[56]: plane-1
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[63]: plane-2
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[70]: plane-3
	crtc=crtc-3
	fb=224
		allocated by = [fbcon]
		refcount=2
		format=RG16 little-endian (0x36314752)
		modifier=0x0
		size=3840x2160
		layers:
			size[0]=3840x2160
			pitch[0]=7680
			offset[0]=0
			obj[0]:
				name=0
				refcount=3
				start=00100000
				size=16588800
				imported=no
	crtc-pos=3840x2160+0+0
	src-pos=3840.000000x2160.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[77]: plane-4
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[84]: plane-5
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
[... 264 lines cut ...]
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
crtc[90]: crtc-5
	enable=0
	active=0
	self_refresh_active=0
	planes_changed=0
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=0
	connector_mask=0
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
connector[32]: HDMI-A-1
	crtc=crtc-3
	self_refresh_aware=0
connector[40]: HDMI-A-2
	crtc=(null)
	self_refresh_aware=0
connector[48]: Writeback-1
	crtc=(null)
	self_refresh_aware=0
```

popcornmix, 2021-08-04:
You need a good hdmi cable to support 4kp60. That would be my first guess.
Otherwise can you post your edid? e.g. output of:
```
base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
```

graysky2, 2021-08-04:
The cable is a [CanaKit Raspberry Pi 4 Micro HDMI](https://www.amazon.com/gp/product/B07TTKD38N/).  According the the page, it supports 4k@60Hz.

* When I boot with `hdmi_enable_4kp60=1`, I have a 0-byte file for `/sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid`
* When I boot with that commented out, here is the output:
```
AP///////wBBDI/BFAYAABkdAQOAPCJ4KmehpVVNoicOUFS/7wDRwLMAlQCBgIFAgcABAQEBTdAA
oPBwPoAwIDUAVVAhAAAao2YAoPBwH4AwIDUAVVAhAAAaAAAA/ABQSEwgMjc2RThWCiAgAAAA/QAX
UB6gPAAKICAgICAgAQsCAzPxTJAEAx8TARJdXl9gYSMJBweDAQAAbQMMABAAOHggAGABAgNn2F3E
AXiAA+MPAAxWXgCgoKApUDAgNQBVUCEAAB4COoAYcTgtQFgsRQBVUCEAAB4BHQByUdAeIG4oVQBV
UCEAAB5NbICgcHA+gDAgOgBVUCEAABoAAAAATg==
```

popcornmix, 2021-08-04:
The 3 one-star reviews there don't fill me with confidence about the cable.
I also didn't see any positive reviews mentioning it working at 4kp60.

graysky2, 2021-08-04:
Agreed.  Do you have a link to a cable you know to work @ 4k@60fps?

popcornmix, 2021-08-04:
https://www.raspberrypi.org/products/micro-hdmi-to-standard-hdmi-a-cable/ would be a known good one.

graysky2, 2021-08-04:
> https://www.amazon.com/CanaKit-Raspberry-Micro-HDMI-Cable/dp/B07TTKD38N

Interesting.  The "Buy Now" button links me to the CanaKit website as an option.  It would seem to be the same as the 2-pack I already bought above.  I emailed CanaKit to confirm.

popcornmix, 2021-08-04:
What country did you select?

graysky2, 2021-08-04:
United States

popcornmix, 2021-08-04:
I ended up [here](https://www.canakit.com/official-micro-hdmi-to-hdmi-cable.html?defpid=4651) which is an official cable (has raspberry logo embossed on the micro-hdmi end of cable.

graysky2, 2021-08-04:
Yes, I found [this](https://www.amazon.com/Waveshare-Official-Raspberry-Standard-Designed/dp/B07YG1C73N) on amazon which, by pic and description is the official one.

popcornmix, 2021-08-04:
That looks like official one.

graysky2, 2021-08-04:
> Otherwise can you post your edid? e.g. output of

Did that edid look right?  I ordered from a 1 meter official cable from Microcenter.

popcornmix, 2021-08-04:
The edid looked fine. It says it supports 4kp60 in 4:4:4, 4:2:2 and 4:2:0 formats.
There are some displays that only support 4:2:0 which isn't supported by Pi4.

graysky2, 2021-08-04:
Thanks for the info.  I'll wait for the replacement cable to arrive and try that.  CanaKit did reply to my email query stating that the 2-pack I purchased does support 4k@60Hz, yet neither of the cables work for me.  If the official one I ordered doesn't work, could there be any hardware damage on the board itself that would explain this?  I assume we ruled out configuration on kernel.

graysky2, 2021-08-06:
The official cable does not work either.  I also tried the official cable in another RPi4 with the same result.  This has to be a software bug, no?

popcornmix, 2021-08-06:
If were a software issue I'm surprised it's not more widespread (4kp60 from fkms driver has been supported for 2 years).
And you say it affects kms and fkms drivers (written by different people).

What's the make and model of TV?
Do you have another 4k TV you can test with?

graysky2, 2021-08-06:
* Correct, present with both kms and fkms.
* I am not using a TV, just a regular monitor, confirmed on two different ones: [Philips 276E](https://www.usa.philips.com/c-p/276E8VJSB_27/4k-ultra-hd-lcd-monitor) and [LG 27UK650](https://www.lg.com/us/monitors/lg-27UK650-W-4k-uhd-led-monitor).

graysky2, 2021-08-06:
Bug confirmed on RPiOS armhf.
```
$ uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
```

popcornmix, 2021-08-06:
Can you try adding `core_freq=600`?
Can you try adding `force_turbo=1`?

graysky2, 2021-08-06:
Working on the RPiOS image:
Adding both options resulted in a correction in the bug.  It behaves as expected.  [Complete dmesg](https://gist.github.com/graysky2/744a3e53b21fd6223eff3a8aadd9b1d0) output here.

graysky2, 2021-08-06:
Applying them both to the Arch ARM image also corrects the bug there as well.

Interestingly, if I remove `force_turbo=1` leaving only `core_freq=600` the bug returns but if remove the `core_freq=600` and leave only `force_turbo=1`, that alone is enough to reverse the bug.

popcornmix, 2021-08-09:
If I add your config.txt settings all is fine. If I switch to your edid, I get the flip down. `force_turbo=1` resolves it.

These are the timings I get with my edid:
```
  #0 3840x2160 60.00 3840 4016 4104 4400 2160 2168 2178 2250 594000 flags: phsync, nvsync; type: preferred, userdef, driver
```
and these with yours:
```
  #0 3840x2160 60.00 3840 3888 3920 4000 2160 2163 2168 2222 533250 flags: phsync, nvsync; type: preferred, userdef, driver
```
The numbers after resolution are:
```vrefresh hdisplay hsync_start hsync_end htotal vdisplay vsync_start vsync_end vtotal clock```

so your display has reduced blanking and a lower pixel clock. 

Out of interest does using my edid give a working 4kp60 mode for you? 

```AP///////wAQrOmgTFRZMAQdAQOAPSN47u6Vo1RMmSYPUFSlSwBxT4GAqcCpQNHA4QABAQEBCOgA
MPJwWoCwWIoAYV0hAAAaAAAA/wBGTjg0SzkxTDBZVEwKAAAA/ABERUxMIFUyNzE4UQogAAAA/QAx
VgqJPAAKICAgICAgAWUCAz7xWGFgX15dEAUEAgcWARQfEhMnICEiAwYRFSMJHweDAQAAbQMMABAA
MHggAGADAgFn2F3EAXjAA+IPA1ZeAKCgoClQMCA1AGFdIQAAGgR0ADDycFqAsFiKAGFdIQAAHr8W
AKCAOBNAMCA6AGFdIQAAGgAAAAAAAAAAAAAALg==
```

decode that with `base64 -d` and place in `/lib/firmware/edid.dat` and add `drm.edid_firmware=edid.dat` to cmdline.txt.

graysky2, 2021-08-09:
Glad you are able to reproduce.  When I use your edid as instructed, I get errors around the file itself:
```
# base64 -d popcorn > /lib/firmware/edid.dat
# cat /boot/cmdline.txt 
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat
```
Rebooting:
```
% dmesg
...
[  +0.002296] [drm:edid_load [drm]] *ERROR* Size of EDID firmware "edid.dat" is invalid (expected 256, got 199
[  +0.025683] [drm:edid_load [drm]] *ERROR* Size of EDID firmware "edid.dat" is invalid (expected 256, got 199
[  +0.105269] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  +0.000059] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  +0.004848] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
```

popcornmix, 2021-08-09:
Ah I had ``` on same line as first line of edid, and github doesn't show it. I've inserted a newline and it now seems to show the whole edid.

graysky2, 2021-08-09:
OK, using your edid as instructed fixes the issue.  For reference:
```
% cat /boot/config.txt
over_voltage=6
arm_freq=2000

dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1

hdmi_enable_4kp60=1
#force_turbo=1
```

And
```
% cat /boot/cmdline.txt 
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat
```

graysky2, 2021-08-09:
So the obvious question I have is why is my edid different?  Property of the cheap monitor?

graysky2, 2021-08-09:
So the obvious question I have is why are the values for the blanking and pixel clock reported as lower from my monitor?  Clearly, I can use your edid and the display works with the higher values.  Bug from the manufacture???

popcornmix, 2021-08-09:
There are standard (CEA) timings for common modes (including 4kp60) which most monitors follow.
Had your monitor advertised that timing, then you would have avoiding this issue.

The fact that `force_turbo` helps means we can support this monitor with a future software update.
I suspect this timing causing some change in requirements for the core clock, but it will take some digging to determine exactly what is happening, and what the fix should be.

graysky2, 2021-08-09:
OK, thanks for the time you spend on this!  Please let me know if I can help moving forward.

graysky2, 2022-12-10:
@popcornmix - On the rpi-5.15.y branch of kernel (currently 5.15.82), keeping these two lines in `/boot/config.txt` works as expected:
```
hdmi_enable_4kp60=1
force_turbo=1
```

BUT, if I use the rpi-6.0.y branch, the monitor goes blank when I boot the box/does not detect a signal.  It only detects a signal if I load kodi on the box.  I don't see any errors or warnings in dmesg.  Thought to troubleshoot?

popcornmix, 2022-12-11:
> @popcornmix - On the rpi-5.15.y branch of kernel (currently 5.15.82), keeping these two lines in /boot/config.txt works as expected:

are you seeing both are required? What happens without force_turbo?

> BUT, if I use the rpi-6.0.y branch, the monitor goes blank when I boot the box/does not detect a signal. It only detects a signal if I load kodi on the box. I don't see any errors or warnings in dmesg. Thought to troubleshoot?

Output of kmsprint (from package kms++-utils) after boot (with blank screen) and after kodi has been run would be useful.
Also for comparison post output on 5.15 kernel.

Your edid may also be useful.

graysky2, 2022-12-11:
Under 5.15.82:
* Both lines commented boots fine and limits Matrix to 2160p @ 30 Hz.
* Uncommented only `hdmi_enable_4kp60=1` boots but renders the display dead.  Output of `dmesg` is filled with errors shown below.
* Uncommenting both lines is required to see the boot output and to use Matrix at 2160p @ 60 Hz.

```
[  +6.623834] vc4-drm gpu: [drm] *ERROR* [CRTC:94:crtc-3] flip_done timed out
[  +0.036928] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[ +10.714951] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  +0.000013] vc4-drm gpu: [drm] *ERROR* [CRTC:94:crtc-3] commit wait timed out
[Dec11 14:56] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  +0.000013] vc4-drm gpu: [drm] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
[ +10.208108] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  +0.000021] vc4-drm gpu: [drm] *ERROR* [PLANE:84:plane-3] commit wait timed out
[ +10.239977] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  +0.000021] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[ +10.239909] vc4-drm gpu: [drm] *ERROR* [CRTC:94:crtc-3] flip_done timed out
[ +14.080253] vc4-drm gpu: [drm] *ERROR* flip_done timed out
```

Interestingly, if I boot to 6.0.12 with both lines uncommented, the monitor is not detected as I mentioned, but if I ssh into the box, then start [kodi.service](https://github.com/graysky2/kodi-standalone-service/blob/master/arm/init/kodi.service), the display magically gets detected and Matrix has 3840x2160p @ 60 Hz available.

Arch ARM doesn't package `kmsprint` it seems. 

Here is the monitor's edid:
```
% base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
AP///////wBBDI/BFAYAABkdAQOAPCJ4KmehpVVNoicOUFS/7wDRwLMAlQCBgIFAgcABAQEBTdAA
oPBwPoAwIDUAVVAhAAAao2YAoPBwH4AwIDUAVVAhAAAaAAAA/ABQSEwgMjc2RThWCiAgAAAA/QAX
UB6gPAAKICAgICAgAQsCAzPxTJAEAx8TARJdXl9gYSMJBweDAQAAbQMMABAAOHggAGABAgNn2F3E
AXiAA+MPAAxWXgCgoKApUDAgNQBVUCEAAB4COoAYcTgtQFgsRQBVUCEAAB4BHQByUdAeIG4oVQBV
UCEAAB5NbICgcHA+gDAgOgBVUCEAABoAAAAATg==
```

popcornmix, 2022-12-11:
> Arch ARM doesn't package kmsprint it seems.

It's pretty simple to build: https://github.com/tomba/kmsxx

graysky2, 2022-12-11:
I build a PKGBUILD for kmsxx: https://aur.archlinux.org/packages/kmsxx-git

Under 5.15.82:
Condition = screen detected/console login screen
```
% kmsprint
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 533.250 3840/48/32/80/+ 2160/3/5/54/- 60 (60.00) P|D 
      Plane 3 (84) fb-id: 338 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 338 3840x2160 RG16
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```
Condition = after started kodi.service:
```
% kmsprint                                       
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 594.000 3840/176/88/296/+ 2160/8/10/72/+ 60 (60.00) D 
      Plane 6 (117) fb-id: 339 (crtcs: 0 1 2 3 4 5) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 339 3840x2160 XR24
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```

Under 6.0.12:
Condition = screen not detected/should be at console login screen
```
% kmsprint
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 533.250 3840/48/32/80/+ 2160/3/5/54/- 60 (60.00) P|D 
      Plane 3 (84) fb-id: 338 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 338 3840x2160 RG16
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```
Condition = after started kodi.service (screen magically came on):
```
% kmsprint
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 594.000 3840/176/88/296/+ 2160/8/10/72/+ 60 (60.00) D 
      Plane 6 (117) fb-id: 340 (crtcs: 0 1 2 3 4 5) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 340 3840x2160 XR24
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```

graysky2, 2022-12-11:
It get more complex... the bug seems inconsistent.  For example, I just rebooted that box with the 6.0.12 kernel and the two lines uncommented.  The screen came on just fine (ie the expected behavior).
```
% kmsprint
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 533.250 3840/48/32/80/+ 2160/3/5/54/- 60 (60.00) P|D 
      Plane 3 (84) fb-id: 338 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 338 3840x2160 RG16
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```

popcornmix, 2022-12-12:
I'm trying to reproduce some form of failure. I'm using your edid. I'm on  5.15.82 (also tried 6.1.0 without failure).
I have `hdmi_enable_4kp60=1`  and do not have `force_turbo=1`

I was hoping that would replicate your `*ERROR* flip_done timed out`, but so far on a few boots, display has been working and dmesg clean.

Can you replicate failure without your display attached - e.g. using `drm.edid_firmware=HDMI-A-1:edid/graysky.edid video=:D`

popcornmix, 2022-12-12:
Ah - I have seen it fail now. It's inconsistent for me.

I may be spotting false patterns, but so far for me it has (with `force_turbo` disabled, and `hdmi_enable_4kp60=1`) and cpufreq governor set to:
performance: always worked
ondemand: sometimes worked, sometimes failed
powersave: always worked

Now performance working is not too surprising (it has a similar effect to a later starting force_turbo).
powersave working is surprising and may suggest a change in clock frequency rather than absolute clock frequency is significant.

I'll leave a reboot loop running - RNG may just be playing with me.

popcornmix, 2022-12-12:
Could you try powersave and performance (without force_turbo)?
Unfortunately ondemand in a reboot loop hasn't failed since the first time.

I'm reboot looping with (in /etc/rc.local):
```sleep 10 && if ! dmesg | grep flip_done; then reboot; fi``

graysky2, 2022-12-12:
> Could you try powersave and performance (without force_turbo)?

Which kernel version did you want me to do these?

popcornmix, 2022-12-12:
> Under which kernel version did you want me to do these? 6.0.12 or 5.15.85?

Any which normally fail. 5.15 would be fine.

graysky2, 2022-12-12:
OK.  I built 5.15.82 with performance as the default gov, and with just `hdmi_enable_4kp60=1`, I am able to boot to the console without the bug.  Tried 3x and worked 3/3 times.

Did the same with powersave.  Wouldn't boot with the monitor detected.  Interestingly, I couldn't ssh into the box either.  Had to comment out the `hdmi_enable 4kp60=1` line.

popcornmix, 2022-12-14:
So your edid reports:
```
   0 3840x2160@60.00  533.250 3840/48/32/80/+    2160/3/5/54/-      60 (60.00) P|D   
   1 3840x2160@60.00  594.000 3840/176/88/296/+  2160/8/10/72/+     60 (60.00) D     16:9
```
It's the same mode, but the former has a lower pixel clock (i.e. reduced blanking), and is preferred, so is chosen by default.

Now, we have a fairly complex calculation that uses pixel clock and blanking periods to calculate what core clock is required. The formula was developed experimentally based on modes we've seen and this one it gets wrong.

It comes down to (treating clocks as MHz):
`max(pixel_clock * hdisplay / htotal + 1, pixel_clock * 9 / 10)` which gives:
`max(533.25 * 3840 / 4000 + 1, 533.25 * 9 / 10) = max(512.9,  480) = 513MHz` for bad mode
`max(594.00 * 3840 / 4400 + 1, 594.00 * 9 / 10), = max(519.4, 534.6) = 534MHz` for good mode

and 513MHz is not enough core clock for this mode.
Experimentally it looks like 518MHz is the lowest that works for this mode.

@timg236 you helped derive this formula. Any thoughts on how to adjust it?

I'm thinking we want to increase a mode that has minimal horizontal blanking, but leave the larger blanking mode (which falls into the `9/10` part of max) alone.
An example would be:
`max(pixel_clock * hdisplay / htotal + 8, pixel_clock * 9 / 10)` which gives:
`max(533.25 * 3840 / 4000 + 8, 533.25 * 9 / 10) = max(519.9,  480) = 520MHz` for bad mode
`max(594.00 * 3840 / 4400 + 8, 594.00 * 9 / 10), = max(526.4, 534.6) = 534MHz` for good mode

So no change for the standard 4kp60 mode, but it boosts the reduced blanking one into the safe area.
It makes this edid work.

timg236, 2022-12-15:
@popcornmix I think what's happening is that the original formula is an estimate based upon the increased FIFO in PV2 but doesn't take into account various small (and not well understood) constant setup costs at the start of each line.
If there is a larger blanking period then these are lost, but it seems like we need to include some sort of minimum setup cost which would increase the clock rate with reduced blanking but not with the larger blanking periods.

popcornmix, 2022-12-15:
A potential fix has been merged into 5.15 tree (and picked to 6.0 and 6.1). Can you test?
Also in rpi-update kernel.

graysky2, 2022-12-15:
@popcornmix - Based on my experience above, a reliably problematic condition was tested with your new commit:
> Did the same with powersave. Wouldn't boot with the monitor detected. Interestingly, I couldn't ssh into the box either. Had to comment out the hdmi_enable 4kp60=1 line.

I setup the test conditions:
* latest commit from rpi-5.15.y
* built with `CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE=y`
* only `hdmi_enable_4kp60=1` without adding `force_turbo=1`

Result: booted with detected display 5/5 times.  For reference, `kmsprint` output below.

```
% kmsprint
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 533.250 3840/48/32/80/+ 2160/3/5/54/- 60 (60.00) P|D 
      Plane 3 (84) fb-id: 338 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 338 3840x2160 RG16
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
```

I think it's safe to declare victory.

closed 2023-01-24 by popcornmix