raspberrypi/linux #4454: Kernel Option CONFIG_SND_SOC_WM8960 Not Set

hftsai256, 2021-07-14:
In [PR#4039](https://github.com/raspberrypi/linux/pull/4039), `wm8960-soundcard` overlay was added to the tree. However the correspond kernel module option was not enabled by that time. Therefore, when I'm building kernel for raspberry pi zero, I have:
```
armbuilder:~/wm8960/linux/rpi-5.4.y$make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcmrpi_defconfig
#
# configuration written to .config
#
armbuilder:~/wm8960/linux/rpi-5.4.y$ grep CONFIG_SND_SOC_WM8960 .config
# CONFIG_SND_SOC_WM8960 is not set
```
And therefore the required kernel module is missing. Are we expecting the option to be `CONFIG_SND_SOC_WM8960=m` in `bcm{2709|2711|2835|rpi}_defconfig`?

pelwell, 2021-07-14:
@shawaj How have you been using the overlay, since the codec driver is not enabled in our defconfigs?

pelwell, 2021-07-16:
The required config setting is now in our defconfigs, and the module will be in future kernel builds.

hftsai256, 2021-07-18:
Thanks @pelwell. May I ask a favor to merge this fix to kernel version 5.4.y? Would this bump the kernel version from 5.4.83+ to a larger y value?

pelwell, 2021-07-18:
It's in the 5.4 branch - see https://github.com/raspberrypi/linux/commit/ec0dcf3064b8ba99f226438214407fcea9870f76. The version is still 5.4.83+, but it will get bumped when the next batch of upstream patches are applied.

closed 2021-07-21 by pelwell