raspberrypi/linux #4455: RPI-LINUX with PREEMT_RT , please make more pair able versions of RPI-LINUX

effect0, 2021-07-15:
hi people, 

i want you to please make more pair able versions of RPI-LINUX with PREEMT_RT, ideally ALL versions.

----> my problem:
i build PREEMT_RT based on
https://wiki.linuxfoundation.org/realtime/start
and
https://github.com/raspberrypi/linux/commits    ----> search repository: Merge remote-tracking branch 'stable/linux-5.10.y' into rpi-5.10.y

examples feb until jul 2021 table:
PREEMT_RT & RPI-LINUX merge with same version
5.10.47 & not exist
5.10.41 & not exist
5.10.35 & exist, and works
5.10.30 & not exist
5.10.27 & exist, and works
5.10.25 & exist, and works
5.10.21 & not exist
....... & ...
there where only some existing compatible in last months
so have to wait until there is a raspberry pi linux with same version than preemt rt, sometimes this takes long time

---->  possible problem solution:

bring more RPI-LINUX versions that can be used with PREEMT_RT

example:
check at PREEMT_RT developer the version upcomming will be released, developer for 5.10-rt is Steven Rostedt 
if preemt_RT for linux 5.10.comming will be released,
then  "merge remote tracking branch for 5.10.comming"

if there is an other !easier! solution dont see then tell please

lategoodbye, 2021-07-15:
What prevents you from using 5.10-rt or 5.12-rt directly?

kdoren, 2021-07-29:
@effect0 I posted a release of 5.10.52-rt47 for RPi here: https://github.com/kdoren/linux/releases/tag/rpi_5.10.52-rt47

@lategoodbye building PREEMPT_RT kernel for RPi is most easily done by applying rt patch set to RPi kernel, but releases should match, and not every 5.10.y release is available on both; every few months the stars line up (i.e. 5.10.52).  You can build with slightly different versions, but then it's hard to be confident that there is no mismatch.  Additional RPi-specific rt patches are required for RPi3 operation.

For my use case (jack audio applications using USB interfaces), I have generally gotten better results with less pain by using low-latency PREEMPT kernel.  I believe this is because RPi uses FIQ, and PREEMPT_RT does not play nice with FIQ (it seems to add USB overhead).  5.10.52-rt47 looks otherwise pretty good, they must have fixed some bugs - it doesn't hang when running jamulus server in multithreaded mode like earlier 5.10 rt versions.

lategoodbye, 2021-07-29:
@kdoren So in your case mainline kernel with rt patches doesn't work for you, because dwc2 USB driver (which doesn't rely on FIQ) doesn't provide the proper performance?

kdoren, 2021-07-29:
@lategoodbye I actually never tried just the mainline (non raspberrypi/linux repo) rt kernel on RPi, I have been assuming (perhaps incorrectly) that RPi required the RPi-specific commits in the raspberrypi/linux repo.  If you think that should work I could try it.

kdoren, 2021-08-02:
PREEMPT_RT kernel 5.10.52-rt47 for RPi is looking pretty good so far,
Latency plots are here:  https://github.com/kdoren/linux/discussions/11
Deb package build is here:  https://github.com/kdoren/linux/releases/tag/rpi_5.10.52-rt47

@lategoodbye I did try building a mainline rt kernel but I could not get it to boot on RPi4.  I didn't put much effort into figuring out why.  It seems to be missing RPi overlay drivers that exist in the raspberrypi/linux repo.

lategoodbye, 2021-08-03:
@kdoren I'm confused, at first you talked about RPi 3 and now about RPi 4. The latter is still missing features in mainline.

What kernel configuration did you tried (rt32.config for both platforms)?

kdoren, 2021-08-03:
@lategoodbye I primarily run/test on RPi4 but I also test on RPi3 because there are a lot of them out there.
The "mainline" linux-stable-rt kernel doesn't appear to contain i.e. some HAT card drivers such as hifiberry. 
I built a PREEMPT_RT kernel from mainline using bcm2835_defconfig (2711 doesn't exist) but it wouldn't boot on RPi4 or RPi3.
The resulting deb package didn't contain any boot/overlays/ files at all (*.dtbo).  
I expect I am doing something wrong.  But it probably doesn't matter, 5.10.52-rt47 made by applying rt patch set to raspberrypi/linux, plus the rpi fiq patches, seems to be working fairly well so far.

lategoodbye, 2021-08-03:
@kdoren Yes, it's a shame that most of the audio HAT driver only available in the Raspberry Pi tree. Unfortunately there is no full overlay support in mainline and all mentioned overlays apply only to Raspberry Pi.

In case you want to build a 32 bit kernel for RPi 3, you will need to choose multi_v7_defconfig and enable LPAE manually for RPi 4 support ( see https://gist.github.com/lategoodbye/c7317a42bf7f9c07f5a91baed8c68f75 ). bcm2835_defconfig is intended for RPi 1 & Zero.

kdoren, 2021-08-04:
@lategoodbye Thanks for the useful info

kdoren, 2021-12-18:
FYI I posted 32 & 64-bit deb packages of new kernel 5.15.10 built with PREEMPT_RT and low-latency PREEMPT.
Also similar packages for kernel 5.10.74.

https://github.com/kdoren/linux/releases