raspberrypi/linux #4458: [FAILED] Failed to start Command from Kernel Command Line

boner-cmd, 2021-07-16:
**Is this the right place for my bug report?**
This is definitely kernel-related, but I don't know if the Raspberry Pi Imager falls under this GitHub.

**Describe the bug**
Something wrong with Raspberry Pi Imager's custom settings not building or installing the kernel correctly.

**To reproduce**
Using Raspberry Pi Imager, install an Ubuntu image with custom settings using control+shift+x

**Expected behaviour**
Image boots normally

**Actual behaviour**
Image fails to boot due to missing kernel dependency, RPI hangs.

**System**

* Which model of Raspberry Pi? RPi 4, 4GB
* Which OS and version (`cat /etc/rpi-issue`)? Ubuntu 21.04 64 bit server, but likely others in Ubuntu 20
* Which firmware version (`vcgencmd version`)? unknown, image doesn't boot
* Which kernel version (`uname -a`)? unknown, image doesn't boot

**Logs**
`"[FAILED] Failed to start Command from Kernel Command Line.
See 'systemctl status kernel-command-line.service' for details
[DEPEND] Dependency failed for Command from Kernel Command Line."`

**Additional context**
https://bugs.launchpad.net/ubuntu/+source/u-boot/+bug/1914878

Specifically these comments:

- https://bugs.launchpad.net/ubuntu/+source/u-boot/+bug/1914878/comments/4
- https://bugs.launchpad.net/ubuntu/+source/u-boot/+bug/1914878/comments/8
- https://bugs.launchpad.net/ubuntu/+source/u-boot/+bug/1914878/comments/10

Please note that while it was an invalid "bug" for Ubuntu itself (the original report was actually for a different error line), the issue I described as identified by other users is most certainly a real and reproducible bug for Raspberry Pi users of the Imager.

BoySanic, 2021-08-11:
I don't have much more to add here, but, it's reproducible on my Pi 4 2GB.

boner-cmd, 2021-08-12:
I probably should have said it outright in my original post, but the workaround is to not use custom settings when using the imager. However, this doesn't change that the imager needs to be fixed so custom settings work again.

popcornmix, 2021-08-12:
This isn't the right place for this. rpi-imager is not part of the linux kernel.

BoySanic, 2021-08-12:
https://github.com/raspberrypi/rpi-imager/issues