raspberrypi/linux #4462: merge of v5.10.51 into rpi-5.10.y conflict

ncopa, 2021-07-19:
**Is this the right place for my bug report?**
I am not sure. I am trying share my finding while preparing the 5.10.51 kernel for alpine linux.

**Describe the bug**
Merge conflict in `drivers/gpu/drm/vc4/vc4_hdmi.c`

**To reproduce**
```
git checkout rpi-5.10.y
git branch -b wip
git merge v5.10.51
```

**Expected behaviour**
well.... conflicts are not really unexpected.

**Actual behaviour**
```
CONFLICT (content): Merge conflict in drivers/gpu/drm/vc4/vc4_hdmi.c
```
there was two hunks that had conflicts:
```c
static void vc4_hdmi_encoder_pre_crtc_configure(struct drm_encoder *encoder,
                                                struct drm_atomic_state *state)
{
        struct vc4_hdmi *vc4_hdmi = encoder_to_vc4_hdmi(encoder);
        struct drm_connector *connector = &vc4_hdmi->connector;
        struct drm_connector_state *conn_state =
                drm_atomic_get_new_connector_state(state, connector);
        struct vc4_hdmi_connector_state *vc4_conn_state =
                conn_state_to_vc4_hdmi_conn_state(conn_state);
        struct drm_display_mode *mode = &encoder->crtc->state->adjusted_mode;
        unsigned long bvb_rate, pixel_rate, hsm_rate;
        int ret;

        ret = pm_runtime_resume_and_get(&vc4_hdmi->pdev->dev);
<<<<<<< HEAD
        if (ret) {
=======
        if (ret < 0) {
>>>>>>> v5.10.51
                DRM_ERROR("Failed to retain power domain: %d\n", ret);
                pm_runtime_put(&vc4_hdmi->pdev->dev);
                return;
        }
```
and
```c
static int vc4_hdmi_bind(struct device *dev, struct device *master, void *data)
...
        /*
         * We need to have the device powered up at this point to call
         * our reset hook and for the CEC init.
         */
        ret = vc4_hdmi_runtime_resume(dev);
        if (ret)
                goto err_put_ddc;

        pm_runtime_get_noresume(dev);
        pm_runtime_set_active(dev);
        pm_runtime_enable(dev);

        if (vc4_hdmi->variant->reset)
                vc4_hdmi->variant->reset(vc4_hdmi);

        if ((of_device_is_compatible(dev->of_node, "brcm,bcm2711-hdmi0") ||
             of_device_is_compatible(dev->of_node, "brcm,bcm2711-hdmi1")) &&
            HDMI_READ(HDMI_VID_CTL) & VC4_HD_VID_CTL_ENABLE) {
                clk_prepare_enable(vc4_hdmi->pixel_clock);
                clk_prepare_enable(vc4_hdmi->hsm_clock);
                clk_prepare_enable(vc4_hdmi->pixel_bvb_clock);
        }
<<<<<<< HEAD
=======

        pm_runtime_enable(dev);
>>>>>>> v5.10.51

        drm_simple_encoder_init(drm, encoder, DRM_MODE_ENCODER_TMDS);
        drm_encoder_helper_add(encoder, &vc4_hdmi_encoder_helper_funcs);

```

**System**
n/a

**Logs**
n/a

**Additional context**

The following comits from  `rpi-5.10.y`:
- a2224f9777778965ff76049548ba483a23d7933e (drm/vc4: hdmi: Switch to pm_runtime_resume_and_get)
- 085260e9d38c047ad89bcaa5960dc3d98e6223e2 (drm/vc4: hdmi: Put the device on error in pre_crtc_configure)

conflicts with upstream `linux-5.10.y`:
- gregkh/linux@f06ea024c176fe44ec7f5746bcddcfab07e21758 (drm/vc4: hdmi: Fix PM reference leak in vc4_hdmi_encoder_pre_crtc_co())

----

And the `rpi-5.10.y` commit
- 94bc403a9a7af3bd042ecdc160e6ab0b08331df8 (drm/vc4: hdmi: Make sure the controller is powered up during bind)

conflicts with upstream `linux-5.10.y`
- gregkh/linux@98bd09d928b370bf5d72a920974db7310404839c (drm/vc4: hdmi: Prevent clock unbalance)

The way I chose to solve this was to follow upstream for the first conflict
```c
        if (ret < 0) {
```
and in the second I followed `HEAD`/`rpi-5.10.y` and removed the `pm_runtime_enable(dev);` as it was moved up in 94bc403a9a7af3bd042ecdc160e6ab0b08331df8. I don't know if this was correct resolution, since I am not familiar with the code.

Issue #4457 might also be related.

pelwell, 2021-07-19:
Thanks for the heads-up. Without looking at the code I would imagine that the correct fix is likely to be reverting one or more downstream patches before applying the equivalent (but not identical) upstream versions.

ncopa, 2021-07-19:
> Thanks for the heads-up. Without looking at the code I would imagine that the correct fix is likely to be reverting one or more downstream patches before applying the equivalent (but not identical) upstream versions.

The downstream a2224f9777778965ff76049548ba483a23d7933e should be safe to revert. The other two needs a closer look. It might be worth try revert 94bc403a9a7af3bd042ecdc160e6ab0b08331df8, merge in `v5.10.51` and see if upstream has fixed #4457.

pelwell, 2021-07-19:
Once mripard has a fix for #4457 we're likely to get it downstream first for testing before he submits it upstream.

popcornmix, 2021-07-22:
Reverting https://github.com/raspberrypi/linux/commit/a2224f9777778965ff76049548ba483a23d7933e and https://github.com/raspberrypi/linux/commit/94bc403a9a7af3bd042ecdc160e6ab0b08331df8, merging upstream and cherry-picking https://github.com/raspberrypi/linux/commit/94bc403a9a7af3bd042ecdc160e6ab0b08331df8 didn't result in any conflicts and
only resulted in trivial change:
```
dom@buildbot:~/projects/linux$ git diff origin/rpi-5.10.y drivers/gpu/drm/vc4/vc4_hdmi.c
diff --git a/drivers/gpu/drm/vc4/vc4_hdmi.c b/drivers/gpu/drm/vc4/vc4_hdmi.c
index 5401b6a6f440..81ea1fbbdaab 100644
--- a/drivers/gpu/drm/vc4/vc4_hdmi.c
+++ b/drivers/gpu/drm/vc4/vc4_hdmi.c
@@ -919,7 +919,7 @@ static void vc4_hdmi_encoder_pre_crtc_configure(struct drm_encoder *encoder,
        int ret;

        ret = pm_runtime_resume_and_get(&vc4_hdmi->pdev->dev);
-       if (ret) {
+       if (ret < 0) {
                DRM_ERROR("Failed to retain power domain: %d\n", ret);
                pm_runtime_put(&vc4_hdmi->pdev->dev);
                return;
```

ncopa, 2021-07-26:
closing this because its merged in into rpi-5.10.y already

closed 2021-07-26 by ncopa