raspberrypi/linux #4465: RPi4/kms: boot hangs if two monitors are connected

HiassofT, 2021-07-21:
With current rpi-update kernel/firmware https://github.com/Hexxeh/rpi-firmware/commit/dd59f0eaeae7a53be8d193c21c4cd83f7a667da7 boot hangs if vc4-kms-v3d is enabled and two monitors are connected (1920x1200 monitors here). With just one monitor connected boot completes successfully.

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.50-v7l+ #1438 SMP Mon Jul 19 10:46:09 BST 2021 armv7l GNU/Linux
root@raspberrypi:~# vcgencmd version
Jul 21 2021 16:21:46     
Copyright (c) 2012 Broadcom
version 6a796bb0062a6c75191c57cba1c13f9300076d02 (clean) (release) (start)
```
serial console log: http://ix.io/3tDb

I initially noticed that on LibreELEC running kernel e6410af

console log from LibreELEC: http://ix.io/3tDe

Bisecting through rpi-update firmwares the last working version is https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2, the first bad vesion is https://github.com/Hexxeh/rpi-firmware/commit/6d1196a54e7197c719299f9f896340bc02123d3b

ping @mripard

popcornmix, 2021-07-22:
The regression started with https://github.com/raspberrypi/linux/pull/4441

Specifically with the switch from downstream to upstream versions of:
https://github.com/raspberrypi/linux/pull/4441/commits/5948c83966438d815c5c2e6a300b2563226fc9f4
https://github.com/raspberrypi/linux/pull/4441/commits/04e385bc99447579c1461a04eacdda07af5145ca

if I revert these two commits (with minor conflicts) the issue is gone.
However the upstream and downstream forms of these commits are not trivially different, and it's tricky to re-apply the downstream, with the other intervening upstream commits.

We may need to revert the whole sequence in the PR (and even that now has conflicts with other commits).

mripard, 2021-07-23:
It's likely to be a kernel crash then. When Linux switches fbcon as soon as the DRM driver is registered, it takes the global console lock, and thus if you happen to have a crash during that first commit it will never output the usual stacktrace.

I'll have a look

popcornmix, 2021-07-23:
Also tested rpi-5.14.y kernel which had a cleaner separation of upstream and downstream commits.
Upstream is okay. The downstream commit where the issue starts is https://github.com/raspberrypi/linux/commit/41aa2712e178a7cb4297446ba5ccf099c8557138 (i.e. same place I was seeing on 5.10 kernel).
I believe the older form of this commit originally merged to 5.10 is okay, but the newer one submitted upstream is not.

mripard, 2021-07-23:
Yeah, upstream doesn't have that commit yet, which kind of confirms that this is the culprit. I've been trying to figure out what's going on so far, but haven't really found anything yet.

timg236, 2021-07-23:
@mripard Do you think this is the same root cause as the clk_request_done panic that we were discussing the other day?

mripard, 2021-07-23:
The PR #4470 should fix this issue.

It's actually something that was found using KASAN, but I never backported it thinking it would reach stable, and yet it never did.. Sorry for that.

@timg236 Yes, I believe it could be the same root cause. If you could give it a quick test it would be awesome

timg236, 2021-07-26:
Yes, I still see the same failure after launching Chromium a couple of times. The first time I also saw
[code]
[   18.075549]  r9:00000001 r8:30c5387d r7:30c5383d r6:ffffffff r5:20000010 r4:b6fd644c
[   18.075555] ---[ end trace 42e645bc91440d3c ]---
[   63.195149] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[   63.195255] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[   73.435041] vc4-drm gpu: [drm] *ERROR* Timed out waiting for flip_done
[/code]

and with a 4K display it looks like we are getting HVS underflows. I'll check whether something is trying to enable 4Kp60 without hdmi_enable_4kp60=1

timg236, 2021-07-26:
It's 4Kp30. FKMS has a steady core-clock frequency of 267218256 and I don't see underflows.  I couldn't tell from the code whether this is safe but if KMS is adjusting core-clock frequency depending on the scene content then it needs to always pick the maximum of the next scene and the on-display scene (for each display)

timg236, 2021-07-26:
The problem is over-scan. FKMS crops, KMS is scaling. Scaling a 4K layer is quite expensive so that needs to be taking into account in the core-clock complexity. Or perhaps better, crop the image instead of scaling.

timg236, 2021-07-26:
Not core-clock related. This corruption is https://github.com/raspberrypi/linux/issues/4475

6by9, 2021-07-27:
> The problem is over-scan. FKMS crops, KMS is scaling. 

Legacy crops. Both FKMS and KMS scale. Unfortunately that was the implementation that mainline chose for margin support, I suspect because they advertise all the timings to userspace so trimming those modes down to reflect the overscan is trickier, and you'd have to retain the untrimmed modes internally in order to configure the actual hardware.

popcornmix, 2021-07-27:
I believe this is resolved with latest rpi-update kernel.

HiassofT, 2021-07-28:
Testing on LibreELEC and RPiOS with two monitors connected was fine. Thanks!

closed 2021-07-28 by HiassofT