raspberrypi/linux #4466: RPI0w Kernel Panic Related to ipv6_rcv

ckelloug3, 2021-07-21:
**Is this the right place for my bug report?**
We're seeing a kernel panic.

**Describe the bug**
The kernel panics on RPI0w.

**To reproduce**
Run the RPI0w for between 5 minutes and a few days.

**Expected behaviour**
The RPI0w should run without crashing.

**Actual behaviour**
The system crashes and halts at the panic.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

```System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

Linux chassis-trak-058 5.10.17+ #1421 Thu May 27 13:58:02 BST 2021 armv6l GNU/Linux
Revision	: 9000c1
Serial		: 00000000b2ce10d5
Model		: Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 27 2021 14:04:13 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       14985136 1678928  12664256  12% /
devtmpfs          187224       0    187224   0% /dev
tmpfs             220204       0    220204   0% /dev/shm
tmpfs             220204    8592    211612   4% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             220204       0    220204   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48800    209295  19% /boot

Filename				Type		Size		Used	Priority
/var/swap                               file		102396		0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
  Installed: (none)
[... 1015 lines cut ...]
/0x454)
[ 3602.001916]  r10:00000000 r9:00000000 r8:00000000 r7:b6757dec r6:00000000 r5:
00000010
[ 3602.014856]  r4:c4875480
[ 3602.019879] [<c01f36b4>] (vfs_write) from [<c01f3cbc>] (ksys_write+0xc0/0xd8)
[ 3602.029660]  r10:00000000 r9:c4bbc000 r8:00000010 r7:b6757dec r6:c0ba5028 r5:
c4875480
[ 3602.042461]  r4:c4875481
[ 3602.047400] [<c01f3bfc>] (ksys_write) from [<c01f3cec>] (sys_write+0x18/0x1c)
[ 3602.057083]  r9:c4bbc000 r8:c00083e4 r7:00000004 r6:00000006 r5:b6757dec r4:0
0000010
[ 3602.069729] [<c01f3cd4>] (sys_write) from [<c0008240>] (ret_fast_syscall+0x0/
0x28)
[ 3602.082244] Exception stack(0xc4bbdfa8 to 0xc4bbdff0)
[ 3602.089858] dfa0:                   00000010 b6757dec 00000006 b6757dec 00000
010 00000000
[ 3602.102917] dfc0: 00000010 b6757dec 00000006 00000004 b67a8cfa b6757dec 00000
000 b6767f9c
[ 3602.116040] dfe0: 00000002 b6757da0 00000000 b6e861bc
[ 3602.123681] Code: bad PC value
[ 3602.129173] ---[ end trace 61524973814c0045 ]---
[ 3602.136250] Kernel panic - not syncing: Fatal exception in interrupt
[ 3602.145109] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
 ]---
```

pelwell, 2021-07-22:
> Run the RPI0w for between 5 minutes and a few days.

That's not a great distribution - more reproducible would be better.

1. Describe your network environment.
2. Are you intentionally using IPv6?
3. What workload are you running?

ckelloug3, 2021-07-22:
>That's not a great distribution - more reproducible would be better.

I've been searching for a better repro case but haven't found one yet.  I've seen the problem on 3 different pi zero w units so I am pretty sure it isn't hardware.

>1.  Describe your network environment.
I'm using the wlan0 interface on the pi using a Linksys WRT54GS 2.4 gHz wifi router/access point.  The access point isn't capable of ipv6.  I've tried this with another access point for which I can't get the model number right now and had the same problem.
>1. Are you intentionally using IPv6?
I was not intentionally using IPv6 however in thoroughly answering your question, I noticed that pigpiod creates IPV6 sockets by default and communicates with my code.  I've reset pigpiod to use ipv4 sockets in my test setup and will see if the oops and panic continue. 
>1. What workload are you running?
I'm running my own application that logs a2d converter values connected to i2c and occasionally toggles gpio pins with pigpiod.

pelwell, 2021-07-23:
I agree that it doesn't have the feel of a hardware fault. I'm curious to see if the pigpiod change has made a difference - if nothing else it might give a route to recreating the issue. Your application doesn't sound particularly intensive, which is also encouraging in terms of reproducibility.

ckelloug3, 2021-07-24:
I changed pigpiod not to use ipv6 and ran the RPI0w for a while and getting a different panic related to swapper.  Whatever's going on looks to me like it's not going to be easily debuggable.  I also tried updating the kernel to 5.10.52 with rpi-update.  The system has run for about a day without panicking where it usually crashed before that.
```
<--- cut here ---
[14258.970323] Unable to handle kernel NULL pointer dereference at virtual addre
ss 00000004
[14258.981767] pgd = 3933c2be
[14258.986248] [00000004] *pgd=00000000
[14258.991578] Internal error: Oops: 17 [#1] ARM
[14258.997653] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm b
luetooth ecdh_generic ecc libaes 8021q garp stp llc rtc_rv3028 regmap_i2c brcmfm
ac brcmutil sha256_generic libsha256 cfg80211 raspberrypi_hwmon rfkill i2c_bcm28
35 bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vch
iq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 vid
eobuf2_common videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tabl
es x_tables ipv6
[14259.054474] CPU: 0 PID: 0 Comm: swapper Tainted: G         C        5.10.17+
#1421
[14259.066347] Hardware name: BCM2835
[14259.072005] PC is at do_idle+0x28/0x108
[14259.078056] LR is at trace_hardirqs_on+0x4c/0x14c
[14259.085002] pc : [<c0053e9c>]    lr : [<c00f5f64>]    psr: 60000013
[14259.093575] sp : c0ba3f38  ip : 00000000  fp : fffffffc
[14259.101091] r10: c0b6fa50  r9 : c0cee000  r8 : 00000000
[14259.108622] r7 : c0b6fa50  r6 : c0ba5020  r5 : c0ba5028  r4 : c0bae5a0
[14259.117442] r3 : 60000093  r2 : 60000093  r1 : 00000000  r0 : c00945d8
[14259.126210] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[14259.135674] Control: 00c5387d  Table: 04568008  DAC: 00000055
[14259.143767] Process swapper (pid: 0, stack limit = 0x027227b1)
[14259.151982] Stack: (0xc0ba3f38 to 0xc0ba4000)
[14259.158643] 3f20:                                                       c0ba3
f6c 00000000
[14259.171141] 3f40: c00f5f64 c0053e94 c0bae5a0 c0ba5028 c0ba5020 c0b6fa50 00000
000 c0cee000
[14259.183484] 3f60: c0ba3f7c c0ba3f70 c0054234 c0053e80 c0ba3f94 c0ba3f80 c086f
fa0 c0054224
[14259.195896] 3f80: c0cee068 c0ba5028 c0ba3fa4 c0ba3f98 c0b39a70 c086ff30 c0ba3
ff4 c0ba3fa8
[14259.208385] 3fa0: c0b39f7c c0b39a64 ffffffff ffffffff 00000000 c0b39654 00000
000 c0b6fa50
[14259.221101] 3fc0: 26627329 00000000 00000000 c0b39330 00000055 00c0387d 00000
c42 1bfe9300
[14259.233997] 3fe0: 410fb767 00c5387d 00000000 c0ba3ff8 00000000 c0b39af8 00000
000 00000000
[14259.247091] Backtrace: invalid frame pointer 0xfffffffc
[14259.254962] Code: e59f40d8 e1a0300d e0033004 e5935000 (e1a050a5)
[14259.264009] ---[ end trace c922b9396d60ae34 ]---
[14259.271341] Kernel panic - not syncing: Attempted to kill the idle task!
[14259.280690] ---[ end Kernel panic - not syncing: Attempted to kill the idle t
ask! ]---
```

bao-eng, 2021-09-29:
I have similar problem(same address bf8447b8) on RPI0w during over the air system update using [mender](https://github.com/mendersoftware/mender) .
```
/etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2
```
```
uname -a
Linux raspberry 5.10.17+ #1414 Fri Apr 30 13:16:27 BST 2021 armv6l GNU/Linux
```
```
vcgencmd version
Apr 30 2021 13:47:07 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)
```
```
[  984.121534] 8<--- cut here ---
[  984.126130] Unable to handle kernel paging request at virtual address bf8447b8
[  984.134914] pgd = fbc527d6
[  984.139121] [bf8447b8] *pgd=00000000
[  984.144234] Internal error: Oops: 80000005 [#1] ARM
[  984.150645] Modules linked in: tun aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc rtc_ds1307 regmap_i2c mcp251x can_dev rndis_wl6
[  984.216889] CPU: 0 PID: 632 Comm: mender Tainted: G         C        5.10.17+ #1414
[  984.228761] Hardware name: BCM2835
[  984.234263] PC is at 0xbf8447b8
[  984.239492] LR is at 0x2a
[  984.244183] pc : [<bf8447b8>]    lr : [<0000002a>]    psr: a0000113
[  984.252652] sp : c871be10  ip : 0000002a  fp : c871be4c
[  984.260145] r10: 0000012c  r9 : c871bec4  r8 : 00000003
[  984.267608] r7 : c2914800  r6 : c3762840  r5 : c0ba5028  r4 : c0cbcae8
[  984.276435] r3 : 00000000  r2 : c06ed948  r1 : 00000500  r0 : c3762840
[  984.285177] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  984.294658] Control: 00c5387d  Table: 086e4008  DAC: 00000055
[  984.302713] Process mender (pid: 632, stack limit = 0x0a643ec1)
[  984.310951] Stack: (0xc871be10 to 0xc871c000)
[  984.317501] be00:                                     00000000 c0ba5028 c0cd6f68 f5adcfbe
[  984.329995] be20: c871be64 c0ba5028 c2914800 f5adcfbe c0ba5028 bf00876c c0cd6f68 00000040
[  984.342430] be40: c871be74 c871be50 c071293c bf008778 00000000 c3762840 bf05b578 f5adcfbe
[  984.354795] be60: c0cd6fb0 00000000 c871be8c c871be78 c071299c c07128e4 c0cd6fb0 00000000
[  984.367307] be80: c871beb4 c871be90 c0712b18 c0712988 c0cd6fb0 00000001 00000000 c0cd6f60
[  984.380045] bea0: 00000040 c871bec4 c871bf04 c871beb8 c07143e8 c0712ab8 c0c252d8 00010b39
[  984.393110] bec0: 00000006 c871bec4 c871bec4 c871becc c871becc f5adcfbe c00f5f64 c0cee760
[  984.406281] bee0: 00000004 00000003 ffffe000 c1076000 00000100 00000008 c871bf54 c871bf08
[  984.419689] bf00: c00094b0 c0714294 00000080 f5adcfbe c00844d0 00400040 00000001 00010b38
[  984.433234] bf20: 00000008 0000000c c0071018 00000000 c0c8d190 00000001 00000000 c1076000
[  984.446901] bf40: ff310aff 018016c0 c871bf6c c871bf58 c0028280 c00093a4 00000000 c0c8d190
[  984.460712] bf60: c871bf94 c871bf70 c006c90c c00281a8 c871bfb0 c0ba5a2c ffffffff 00c5387d
[  984.474536] bf80: 00c5387d ff310aff c871bfac c871bf98 c000933c c006c8b8 001a5658 80000010
[  984.488331] bfa0: 00000000 c871bfb0 c0008fcc c000930c 018e8240 00007dc0 00000029 70fa7b29
[  984.502186] bfc0: ef69b3f7 8fc4be72 a530e50d ff3b427b 8e1ee438 ff310aff 018016c0 007b61f4
[  984.516059] bfe0: 821a4018 018b533c 001a5410 001a5658 80000010 ffffffff 00000000 00000000
[  984.529920] Backtrace: 
[  984.535681] [<bf00876c>] (ipv6_rcv [ipv6]) from [<c071293c>] (__netif_receive_skb_one_core+0x64/0x84)
[  984.550519]  r7:00000040 r6:c0cd6f68 r5:bf00876c r4:c0ba5028
[  984.558994] [<c07128d8>] (__netif_receive_skb_one_core) from [<c071299c>] (__netif_receive_skb+0x20/0x7c)
[  984.573962]  r5:00000000 r4:c0cd6fb0
[  984.580190] [<c071297c>] (__netif_receive_skb) from [<c0712b18>] (process_backlog+0x6c/0x110)
[  984.594002]  r5:00000000 r4:c0cd6fb0
[  984.600117] [<c0712aac>] (process_backlog) from [<c07143e8>] (net_rx_action+0x160/0x414)
[  984.613331]  r9:c871bec4 r8:00000040 r7:c0cd6f60 r6:00000000 r5:00000001 r4:c0cd6fb0
[  984.626181] [<c0714288>] (net_rx_action) from [<c00094b0>] (__do_softirq+0x118/0x3e0)
[  984.639049]  r10:00000008 r9:00000100 r8:c1076000 r7:ffffe000 r6:00000003 r5:00000004
[  984.651943]  r4:c0cee760
[  984.656932] [<c0009398>] (__do_softirq) from [<c0028280>] (irq_exit+0xe4/0xf4)
[  984.666746]  r10:018016c0 r9:ff310aff r8:c1076000 r7:00000000 r6:00000001 r5:c0c8d190
[  984.679620]  r4:00000000
[  984.684587] [<c002819c>] (irq_exit) from [<c006c90c>] (__handle_domain_irq+0x60/0xb8)
[  984.697366]  r5:c0c8d190 r4:00000000
[  984.703373] [<c006c8ac>] (__handle_domain_irq) from [<c000933c>] (bcm2835_handle_irq+0x3c/0x44)
[  984.717055]  r9:ff310aff r8:00c5387d r7:00c5387d r6:ffffffff r5:c0ba5a2c r4:c871bfb0
[  984.729814] [<c0009300>] (bcm2835_handle_irq) from [<c0008fcc>] (__irq_usr+0x4c/0x60)
[  984.742743] Exception stack(0xc871bfb0 to 0xc871bff8)
[  984.750319] bfa0:                                     018e8240 00007dc0 00000029 70fa7b29
[  984.763458] bfc0: ef69b3f7 8fc4be72 a530e50d ff3b427b 8e1ee438 ff310aff 018016c0 007b61f4
[  984.776549] bfe0: 821a4018 018b533c 001a5410 001a5658 80000010 ffffffff
[  984.785656]  r5:80000010 r4:001a5658
[  984.791735] Code: bad PC value
[  984.797193] ---[ end trace 6de158139403431f ]---
[  984.804230] Kernel panic - not syncing: Fatal exception in interrupt
[  984.813044] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

pastcompute, 2021-11-17:
I am seeing the same problem: I have a Zero W and all it is running is raspivid and cvlc, and I randomly get a hard hang, and once I connected a serial cable discovered the same ipv6_rcv stack trace as above, in a random process, anywhere between 5 and 30 minutes after power on. I have a NOIR camera and a HC-SR501 PIR sensor and with it connected to a 2.4A USB power supply anda USB current monitor it never draws more than about 0.7A so I don't think it is power related.

```
Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage4

Linux zcam1 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
Revision	: 9000c1
Serial		: 0000000010fd0997
Model		: Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=1
<snipped>
```

Ironically it panicked when I ran raspinfo to get the above information!

pastcompute, 2021-11-17:
I am going to disable ipv6 (for example https://www.howtoraspberry.com/2020/04/disable-ipv6-on-raspberry-pi/ ) because I don't need it, and see if the problem goes away

bao-eng, 2021-11-17:
@pastcompute it would be great to know if it worked

pastcompute, 2021-11-18:
@bao-eng well, this time it stayed up overnight. But instead I was seeing non-panic crashes in dmesg instead, so the problem has "moved". On one occasion the ssh connection hung but I still had the serial console and could reboot from there.

Now I have gone a rabbit hole, I tried increasing the amount of RAM by reducing the GPU size, and then by upgrading the kernel which it turned out broke raspivid and I have to use libcamera instead, and now I'm seeing this flooding domes instead: `[  157.922466] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000` I am just about ready to spin a fresh sd card and start again...

pastcompute, 2021-11-18:
On the occasion I lost ssh but still had a serial console, there was this in dmesg:
```
[ 8387.477044] 8<--- cut here ---
[ 8387.483377] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[ 8387.497240] pgd = 41cf44ae
[ 8387.502866] [00000004] *pgd=00000000
[ 8387.508987] Internal error: Oops: 17 [#1] ARM
[ 8387.515546] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes spidev 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 raspberrypi_hwmon cfg80211 rfkill i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig spi_bcm2835 videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[ 8387.576577] CPU: 0 PID: 1295 Comm: kworker/u2:0 Tainted: G         C        5.10.17+ #1403
[ 8387.589702] Hardware name: BCM2835
[ 8387.596876] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
[ 8387.606530] PC is at __mmc_start_request+0x20/0x16c
[ 8387.615130] LR is at __mmc_start_request+0x20/0x16c
[ 8387.622945] pc : [<c068ebd0>]    lr : [<c068ebd0>]    psr: 60000013
[ 8387.632242] sp : c34b1c18  ip : c34b1c18  fp : c34b1c34
[ 8387.640573] r10: ad000002  r9 : 00000001  r8 : 00000002
[ 8387.648061] r7 : c470a818  r6 : 00000000  r5 : c34b1cd8  r4 : c125b800
[ 8387.656905] r3 : 00000002  r2 : 00000000  r1 : 00000000  r0 : 00000000
[ 8387.667020] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 8387.676544] Control: 00c5387d  Table: 04d34008  DAC: 00000055
[ 8387.685837] Process kworker/u2:0 (pid: 1295, stack limit = 0x60220668)
[ 8387.695581] Stack: (0xc34b1c18 to 0xc34b2000)
[ 8387.702986] 1c00:                                                       c125b800 c34b1cd8
[ 8387.715747] 1c20: 00000000 c470a818 c34b1c54 c34b1c38 c068eda8 c068ebbc c34b1cd8 80000000
[ 8387.728213] 1c40: c125b800 c470a818 c34b1c74 c34b1c58 c068eee0 c068ed28 c18fc800 80000000
[ 8387.741056] 1c60: 00080000 c470a818 c34b1d74 c34b1c78 c069b5d8 c068ee84 00000040 c18fc800
[ 8387.755418] 1c80: 3b9aca00 00000000 00000040 00000001 d78afd6a 00000818 00000400 00000000
[ 8387.769332] 1ca0: 3b9aca00 00000000 00000200 00000002 00000000 00000000 00000100 00000000
[ 8387.782617] 1cc0: 00000000 c34b1cd8 00000001 00000000 c34b1c90 00000000 00000000 c34b1d14
[ 8387.795773] 1ce0: c34b1ca0 00000000 00000000 c34b1cec c34b1cec 00000000 c34b1cf8 c34b1cf8
[ 8387.809351] 1d00: c068dc38 00000000 00000000 00000000 00000000 00000035 ad000002 00000000
[ 8387.824392] 1d20: 00000000 00000000 00000000 000001b5 00000000 00000000 00000000 c34b1ca0
[ 8387.839427] 1d40: c34b1cd8 fe1bc978 c0c252d8 00000400 000005bc c1045c00 c470a818 00000001
[ 8387.853186] 1d60: 00008000 000001ff c34b1dbc c34b1d78 c069d0d8 c069b3b8 00000001 c470a818
[ 8387.868055] 1d80: 00000002 00000200 c34b1dbc 00000001 c05ae134 c4735480 c3da1800 00008000
[ 8387.882974] 1da0: c34b1e84 00000000 00000002 c3da0e10 c34b1dd4 c34b1dc0 c069d2ec c069d010
[ 8387.898621] 1dc0: c470a818 000005bc c34b1dec c34b1dd8 bf54e680 c069d2c8 c4735480 c3da1800
[ 8387.913554] 1de0: c34b1e1c c34b1df0 bf54ede4 bf54e650 00000001 000005bc 00000002 c34b1e84
[ 8387.927458] 1e00: c34b1e84 0000000f 0000000f 000005bc c34b1f0c c34b1e20 bf54ba90 bf54ed3c
[ 8387.941810] 1e20: 00000000 c0ba5028 00000001 c0871300 c34b1e74 c34b1e40 c00144a0 18002020
[ 8387.957572] 1e40: 000000ff 00000001 00000000 c3da0c20 00000014 c004c894 c3da0c10 c0871300
[ 8387.973067] 1e60: c0871300 c19a5000 c34b1e90 000005bc c3da0c00 00000000 00000000 00000000
[ 8387.987032] 1e80: 00000002 c4735480 c4735480 00000001 05bc020e 00000000 00000116 c003df18
[ 8388.002175] 1ea0: c34b1ee4 c34b1eb0 c00144a0 c0013658 c10ee018 00000000 c003df18 c34b1f74
[ 8388.017463] 1ec0: c0c252d8 c003dbf8 00000040 c003df18 00000000 00000000 c34b1f0c fe1bc978
[ 8388.033000] 1ee0: c00f5f64 c3da0e10 c292bba0 00000040 c18fb900 00000000 00000000 c3da0e14
[ 8388.047652] 1f00: c34b1f44 c34b1f10 c003dc0c bf54a1a4 c003e17c 00000001 c34b1f44 c292bba0
[ 8388.061578] 1f20: c1045200 c292bbb4 c1045214 c0c19e40 00000088 c1045200 c34b1f74 c34b1f48
[ 8388.075785] 1f40: c003df18 c003da10 00000000 c7062000 c735a840 00000000 c34b0000 c003dee4
[ 8388.091258] 1f60: c292bba0 c34a1e88 c34b1fac c34b1f78 c0043848 c003def0 c7062020 c7062020
[ 8388.106682] 1f80: c002fc50 c735a840 c0043700 00000000 00000000 00000000 00000000 00000000
[ 8388.120608] 1fa0: 00000000 c34b1fb0 c00082ec c004370c 00000000 00000000 00000000 00000000
[ 8388.135918] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 8388.151419] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[ 8388.166955] Backtrace:
[ 8388.172785] [<c068ebb0>] (__mmc_start_request) from [<c068eda8>] (mmc_start_request+0x8c/0xbc)
[ 8388.186893]  r7:c470a818 r6:00000000 r5:c34b1cd8 r4:c125b800
[ 8388.195406] [<c068ed1c>] (mmc_start_request) from [<c068eee0>] (mmc_wait_for_req+0x68/0xe0)
[ 8388.209506]  r7:c470a818 r6:c125b800 r5:80000000 r4:c34b1cd8
[ 8388.219136] [<c068ee78>] (mmc_wait_for_req) from [<c069b5d8>] (mmc_io_rw_extended+0x22c/0x35c)
[ 8388.234747]  r7:c470a818 r6:00080000 r5:80000000 r4:c18fc800
[ 8388.243615] [<c069b3ac>] (mmc_io_rw_extended) from [<c069d0d8>] (sdio_io_rw_ext_helper+0xd4/0x1c4)
[ 8388.258103]  r10:000001ff r9:00008000 r8:00000001 r7:c470a818 r6:c1045c00 r5:000005bc
[ 8388.272564]  r4:00000400
[... cut ...]
[ 8388.502694] 1fa0:                                     00000000 00000000 00000000 00000000
[ 8388.516110] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 8388.529293] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 8388.538684]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0043700
[ 8388.554031]  r4:c735a840
[ 8388.559819] Code: e8bd4000 e1a06001 e1a04000 eb000def (e5962004)
[ 8388.569441] ---[ end trace dc53238821f16cb7 ]---
```