raspberrypi/linux #4467: CVE-2021-33909

yarmooh, 2021-07-22:
Hello,

I'd like to know which kernel version is or is not affected by this CVE, is RaspberryPi (CM3 in my case) affected at all?
This CVE has been reported and I see already debian-related patched kernels, but can't find information regarding rpi.

pelwell, 2021-07-22:
A description of the vulnerability was published two days ago: https://nvd.nist.gov/vuln/detail/CVE-2021-33909

The Linux kernel devs will prepare patches and back-port them to earlier kernels including 5.10 - our current release - at which point our kernel package will be updated.

XECDesign, 2021-07-22:
Should be fixed in 5.10.52

https://www.linuxkernelcves.com/cves/CVE-2021-33909

pelwell, 2021-07-22:
You beat me to it. The patches are already available upstream and will be picked up in the next version bump.

pelwell, 2021-07-23:
The current `rpi-update` kernel includes the relevant patch (174c34d9cda1).

XECDesign, 2021-07-23:
@popcornmix Does firmware/stable need to be updated?

pelwell, 2021-07-23:
Not until #4465 is sorted.