raspberrypi/linux #4471: [UART] Serial problem

RuffaloLavoisier, 2021-07-23:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

serial port not working

**To reproduce**
List the steps required to reproduce the issue.

1. start minicom
```
$ sudo apt-get install minicom #setting 115200 and /dev/ttyUSB0
```
2. power on pi
```
$ sudo minicom
```
**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

break serial value

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW - Raspberry pi 3B
* Which OS and version (`cat /etc/rpi-issue`)?

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

* Which firmware version (`vcgencmd version`)?

May 27 2021 14:05:31 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

* Which kernel version (`uname -a`)? -
Linux fileserver 5.10.52-v7+ #2 SMP Sat Jul 24 00:13:41 KST 2021 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

no DMA platform data

**Additional context**
Add any other relevant context for the problem.

JamesH65, 2021-07-23:
Sorry, it is not clear from this report what issue you are seeing. Can you please clarify.

RuffaloLavoisier, 2021-07-23:
Thank you

/boot/config.txt and enable_uart=1.I tried to connect to the serial raspberry from putty and minicom after kernel build, but the connection failed.

kripton, 2021-07-23:
Are you trying to connect from the Pi to somewhere or from your PC to the Pi?
Can you try to do a serial port loopback test on the Pi in order to rule out problems with connected hardware? https://amitasinghchauhan.medium.com/serial-port-debugging-101-loopback-test-4a7e40da9055

RuffaloLavoisier, 2021-07-24:
> Are you trying to connect from the Pi to somewhere or from your PC to the Pi?
> Can you try to do a serial port loopback test on the Pi in order to rule out problems with connected hardware? https://amitasinghchauhan.medium.com/serial-port-debugging-101-loopback-test-4a7e40da9055

Thank you 

I tried debugging according to the link, but the results are not output.

RuffaloLavoisier, 2021-07-24:
> Thank you
>
> I tried debugging according to the link, but the results are not output.

My Develop Enviroment Linux Ubuntu and This is the result of connecting TX and RX to each other and using minicom for pie.

JamesH65, 2021-07-24:
Since this is not a bug (I use this all the time and I know it should work), I think the best place to deal with this will be the forums. ww.raspberrypi.org/forums

closed 2021-07-24 by JamesH65

RuffaloLavoisier, 2021-07-24:
> Since this is not a bug (I use this all the time and I know it should work), I think the best place to deal with this will be the forums. ww.raspberrypi.org/forums

Thank you. I'll look for it. What keyword should I approach?

JamesH65, 2021-07-24:
Just add a new post with a descriptive title, describe what you are trying to do, how, and what isn't working.

RuffaloLavoisier, 2021-07-24:
> Just add a new post with a descriptive title, describe what you are trying to do, how, and what isn't working.

Ok , Thank You

RuffaloLavoisier, 2021-08-03:
> Just add a new post with a descriptive title, describe what you are trying to do, how, and what isn't working.

Hello, 11 days ago, there was an issue regarding serial issues.
I solved the problem today.
After rebuilding the kernel and changing the serial cable, it was recognized successfully.

>uname -a
>>Linux fileserver 5.10.52+ #1 Fri Jul 23 02:19:04 KST 2021 armv6l GNU/Linux

Thank you