raspberrypi/linux #4474: Possible VC4 memory leak introduced in 5.10.47/5.10.48

kwirk, 2021-07-25:
**Describe the bug**
After upgrading from 5.10.46 to 5.10.48, I had issues with oom-killer killing Kodi in about 20mins of starting to watch a video. The system then became unstable, with other processes being randomly killed, and a system reboot was required.

**To reproduce**
Play video on Kodi. Monitor *kmalloc-128* increasing via `slaptop -s c`

**Expected behaviour**
Kodi not be oom-killed

**Actual behaviour**
Kodi oom-killed, along with other processes. System unstable requiring reboot.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Model: Raspberry Pi 4
OS: Arch Linux ARM
Firmware: Jul 21 2021 16:21:46 version 6a796bb0062a6c75191c57cba1c13f9300076d02
Kernel: 5.10.52 (issue seen since updating to 5.10.48; but may have also been present on slightly earlier version)

**Logs**
```
$ sudo slaptop -s c
Active / Total Objects (% used)    : 266001 / 289256 (92.0%)
 Active / Total Slabs (% used)      : 13992 / 13992 (100.0%)
 Active / Total Caches (% used)     : 137 / 265 (51.7%)
 Active / Total Size (% used)       : 93490.38K / 103413.99K (90.4%)
 Minimum / Average / Maximum Object : 0.17K / 0.36K / 16.00K

  OBJS ACTIVE  USE OBJ SIZE  SLABS OBJ/SLAB CACHE SIZE NAME                   
 83628  83586  99%    0.38K   3987       21     31896K kmalloc-128
 15316  14367  93%    0.55K    547       28      8752K inode_cache
 30720  30691  99%    0.27K   1024       30      8192K ext4_groupinfo_4k
 31424  28348  90%    0.25K   1964       16      7856K kmalloc-64
 24975  17937  71%    0.29K    925       27      7400K dentry
 26192  26153  99%    0.24K   1637       16      6548K kernfs_node_cache
 11169  11152  99%    0.45K    657       17      5256K radix_tree_node
  3728   1177  31%    0.95K    233       16      3728K ext4_inode_cache
 11222  10974  97%    0.26K    362       31      2896K vm_area_struct
  1168   1127  96%    2.00K     73       16      2336K kmalloc-1k
  1872   1779  95%    1.00K    117       16      1872K kmalloc-512
   392    378  96%    3.94K     49        8      1568K task_struct
```

Noted high number of `drm_atomic_helper_setup_commit`:
```
$ sudo sort -rn /sys/kernel/slab/kmalloc-128/alloc_calls | head
  80462 drm_atomic_helper_setup_commit+0x21c/0x7bc [drm_kms_helper] age=3/84268/167031 pid=276-602 cpus=0-3
    153 drm_property_create+0xa0/0x190 [drm] age=167039/167040/167044 pid=276 cpus=1
     93 snd_info_create_entry+0x28/0xe0 [snd] age=167039/167148/167184 pid=244-276 cpus=1
     47 drm_mode_duplicate+0x20/0x4c [drm] age=167032/167032/167032 pid=276 cpus=1
     28 drm_universal_plane_init+0x118/0x604 [drm] age=167039/167039/167039 pid=276 cpus=1
     17 __ipv6_dev_mc_inc+0x154/0x3c4 [ipv6] age=165848/166512/167270 pid=1-594 cpus=0-3
     14 v3d_fence_create+0x28/0x9c [v3d] age=3768/64040/165436 pid=322-323 cpus=0-2
     11 devm_snd_soc_register_component+0x2c/0x80 [snd_soc_core] age=167040/167054/167107 pid=261-276 cpus=1-2
     10 snd_ctl_new+0x34/0x8c [snd] age=167039/167068/167182 pid=244-276 cpus=1
      9 __drmm_add_action_or_reset+0x14/0x30 [drm] age=167038/167047/167078 pid=271-276 cpus=1
```

I downgraded to 5.10.46 and this has resolved the issue:
```
$ sudo grep drm_atomic_helper_setup_commit /sys/kernel/slab/kmalloc-128/alloc_calls
      9 drm_atomic_helper_setup_commit+0x214/0x790 [drm_kms_helper] age=1/236240/305496 pid=113-590 cpus=0-3
      1 drm_atomic_helper_setup_commit+0x538/0x790 [drm_kms_helper] age=1 pid=590 cpus=1
      1 drm_atomic_helper_setup_commit+0x6f4/0x790 [drm_kms_helper] age=304206 pid=590 cpus=0
```

**Additional context**
Also report on anholt/linux#156
Forum post:  https://www.raspberrypi.org/forums/viewtopic.php?f=53&t=316394

kwirk, 2021-07-25:
I've also built latest rpi-5.10.y (7aa66c086ae01c4f77d54aaeb1d9f62573a7b748) including #4470 applied as a patch, which results in same behaviour: increasing number of allocation calls from `drm_atomic_helper_setup_commit` in *kmalloc-128*.

popcornmix, 2021-07-25:
I would be suspecting one of these : https://github.com/raspberrypi/linux/pull/4441
Can you try building at `3282562~` (i.e. one commit before that PR) and at `3d7507e` (i.e. end of PR) and check if each is okay?

kwirk, 2021-07-25:
So I'm currently running `3282562~` (64f5ffe) without issue (was packaged as 5.10.42-2 on Arch Linux ARM).

Currently building 3d7507e…

kwirk, 2021-07-25:
Looks like 3d7507e has the issue.

I saw on #4441 f14d057 introduced a `atomic_commit_setup` function and 2f3e2d8 switched to using `drm_atomic_helper_commit`, and `drm_atomic_helper_setup_commit` is where I'm seeing constant allocations coming from…

popcornmix, 2021-07-25:
Sounds likely. Are you able to confirm memory leak starts with that commit?

kwirk, 2021-07-25:
So the issue was introduced with 135abe8. I also applied e621e4e as a patch on top of this, as noted this commit related to same section of code (I had tested e621e4e on latest main previously but wanted to double check), but the issue persists.

popcornmix, 2021-07-26:
@mripard any thoughts on a memory leak introduced with "drm/vc4: kms: Wait on previous FIFO users before a commit"?

mripard, 2021-07-26:
I think that we have the drm_crtc_commit ref counting wrong somehow and thus never free it for each commit.

However, I started to run an application that does a commit each frame, and while I can definitely see the kmalloc-128 slab increase over time, I'm not getting the drm_atomic_helper_setup_commit call, at all, which is fairly weird.

kwirk, 2021-07-26:
Built and tested latest rpi-5.10.y (9a22b00c1e40799b4014a96644d811307e7114d2), now #4478 is merged. Working fine so far with no sign on *kmalloc-128* increasing :+1:

Marcool04, 2021-08-03:
I have been experiencing out of memory issues on a raspberry pi 4, 8Gb RAM, armv7h system, using kernel 5.10.52. In my case, the reversions in pull request https://github.com/raspberrypi/linux/pull/4478 have not solved the issue. Archlinuxarm [linux-raspberrypi4](https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4) package at version 5.10.52-5 is built from commit https://github.com/raspberrypi/linux/commit/6495fa3396fb387d3871ced42a30281fe7be5352, and therefore includes the reversions made in https://github.com/raspberrypi/linux/pull/4478 that are supposed to have solved this issue according to above comment, but still has the oom behaviour on my system.

Change occurred somewhere between https://github.com/raspberrypi/linux/commit/2697f7403187bb2bb61cc716f33ee9f6cfb9af7c (which I'm running now with no oom issues) and https://github.com/raspberrypi/linux/commit/86729e78125d4f3d203457940feee8bc97b11f6c (when they started, and were not fixed by https://github.com/raspberrypi/linux/pull/4478). At least one other user has reported a similar situation on the archlinuxarm forum: https://archlinuxarm.org/forum/viewtopic.php?f=60&t=15460&p=67115#p67115

It would seem that on aarch64 systems the reversions did indeed fix the problem though.

kwirk, 2021-08-03:
@Marcool04: It would be interesting to see if you have the memory leak in the same place, as I don't believe the code which caused this memory leak is reachable (not 100% confident however…).

Could you add `slub_debug=U` to your boot parameters (in `/boot/cmdline.txt`) and look at `slabtop -s -c` and `allloc_calls` similar to my original comment?

kwirk, 2021-08-03:
I've been using 5.10.52-4 without issue (86729e78125d4f3d203457940feee8bc97b11f6c), and just upgraded to 5.10.52-6 (4e5d621498df7fce8b2cf90ac6bb7b2efbdbec5e), and again no issues.
```
$ uname -mr
5.10.52-6-ARCH armv7l
```

Marcool04, 2021-08-03:
Indeed @kwirk there would appear to be some error on my part. Either this has been resolved between 5.10.52-5 and 5.10.52-6 (but no commit in between seems to indicate it should have), or I have an issue somewhere else, as I am now running the latest archlinuxarm kernel just fine…

````
# uname -mr
5.10.52-6-ARCH armv7l
````

````
# slabtop -s c
Active / Total Objects (% used)    : 304838 / 309615 (98.5%)
 Active / Total Slabs (% used)      : 15203 / 15203 (100.0%)
 Active / Total Caches (% used)     : 141 / 270 (52.2%)
 Active / Total Size (% used)       : 103991.14K / 106731.19K (97.4%)
 Minimum / Average / Maximum Object : 0.17K / 0.34K / 16.00K

  OBJS ACTIVE  USE OBJ SIZE  SLABS OBJ/SLAB CACHE SIZE NAME
 13776  13766  99%    0.95K    861       16     13776K ext4_inode_cache
 37557  37535  99%    0.29K   1391       27     11128K dentry
 21828  21813  99%    0.45K   1284       17     10272K radix_tree_node
 16016  15974  99%    0.55K    572       28      9152K inode_cache
 31651  31627  99%    0.26K   1021       31      8168K vm_area_struct
  1576   1571  99%    3.94K    197        8      6304K task_struct
 23856  23802  99%    0.24K   1491       16      5964K kernfs_node_cache
431086  31073  99%    0.18K   1413       22      5652K dm_exception
 27048  27047  99%    0.19K   1288       21      5152K anon_vma_chain
 20512  20199  98%    0.25K   1282       16      5128K kmalloc-64
 18090  18073  99%    0.22K   1005       18      4020K buffer_head
52  14646  99  98%    0.21K    814       18      3256K anon_vma
 14070  14055  99%    0.19K    670       21      2680K ext4_extent_status
  1808   1661  91%    1.00K    113       16      1808K kmalloc-512
1  816    772  94%    2.00K     51       16      1632K kmalloc-1k
906  99  2886  98%    0.38K    139       21      1112K kmalloc-128
````

````
# sort -rn /sys/kernel/slab/kmalloc-128/alloc_calls | head
    897 nf_ct_ext_add+0x88/0x104 [nf_conntrack] age=5/3598/9854 pid=0-2533 cpus=0-3
    153 drm_property_create+0xa0/0x190 [drm] age=9740/9744/9757 pid=81 cpus=3
     49 init_rescuer+0x3c/0xe0 age=9766/12459/13848 pid=1-479 cpus=0-3
     42 create_worker+0x58/0x1e0 age=7772/12630/13848 pid=1-560 cpus=0-3
     39 add_sysfs_param+0x3c/0x1d0 age=9778/12740/13827 pid=1-480 cpus=0-3
     33 snd_info_create_entry+0x28/0xe0 [snd] age=9749/9772/9815 pid=81-471 cpus=1,3
     30 __devm_request_region+0x34/0xac age=9741/12089/13831 pid=1-480 cpus=0-3
     28 drm_universal_plane_init+0x118/0x604 [drm] age=9740/9740/9747 pid=81 cpus=3
     14 __ipv6_dev_mc_inc+0x154/0x3c4 [ipv6] age=9540/10017/10340 pid=1-574 cpus=1,3
````

This time, the difference is I did not upgrade my AUR packages (which I usually do at the same time as I upgrade with pacman)…

Maybe one of `electron12-bin-12.0.15-1  highway-0.14.0-1  moarvm-2021.07-1  nqp-2021.07-1  pbget-2021.7-1  pm2ml-2021.8-1  python-mautrix-0.9.10-1 python-tifffile-2021.7.30-1  python3-xcpf-2021-1  rakudo-2021.07-1  rompr-1.59.1-1  mx-puppet-discord-git-r248.c17384a-3` is causing a memory leak… All of those packages are used by systemd services that start on boot so they are candidates. I'll have to upgrade them one by one and see I guess!
Thanks for taking the time to help.
Regards,
Mark.

closed 2021-08-04 by kwirk