raspberrypi/linux #4475: Display Corruptions with KMS and Overscan Margins on the RaspberryPi 4

mripard, 2021-07-26:
Moving the cursor along the edges of the display creates some visual artifacts when disable_overscan isn't set.

**To reproduce**

Build a kernel based on current the rpi-5.10.y branch (7aa66c086ae0 in my case), start X and move the cursor along the edges. Complete corruption of the display should occur.

timg236, 2021-07-26:
Yes, we can see that here

popcornmix, 2021-07-26:
Also noticed that top right corner causes:
```
[  227.049120] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  227.049163] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  237.289091] vc4-drm gpu: [drm] *ERROR* Timed out waiting for flip_done
```

mripard, 2021-07-26:
For the record, I noticed this this morning while debugging something else and just opened the issue so that it doesn't get forgotten.

timg236, 2021-07-26:
I'm finding that when the mouse is stationary everything is ok. However, if it moves and is within 62 pixels of the bottom or right edge then the corrupt appears.
This seems to correspond with the scaling mode changing from zero to 4 or 5 for that d-list element. In the stationary state the scaling mode is always zero. 
scl0_mode, scl1_mode in controlword0

popcornmix, 2021-07-26:
A possibly related issue if with kms, and two displays, from desktop rapidly switching mouse pointer between displays results in occasional glitches (on the framebuffer). It doesn't seem to be clock related (forcing it high doesn't avoid it).
I'm suspicious that the mouse overlay being removed/added is causing a one frame glitch.
When it doesn't change screen it presumably uses the simpler code path allowed by `vc4_plane_atomic_async_check`.

popcornmix, 2021-07-26:
I've reverted most of https://github.com/raspberrypi/linux/pull/4441 and the screen corruption when cursor near edges and glitches when mouse pointer switches screens issues are gone.

popcornmix, 2021-07-27:
This should be resolved with latest rpi-update kernel.

mripard, 2021-08-30:
I reverted your revert and tried to fix all the pending bugs we had. I've been working on this one today and followed the suggestion from @timg236 on #4465 that it might be related to the core clock being too low on 4k.

I'm seeing this on 1080p, but I still raised the core clock to 500MHz which is more than the double of what is computed (220MHz). I'm still getting both the corruptions (even though I'm getting less corruptions, possibly from my other fixes), and some timeouts.

Moving the cursor in itself (using xdotool mousemove) doesn't trigger it, so it looks like it's the sequence of commits when moving the cursor that triggers it.

mripard, 2021-11-18:
For the record, this also affects rpi-5.15.y with a single display

PR #4776 (2021-12-17): Fix for cursor corruption with overscan
Hi,

Here's a PR that fixes #4475.

The last commit has most of the details, but the basic idea is that when we move the cursor along the edges, we end up reusing and corrupting the active dlist.

Let me know what you think,
Maxime