raspberrypi/linux #4477: SC16IS750 driver aborts I2C transaction

requireiot, 2021-07-26:
I am using the driver for the I2C-to-UART bridge. The line in /boot/config.txt is `dtoverlay=sc16is74x-i2c,addr=0x48,int_pin=27` . I have tried this on multiple Raspberry Pi 2B boards, same result.

Observing the I2C communication on a scope (see attached screenshot), i can see that sometimes there is an attempted _write_ access to index 0x40 (register 0x08), which doesn't exist on these chips according to the datasheet, and then the start of a new transmission, just 2 bits, and then SDA and SCL go high, which confuses the slave.
This happens every couple of minutes, while there is a repeated access to /dev/ttySC0 from lcdproc.
[image]

I would expect only complete byte transfers over I2C, and I would not expect any write access to index 0x40 = register 8 = TXLVL, which is a read-only register

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? **Pi2B**
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2020-02-13 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
* Which firmware version (`vcgencmd version`)? May 27 2021 14:04:13  Copyright (c) 2012 Broadcom  version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)
* Which kernel version (`uname -a`)? Linux raspi10 5.10.17-v7+ # 1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux

pelwell, 2021-07-26:
I can't completely explain your problem yet, but looking that the sc16is750-i2c overlay (and the other sc16is75x overlays) I'm bothered by the fact that it uses (by default) GPIO 24 as a falling-edge interrupt without enabling a pull-up.

GPIO 24 (like GPIO 27 that you are using) pulls low by default, but the SC.16IS750 is likely to be driving the interrupt pin in an open-drain fashion, driving it low for an interrupt and otherwise letting it float.

Can you add a trace to your scope that shows GPIO 27? It looks like you have spare channels.

pelwell, 2021-07-26:
Ah - the datasheet says of INT:
> An external resistor (1 kOhm for 3.3 V, 1.5 kOhm for 2.5 V) must be connected between this pin and VDD.

That's a strong pull-up, more than the built-in pull-ups could achieve, so there must be a resistor on the board/module.

requireiot, 2021-07-26:
Well, I'm not using a "real" SC16IS750, but rather an ATmega328 that emulates the SC16IS750 and drives the IRQ line GPIO27 with a push-pull output, so shouldn't be an issue that there is no external pull-up resistor. 
If I had seen SDA go permanently _LOW_, then I would have assumed a but in my AVR code, but can't see how the slave could force SDA _HIGH_.

pelwell, 2021-07-27:
Sorry, but I'm out. It was disingenuous to not mention upfront that this wasn't a real SC16IS75x, and I don't trust the virtual device not to (for example) start to ACK too early, pulling SDA low and making a read look a write (there is nowhere in the code where there is a write to the TXLVL register, and nobody else has reported a similar problem).

closed 2021-07-27 by pelwell

requireiot, 2021-07-27:
Sorry, didn't mean to be "disingenuous", too bad it came across that way. 

My main worry here isn't the apparent write to TXLVL (which theoretically could be caused by faulty slave behavior), it is the "partial byte" that ends after 2 bits, with SDA then staying HIGH. Can't see how even a faulty slave could cause that. Anyway, I'll keep investigating on my own.

pelwell, 2021-07-27:
Applying Occam's razor I think both effects are symptoms of some underlying "badness" in the slave or the wiring - you could check the kernel log ("dmesg") for error messages.