raspberrypi/linux #4479: Microsoft 2.4GHz Wireless Mouse/Keyboard freezes requiring removal/reinsertion of USB dongle

mattmcneilgit, 2021-07-26:
**Describe the bug**
My Microsoft 2.4GHz wireless mouse/keyboard becomes unresponsive requiring the USB dongle to be quickly removed and re-inserted into the Pi4B

**To reproduce**
Reproduces at random

**Expected behaviour**
After successful boot, the USB transceiver/dongle operating the mouse and keyboard should not require occasional eject/reinsertion

**Actual behaviour**
Dongle becomes completely unresponsive

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Raspberry Pi 4 Model B Rev 1.2
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2020-12-02
* Which firmware version (`vcgencmd version`)? version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d
* Which kernel version (`uname -a`) 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux

**Logs**
From DMSEG, possibly relevant section:

[   17.040971] Bluetooth: RFCOMM ver 1.11
[ 3935.304095] v3d fec00000.v3d: MMU error from client L2T (0) at 0x56e1000, pte invalid
[ 4895.887330] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 783360)
[ 4895.888956] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 783360)
[146744.801190] usb 1-1.1: USB disconnect, device number 3
[149297.751549] usb 1-1.1: new full-speed USB device number 4 using xhci_hcd
[149297.890193] usb 1-1.1: New USB device found, idVendor=045e, idProduct=0745, bcdDevice= 6.56
[149297.890213] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[149297.890229] usb 1-1.1: Product: Microsoft\u00ae 2.4GHz Transceiver v8.0
[149297.890244] usb 1-1.1: Manufacturer: Microsoft
[149297.900372] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:045E:0745.0004/input/input6
[149297.961972] hid-generic 0003:045E:0745.0004: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input0
[149297.978158] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:045E:0745.0005/input/input7
[149297.978693] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:045E:0745.0005/input/input8
[149298.041534] hid-generic 0003:045E:0745.0005: input,hidraw1: USB HID v1.11 Mouse [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input1
[149298.052656] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:045E:0745.0006/input/input9
[149298.121539] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:045E:0745.0006/input/input11
[149298.122092] hid-generic 0003:045E:0745.0006: input,hiddev96,hidraw2: USB HID v1.11 Device [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input2
[150249.610354] usb 1-1.1: USB disconnect, device number 4
[150252.486466] usb 1-1.1: new full-speed USB device number 5 using xhci_hcd
[150252.625583] usb 1-1.1: New USB device found, idVendor=045e, idProduct=0745, bcdDevice= 6.56
[150252.625599] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[150252.625612] usb 1-1.1: Product: Microsoft\u00ae 2.4GHz Transceiver v8.0
[150252.625624] usb 1-1.1: Manufacturer: Microsoft
[150252.635563] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:045E:0745.0007/input/input12
[150252.697279] hid-generic 0003:045E:0745.0007: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input0
[150252.713337] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:045E:0745.0008/input/input13
[150252.713931] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:045E:0745.0008/input/input14
[150252.777003] hid-generic 0003:045E:0745.0008: input,hidraw1: USB HID v1.11 Mouse [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input1
[150252.790369] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:045E:0745.0009/input/input15
[150252.856921] input: Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0 System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:045E:0745.0009/input/input17
[150252.857387] hid-generic 0003:045E:0745.0009: input,hiddev96,hidraw2: USB HID v1.11 Device [Microsoft Microsoft\u00ae 2.4GHz Transceiver v8.0] on usb-0000:01:00.0-1.1/input2
[150611.313290] usb 1-1.1: USB disconnect, device number 5

P33M, 2021-07-29:
Please post the output of `lsusb -v` of the device in question.

mattmcneilgit, 2021-07-29:
Bus 001 Device 010: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0745 Nano Transceiver v1.0 for Bluetooth
  bcdDevice            6.56
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      57
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     295
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     319
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1

P33M, 2021-07-30:
Are there any other USB devices plugged in? Are you using the on-board wifi?

mattmcneilgit, 2021-07-30:
No other USB connections and wifi is turned off.