raspberrypi/linux #4482: "bcm2835_codec_buf_prepare data will not fit into plane" messages from bcm2835-codec

pelwell, 2021-07-27:
Recently I've started seeing some new error messages from the bcm2835-codec driver (a V4L2 wrapper around the VPU-based codecs). The messages are of the following form:
```
bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 792576)
```
I'll add more details (kernel versions, circumstances) as I find them, but until then this is here as a placeholder and early warning.

pelwell, 2021-07-27:
@jc-kynesim You may want to keep one eye on this thread.

randomMesh, 2021-07-27:
> I'll add more details (kernel versions, circumstances) as I find them

I see a lot of these on my Raspberry Pi 400 and here's the output of `raspinfo`. Hope this helps.
```
System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage5

Linux alix 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Revision	: c03130
Serial		: 1000000000b1dcd0
Model		: Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 27 2021 14:02:15 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Dateisystem    1K-Bl\u00f6cke Benutzt Verf\u00fcgbar Verw% Eingeh\u00e4ngt auf
/dev/root       14987608 9350632   4928724   66% /
devtmpfs         1736388       0   1736388    0% /dev
tmpfs            1868484   20652   1847832    2% /dev/shm
tmpfs            1868484    8736   1859748    1% /run
tmpfs               5120       4      5116    1% /run/lock
tmpfs            1868484       0   1868484    0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48788    209307   19% /boot
tmpfs             373696       4    373692    1% /run/user/1001

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installiert:           1.20210706
raspberrypi-sys-mods:
  Installiert:           20210706
openbox:
  Installiert:           3.6.1-8+rpt5
lxpanel:
  Installiert:           0.10.0-2+rpt15
pcmanfm:
  Installiert:           1.3.1-1+rpt27
rpd-plym-splash:
  Installiert:           0.26

Networking Information
----------------------

eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether m.m.m.m  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
[... 544 lines cut ...]
[ 1052.037995] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 43008)
[ 1052.244874] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.244938] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.245398] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.247635] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.248692] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.250395] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.250736] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.253011] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1052.253261] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1054.541061] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 353280)
[ 1054.541700] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 353280)
[ 1075.257407] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1075.258564] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1075.259052] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1075.259436] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 288000)
[ 1138.702228] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 299520)
[ 1138.704508] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 299520)
[ 1155.730425] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 276480)
[ 1155.732221] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 276480)
[ 1155.734107] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 276480)
[ 3427.998574] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 149760)
[ 3428.000107] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 149760)
[ 3428.005499] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 149760)
```

pelwell, 2021-07-27:
I'm getting them from the latest rpi-update kernel (b986fda889737f6ce0a4acda89d447f78d0a1a73), viewing a video in YouTube on Chromium and changing the quality setting. No - _reducing_ the quality setting.

Is this some pipelining effect where higher quality (resolution) video that has already been downloaded can't be decoded into buffers allocated for the new lower resolution?

pelwell, 2021-07-27:
Is 1536 bytes the size of a frame of 32x32 video in 4:2:0?

JamesH65, 2021-07-27:
Yes (32 x 32 x 12bpp) / 8

pelwell, 2021-07-27:
Obvs, but is that the significance of the 1536 in the logs above? Hmmm?

JamesH65, 2021-07-27:
I vaguely remember someone talking about a minimal size of video used somewhere in the system, for initialisation of the pipe perhaps? @jc-kynesim maybe?

6by9, 2021-07-27:
jc-kynesim is on holiday this week.

Yes the resolution is set to 32x32 by Chromium (I'm assuming this is Chromium we're looking at), and then relies on the codec producing a resolution changed event to set up the correct resolution. It sounds like we're being given buffers without the resolution having been set correctly.
This may just be a race condition between having been given encoded buffers to decode which trigger the resolution changed, and getting the raw buffers to fill.

JC needs me to have a look at have a look at the driver to make it match the spec slightly more closely over resolution changes for the next version of Chromium, so I'll bump it up the priorities list.

jc-kynesim, 2021-08-02:
Yeah this'll be part of the long standing Chrome V4L2 stateful startup problem.  AFAIK it is a driver/firmware issue.

jc-kynesim, 2021-08-02:
Its got worse with increasing chrome versions (probably timing / race as they've tuned their decode pipe)

pelwell, 2021-08-02:
I'm closing this as a known glitch that will get fixed eventually but isn't causing any malfunctions at present.

closed 2021-08-02 by pelwell

Botspot, 2022-05-21:
This error still seems to have resurfaced on Chromium 101.