raspberrypi/linux #4484: Monitor Power Saving (DPMS) does not work as intended

kphillisjr, 2021-07-28:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
DPMS Does not work as intended to turn Displays on or off.

**To reproduce (Method A):**
* Power on raspberrry pi
* wait for desktop to appear.
* open terminal
* run the command ```sudo reboot```

**Expected behavior (Method A):**
To have video when the raspberry Pi finishes rebooting.

**Actual behavior (Method A):**
The Raspberry Pi 4 reboots, and the display goes into low power mode to never return without user interruption on the display. 

**To reproduce (Method B):**
Change any setting requiring a reboot. Examples are any setting under the 'Display' tab of the 'Raspberry Pi configuration' utility.

**Expected behavior (Method B):**
To have video when the raspberry Pi finishes rebooting.

**Actual behavior (Method A):**
The Raspberry Pi 4 reboots, and the display goes into low power mode to never return without user interruption on the display. The Raspberry Pi 3 is not effected.

**To reproduce (Method C):**
* power on Raspberry Pi
* wait for Desktop to appear.
* Leave the system alone at desktop for 15 to 30 minutes.
* The Monitor should turn on low power mode.
* Move mouse, and type on keyboard 

**Expected behavior (Method C):**
For the Monitor to enter Low power mode and wake up.

**Actual behavior (Method C):**
The Monitor enters low power Mode on the Raspberry Pi 4, but does not wake back up. The Raspberry Pi 3 claims it is in low power mode, but the monitor does not turn off as intended.

**To reproduce (Method D):**
* power on Raspberry Pi
* wait for Desktop to appear.
* Open up terminal.
* run the command... ```xset dpms force suspend```
* wait about 5 to 15 minutes. The monitor should enter low power mode.
* Move mouse, and type on keyboard 

**Expected behavior (Method D):**
For the monitor to enter low power mode, and turn back on when your done.

**Actual behavior (Method D):**
The Monitor will enter low power mode on the Raspberry Pi 4, and fail to return. The Raspberry PI 3 claims to be in low power mode, but the monitor never enters low power mode.

**System**
Logs from Raspberry Pi 3: 
https://gist.githubusercontent.com/kphillisjr/070c538659e616992b5952885a044f18/raw/9ac299f3f2116c35598819c211bf0d2d3512c633/Raspinfo.rpi3b.txt 

Logs from Raspberry Pi 4: 
https://gist.githubusercontent.com/kphillisjr/070c538659e616992b5952885a044f18/raw/9ac299f3f2116c35598819c211bf0d2d3512c633/Raspinfo.rpi4b.txt

**Note:** While not exactly related to this bug, I should probably mention that the Raspberry Pi 4 fails to show any splash screens before showing the desktop.

xgqfrms, 2023-04-10:
👀 watching...

pelwell, 2023-04-10:
Can you clarify something?

> The Monitor enters low power Mode on the Raspberry Pi 4, but does not wake back up. The Raspberry Pi 3 claims it is in low power mode, but the monitor does not turn off as intended.

To me this says that the monitor enters low power _and_ does not wake up _but_ does not turn off. That sounds contradictory - is it?

kphillisjr, 2023-04-16:
> Can you clarify something?
>
>
> To me this says that the monitor enters low power _and_ does not wake up _but_ does not turn off. That sounds contradictory - is it?

Yea, it is a bit confusing, so i will clarify the original statement...

Raspberry Pi 4 - I tell the dpms tool to place the monitor in low power mode. The monitor goes into low power mode, and will not leave low power mode no matter what.

Raspberry Pi 3 - The monitor does not enter low power mode when using the dpms tool. The tool will claim it is in low power mode, but clearly the monitor is not in low power mode.