raspberrypi/linux #4485: eth0 (bcmgenet): transmit queue timed out

atsies, 2021-07-28:
**Describe the bug**
eth0 (bcmgenet) Ethernet stops working when plugged-in and hangs until a next reboot.

**To reproduce**
Boot Raspi OS, plug in Ethernet cable, observe kernel output.

**Expected behaviour**
eth0 (bcmgenet) Ethernet is expected to function.

**Actual behaviour**
eth0 (bcmgenet) Ethernet stops working, TCP/IP networking on eth0 becomes non-functional.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B Rev 1.2

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

* Which firmware version (`vcgencmd version`)?
Apr 30 2021 13:45:52 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

* Which kernel version (`uname -a`)?
`Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1414 SMP Fri Apr 30 13:20:47 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50700 r8192 d23028 u81920
[    0.000000] pcpu-alloc: s50700 r8192 d23028 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=m.m.m.m vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x4000000t
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000017cc0000-0x000000001bcc0000] (64MB)
[... 326 lines cut ...]
[   18.955026]  r4:ffffe000
[   18.955045] [<c02273a4>] (irq_exit) from [<c0287540>] (__handle_domain_irq+0x70/0xc4)
[   18.955056]  r5:00000000 r4:c1094d10
[   18.955072] [<c02874d0>] (__handle_domain_irq) from [<c020135c>] (gic_handle_irq+0x90/0xa4)
[   18.955086]  r9:c193a000 r8:c1094d1c r7:c193bf30 r6:f081400c r5:f0814000 r4:c1205b3c
[   18.955101] [<c02012cc>] (gic_handle_irq) from [<c0200abc>] (__irq_svc+0x5c/0x7c)
[   18.955115] Exception stack(0xc193bf30 to 0xc193bf78)
[   18.955132] bf20:                                     00000000 0000c90c eff4f2c4 c021ac00
[   18.955147] bf40: ffffe000 c120509c c12050e4 00000008 00000001 c133ba3f c0e1edfc c193bf8c
[   18.955164] bf60: c193bf90 c193bf80 c02088c0 c02088c4 60000013 ffffffff
[   18.955181]  r9:c193a000 r8:00000001 r7:c193bf64 r6:ffffffff r5:60000013 r4:c02088c4
[   18.955202] [<c020887c>] (arch_cpu_idle) from [<c0b72ef0>] (default_idle_call+0x4c/0x118)
[   18.955220] [<c0b72ea4>] (default_idle_call) from [<c02587d4>] (do_idle+0x118/0x168)
[   18.955237] [<c02586bc>] (do_idle) from [<c0258af8>] (cpu_startup_entry+0x28/0x30)
[   18.955253]  r10:00000000 r9:410fd083 r8:00007000 r7:c13543a0 r6:30c0387d r5:00000003
[   18.955265]  r4:00000091 r3:60000093
[   18.955284] [<c0258ad0>] (cpu_startup_entry) from [<c020f7b8>] (secondary_start_kernel+0x16c/0x178)
[   18.955305] [<c020f64c>] (secondary_start_kernel) from [<002018ac>] (0x2018ac)
[   18.955317]  r5:00000000 r4:018ceac0
[   18.955338] ---[ end trace b47be6fb4ffe5036 ]---

**Additional context**
Add any other relevant context for the problem.
kern.log attached.
[kern.log](https://github.com/raspberrypi/linux/files/6891025/kern.log)

mdevaev, 2021-07-28:
Perhaps related: #3401

mdevaev, 2021-07-28:
@atsies please show `ifconfig` also. Maybe you have some unusual network configuration.

2all: the user informed me that the bug was reproduced on two different boards.

atsies, 2021-07-28:
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.166.156  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::68d7:f1ed:107b:9c79  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:a0:ab:a2  txqueuelen 1000  (Ethernet)
        RX packets 3608  bytes 406771 (397.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3  bytes 580 (580.0 B)
        TX errors 144  dropped 0 overruns 0  carrier 0  collisions 0

Interface is expected to obtain a DHCP lease from the server, but fails before that. Thus the obtained address is link-local.

atsies, 2021-07-28:
Transmission queue(s) remain locked, the number of TX-ERR increments:
Iface      MTU    RX-OK RX-ERR RX-DRP RX-OVR    TX-OK TX-ERR TX-DRP TX-OVR Flg
eth0      1500     8118      0          0             0               3          321        0            0            BMRU

atsies, 2021-07-28:
ethtool -S eth0
NIC statistics:
     rx_packets: 13652
     tx_packets: 3
     rx_bytes: 1514002
     tx_bytes: 580
     rx_errors: 0
     tx_errors: 551
     rx_dropped: 0
     tx_dropped: 0
     multicast: 7922
     rx_64_octets: 67934
     rx_65_127_oct: 10143
     rx_128_255_oct: 1965
     rx_256_511_oct: 468
     rx_512_1023_oct: 0
     rx_1024_1518_oct: 0
     rx_vlan_1519_1522_oct: 0
     rx_1522_2047_oct: 0
     rx_2048_4095_oct: 0
     rx_4096_9216_oct: 0
     rx_pkts: 80510
     rx_bytes: 5860512
     rx_multicast: 74761
     rx_broadcast: 5730
     rx_fcs: 0
     rx_control: 65929
     rx_pause: 65929
     rx_unknown: 0
     rx_align: 0
     rx_outrange: 0
     rx_code: 0
     rx_carrier: 0
     rx_oversize: 0
     rx_jabber: 0
     rx_mtu_err: 0
     rx_good_pkts: 80510
     rx_unicast: 19
     rx_ppp: 0
     rx_crc: 0
     tx_64_octets: 0
     tx_65_127_oct: 0
     tx_128_255_oct: 0
     tx_256_511_oct: 0
     tx_512_1023_oct: 0
     tx_1024_1518_oct: 0
     tx_vlan_1519_1522_oct: 0
     tx_1522_2047_oct: 0
     tx_2048_4095_oct: 0
     tx_4096_9216_oct: 0
     tx_pkts: 0
     tx_multicast: 0
     tx_broadcast: 0
     tx_pause: 0
     tx_control: 0
     tx_fcs_err: 0
     tx_oversize: 0
     tx_defer: 0
     tx_excess_defer: 0
     tx_single_col: 0
     tx_multi_col: 0
     tx_late_col: 0
     tx_excess_col: 0
     tx_frags: 0
     tx_total_col: 0
     tx_jabber: 0
     tx_bytes: 0
     tx_good_pkts: 0
     tx_unicast: 0
     rx_runt_pkts: 0
     rx_runt_valid_fcs: 0
     rx_runt_inval_fcs_align: 0
     rx_runt_bytes: 0
     rbuf_ovflow_cnt: 0
     rbuf_err_cnt: 0
     mdf_err_cnt: 4743
     alloc_rx_buff_failed: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     tx_realloc_tsb: 0
     tx_realloc_tsb_failed: 0
     txq0_packets: 0
     txq0_bytes: 0
     rxq0_bytes: 0
     rxq0_packets: 0
     rxq0_errors: 0
     rxq0_dropped: 0
     txq1_packets: 0
     txq1_bytes: 0
     rxq1_bytes: 0
     rxq1_packets: 0
     rxq1_errors: 0
     rxq1_dropped: 0
     txq2_packets: 1
     txq2_bytes: 384
     rxq2_bytes: 0
     rxq2_packets: 0
     rxq2_errors: 0
     rxq2_dropped: 0
     txq3_packets: 1
     txq3_bytes: 86
     rxq3_bytes: 0
     rxq3_packets: 0
     rxq3_errors: 0
     rxq3_dropped: 0
     txq16_packets: 1
     txq16_bytes: 110
     rxq16_bytes: 1514002
     rxq16_packets: 13652
     rxq16_errors: 0
     rxq16_dropped: 0

atsies, 2021-08-30:
Ordered a 3'rd RPi4 B rev. 1.2 . The same issue.
It looks like the network traffic is the cause of the problem. After switching to another network the RPi ethernet stopped hanging.
On switching back to the 1st network, it hangs again regularly. Unfortunately, I am not yet able to record the traffic on Ethernet port to see any anomalies.

graysky2, 2021-12-10:
I am experiencing this on RPi4B running OpenWRT which does not use the RPi Foundation kernel, it is mainline.  So far, I have only seen this on 5.10.84.  I downgraded to 5.10.83 and am waiting to see if it pops up.

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.84 (squishy@quadruple) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r0+18161-8831bfab8a) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed Dec 8 11:15:57 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45272 r8192 d32552 u86016
[    0.000000] pcpu-alloc: s45272 r8192 d32552 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:02:C1:C1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 1836040K/2019328K available (7486K kernel code, 870K rwdata, 1904K rodata, 448K init, 532K bss, 117752K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2d0/0x454 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000181] Console: colour dummy device 80x25
[    0.000473] printk: console [tty1] enabled
[    0.000509] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000535] pid_max: default: 32768 minimum: 301
[    0.000701] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000745] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001389] cgroup: Disabling memory control group subsystem
[    0.002329] rcu: Hierarchical SRCU implementation.
[    0.002499] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003004] smp: Bringing up secondary CPUs ...
[    0.003418] Detected PIPT I-cache on CPU1
[    0.003483] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003965] Detected PIPT I-cache on CPU2
[    0.004005] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004462] Detected PIPT I-cache on CPU3
[    0.004502] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004577] smp: Brought up 1 node, 4 CPUs
[    0.004634] SMP: Total of 4 processors activated.
[    0.004649] CPU features: detected: 32-bit EL0 Support
[    0.004663] CPU features: detected: CRC32 instructions
[    0.004733] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004751] CPU: All CPU(s) started at EL2
[    0.004782] alternatives: patching kernel code
[    0.012880] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 228 lines cut ...]
[ 2688.241492] x13: 0000000000000116 x12: ffffffc010b43a48 
[ 2688.246806] x11: ffffffc010a1f790 x10: 00000000fffff000 
[ 2688.252120] x9 : ffffffc010a1f790 x8 : 0000000000000000 
[ 2688.257434] x7 : ffffffc0109c7790 x6 : 0000000000000008 
[ 2688.262748] x5 : 0000000000000003 x4 : 0000000000000000 
[... cut ...]
[ 2688.293688]  run_timer_softirq+0x3c/0x74
[ 2688.297609]  _stext+0x124/0x290
[ 2688.300751]  irq_exit+0xa0/0xf0
[ 2688.303893]  __handle_domain_irq+0x80/0xe0
[ 2688.307992]  gic_handle_irq+0x78/0xa0
[ 2688.311652]  el1_irq+0xcc/0x180
[ 2688.314793]  arch_cpu_idle+0x18/0x30
[ 2688.318369]  default_idle_call+0x20/0x70
[ 2688.322292]  do_idle+0x21c/0x280
[ 2688.325518]  cpu_startup_entry+0x28/0x60
[ 2688.329440]  secondary_start_kernel+0x128/0x150
[ 2688.333968] ---[ end trace ddbc004725827c20 ]---

```

johnp789, 2022-01-25:
This happened last night on my 2GB Pi 4 Model B running an Arch Linux ARM aarch64 kernel.  This Pi 4B serves as a router, with the eth0 interface serving as the "LAN" port and connected to an unmanaged gigabit switch along with a MacBook Pro and a Raspberry Pi 3 Model B.  The "WAN" port is a USB 3.0 GbE interface.

```
$ uname -a
Linux pi4-1 5.11.4-1-ARCH #1 SMP Sun Mar 7 23:46:10 UTC 2021 aarch64 GNU/Linux

$ uptime
 01:15:59 up 47 days, 10:56,  1 user,  load average: 0.00, 0.01, 0.00

$ ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.16.1  netmask 255.255.252.0  broadcast 192.168.19.255
        inet6 fe80::dea6:aaaa:aaaa:aaaa  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:aa:aa:aa:aa  txqueuelen 1000  (Ethernet)
        RX packets 36926543  bytes 12922639839 (12.0 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 58183599  bytes 52461760482 (48.8 GiB)
        TX errors 6097  dropped 0 overruns 0  carrier 0  collisions 0

[4043938.220080] pi4-1 kernel: ------------[ cut here ]------------
[4043938.224979] pi4-1 kernel: NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 1 timed out
[4043938.232009] pi4-1 kernel: WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:442 dev_watchdog+0x310/0x320
[4043938.240584] pi4-1 kernel: Modules linked in: nft_fib_ipv6 nft_fib_ipv4 nft_fib overlay nft_masq nft_nat nft_chain_nat nf_nat nf_log_ipv6 nf_log_ipv4 nf_log_common nft_log nft_reject_inet nf_reject_ip>[4043938.306760] pi4-1 kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.11.4-1-ARCH #1
[4043938.314967] pi4-1 kernel: Hardware name: Raspberry Pi 4 Model B (DT)
[4043938.320349] pi4-1 kernel: pstate: 20000005 (nzCv daif -PAN -UAO -TCO BTYPE=--)
[4043938.326615] pi4-1 kernel: pc : dev_watchdog+0x310/0x320
[4043938.330854] pi4-1 kernel: lr : dev_watchdog+0x310/0x320
[4043938.335092] pi4-1 kernel: sp : ffff800010003d40
[4043938.338623] pi4-1 kernel: x29: ffff800010003d40 x28: 0000000000000004 
[4043938.344186] pi4-1 kernel: x27: 0000000000000000 x26: 0000000000000001 
[4043938.349748] pi4-1 kernel: x25: 0000000000000140 x24: 00000000ffffffff 
[4043938.355309] pi4-1 kernel: x23: 0000000000000000 x22: ffff000042698480 
[4043938.360871] pi4-1 kernel: x21: ffff800011d37000 x20: ffff000042698000 
[4043938.366433] pi4-1 kernel: x19: 0000000000000001 x18: 0000000000000030 
[4043938.371994] pi4-1 kernel: x17: 0000000000000000 x16: 0000000000000000 
[4043938.377556] pi4-1 kernel: x15: ffff800011d45ff0 x14: ffff800011d59670 
[4043938.383118] pi4-1 kernel: x13: 0000000000000000 x12: 0000000000000942 
[4043938.388679] pi4-1 kernel: x11: 0000000000000316 x10: ffff800011e09670 
[4043938.394241] pi4-1 kernel: x9 : 00000000ffffe000 x8 : 80000000ffffe000 
[4043938.399803] pi4-1 kernel: x7 : ffff800011d59670 x6 : 0000000000000001 
[4043938.405365] pi4-1 kernel: x5 : 0000000000000000 x4 : 0000000000000000 
[4043938.410926] pi4-1 kernel: x3 : ffff800011d3f310 x2 : 0000000000000004 
[4043938.416488] pi4-1 kernel: x1 : 56de5a16e8fc0800 x0 : 0000000000000000 
[4043938.422050] pi4-1 kernel: Call trace:
[4043938.424700] pi4-1 kernel:  dev_watchdog+0x310/0x320
[4043938.428588] pi4-1 kernel:  call_timer_fn.constprop.0+0x24/0x80
[4043938.433445] pi4-1 kernel:  expire_timers+0xe4/0x10c
[4043938.437331] pi4-1 kernel:  __run_timers.part.0+0x1ac/0x1f0
[4043938.441833] pi4-1 kernel:  run_timer_softirq+0x3c/0x74
[4043938.445983] pi4-1 kernel:  _stext+0x124/0x2ac
[4043938.449340] pi4-1 kernel:  irq_exit+0xdc/0xfc
[4043938.452699] pi4-1 kernel:  __handle_domain_irq+0x80/0xe0
[4043938.457026] pi4-1 kernel:  gic_handle_irq+0x6c/0xa0
[4043938.460914] pi4-1 kernel:  el1_irq+0xb4/0x140
[4043938.464270] pi4-1 kernel:  arch_cpu_idle+0x18/0x30
[4043938.468069] pi4-1 kernel:  default_idle_call+0x24/0x6c
[4043938.472218] pi4-1 kernel:  cpuidle_idle_call+0x158/0x1ac
[4043938.476547] pi4-1 kernel:  do_idle+0xa8/0x100
[4043938.479905] pi4-1 kernel:  cpu_startup_entry+0x24/0x90
[4043938.484056] pi4-1 kernel:  rest_init+0xbc/0xcc
[4043938.487501] pi4-1 kernel:  arch_call_rest_init+0x10/0x1c
[4043938.491831] pi4-1 kernel:  start_kernel+0x3f0/0x424
[4043938.495718] pi4-1 kernel: ---[ end trace 753ccc04bb8fd67a ]---
```
This was followed by many lines about the tx ring 0 full.
```
[4044005.261046] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
[4044007.261080] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
[4044009.261145] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
[4044011.261144] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
[4044013.261174] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
```
Much later, the last of these messages around the time I disconnected and reconnected the MacBook Pro.  After doing this, without any intervention on the Pi 4B, traffic was able to flow again.  
```
[4056163.431011] pi4-1 kernel: bcmgenet fd580000.ethernet eth0: bcmgenet_xmit: tx ring 0 full when queue 1 awake
```

graysky2, 2022-01-25:
@johnp789 - overclocking?

johnp789, 2022-01-25:
> @johnp789 - overclocking?

I'm not overclocking.  My `config.txt` has only `enable_uart=1` in it.

graysky2, 2022-01-25:
> I'm not overclocking. My config.txt has only enable_uart=1 in it.

Well, that was timely.  I sought to test out whether or not my overclock could be a reason this bug is manifesting.  After running without it (stock) for a while, the bug got triggered this morning for me.  The overclock was a red herring.

```
[14105.382014] ------------[ cut here ]------------
[14105.386662] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 0 timed out
[14105.393425] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:467 dev_watchdog+0x2c8/0x2d0
[14105.401688] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat brcmfmac xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat nf_flow_table nf_conntrack libchacha20poly1305 libblake2s ipt_REJECT chacha_neon cfg80211 cdc_ncm cdc_ether ax88179_178a xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet usbhid slhc sch_cake rtl8150 r8152 poly1305_neon nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libchacha libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hid_generic crc_ccitt compat brcmutil br_netfilter sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[14105.401953]  act_mirred act_gact snd_bcm2835(C) hid evdev nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel veth snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 vfat fat nls_iso8859_1 nls_cp437
[14105.520204] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.10.93 #0
[14105.527597] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[14105.533429] pstate: 60400005 (nZCv daif +PAN -UAO -TCO BTYPE=--)
[14105.539433] pc : dev_watchdog+0x2c8/0x2d0
[14105.543439] lr : dev_watchdog+0x2c8/0x2d0
[14105.547443] sp : ffffffc010b2bd70
[14105.550753] x29: ffffffc010b2bd70 x28: 0000000000000140 
[14105.556066] x27: 00000000ffffffff x26: 0000000000000000 
[14105.561378] x25: 0000000000000000 x24: 0000000000000000 
[14105.566689] x23: ffffff8040c28480 x22: 0000000000000001 
[14105.572001] x21: ffffffc0109b6000 x20: ffffff8040c28000 
[14105.577312] x19: 0000000000000000 x18: 0000000000000132 
[14105.582623] x17: 00004fb9fe4107dc x16: 00004a2c97c7a9b6 
[14105.587935] x15: ffffffc0109c7890 x14: 0000000000000396 
[14105.593246] x13: 0000000000000132 x12: ffffffc010b2ba48 
[14105.598557] x11: ffffffc010a1f890 x10: 00000000fffff000 
[14105.603869] x9 : ffffffc010a1f890 x8 : 0000000000000000 
[14105.609180] x7 : ffffffc0109c7890 x6 : 0000000000000001 
[14105.614491] x5 : 0000000000000000 x4 : 0000000000000000 
[14105.619802] x3 : ffffffc0109b8f8c x2 : 0000000000000004 
[14105.625113] x1 : 0000000000000005 x0 : 000000000000003c 
[14105.630425] Call trace:
[14105.632869]  dev_watchdog+0x2c8/0x2d0
[14105.636531]  call_timer_fn.constprop.0+0x24/0x80
[14105.641145]  __run_timers.part.0+0x240/0x280
[14105.645411]  run_timer_softirq+0x3c/0x74
[14105.649330]  _stext+0x124/0x290
[14105.652470]  irq_exit+0xa0/0xf0
[14105.655608]  __handle_domain_irq+0x80/0xe0
[14105.659705]  gic_handle_irq+0x78/0xa0
[14105.663363]  el1_irq+0xcc/0x180
[14105.666502]  arch_cpu_idle+0x18/0x30
[14105.670076]  default_idle_call+0x20/0x70
[14105.673996]  do_idle+0x21c/0x280
[14105.677221]  cpu_startup_entry+0x24/0x60
[14105.681140]  rest_init+0xbc/0xcc
[14105.684363]  arch_call_rest_init+0x10/0x1c
[14105.688455]  start_kernel+0x438/0x454
[14105.692112] ---[ end trace 45fcfc017fd545ae ]---
```

Again, this is running at stock clock/voltage.

johnp789, 2022-01-26:
I caught only one such error at 47 days of uptime, so troubleshooting may be a very slow process!

graysky2, 2022-01-26:
> I caught only one such error at 47 days of uptime, so troubleshooting may be a very slow process!

I don't think so... you can trigger it using an `iperf3` test pounding the RPi4:
* On the RPi4, run it to listen: `iperf3 -s -f M`
* On a PC connected, run it to send: `iperf3 -n 10G -P 4 -c 10.1.10.1`

For me it only took 30 gigs or so to trigger the bug:
```
# dmesg
...
[   11.787446] r8152 2-1:1.0 eth1: carrier on
[  520.956114] ------------[ cut here ]------------
[  520.960747] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 0 timed out
[  520.967553] WARNING: CPU: 3 PID: 26 at net/sched/sch_generic.c:467 dev_watchdog+0x2c8/0x2d0
[  520.975898] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat brcmfmac xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat nf_flow_table nf_conntrack libchacha20poly1305 libblake2s ipt_REJECT chacha_neon cfg80211 cdc_ncm cdc_ether ax88179_178a xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet usbhid slhc sch_cake rtl8150 r8152 poly1305_neon nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libchacha libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hid_generic crc_ccitt compat brcmutil br_netfilter sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[  520.976044]  act_mirred act_gact snd_bcm2835(C) hid evdev nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel veth snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 vfat fat nls_iso8859_1 nls_cp437
[  521.094218] CPU: 3 PID: 26 Comm: ksoftirqd/3 Tainted: G         C        5.10.93 #0
[  521.101866] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  521.107692] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[  521.113691] pc : dev_watchdog+0x2c8/0x2d0
[  521.117692] lr : dev_watchdog+0x2c8/0x2d0
[  521.121691] sp : ffffffc010c2bc00
[  521.124996] x29: ffffffc010c2bc00 x28: 0000000000000140 
[  521.130301] x27: 00000000ffffffff x26: 0000000000000000 
[  521.135606] x25: 0000000000000003 x24: 0000000000000000 
[  521.140911] x23: ffffff8040c28480 x22: 0000000000000001 
[  521.146215] x21: ffffffc0109b6000 x20: ffffff8040c28000 
[  521.151519] x19: 0000000000000000 x18: 0000000000000126 
[  521.156824] x17: 0000c3aa115ea6c6 x16: 0000c3aa115ea6c6 
[  521.162128] x15: ffffffc0109c7890 x14: 0000000000000372 
[  521.167432] x13: 0000000000000126 x12: ffffffc010c2b8d8 
[  521.172737] x11: ffffffc010a1f890 x10: 00000000fffff000 
[  521.178041] x9 : ffffffc010a1f890 x8 : 0000000000000000 
[  521.183346] x7 : ffffffc0109c7890 x6 : 0000000000000001 
[  521.188650] x5 : 0000000000000000 x4 : 0000000000000000 
[  521.193954] x3 : 0000000000000000 x2 : ffffff807fbd4068 
[  521.199258] x1 : ffffffc06f22d000 x0 : 000000000000003c 
[  521.204564] Call trace:
[  521.207002]  dev_watchdog+0x2c8/0x2d0
[  521.210659]  call_timer_fn.constprop.0+0x24/0x80
[  521.215267]  __run_timers.part.0+0x240/0x280
[  521.219528]  run_timer_softirq+0x3c/0x74
[  521.223443]  _stext+0x124/0x290
[  521.226577]  run_ksoftirqd+0x54/0x70
[  521.230146]  smpboot_thread_fn+0x154/0x1a0
[  521.234235]  kthread+0x124/0x12c
[  521.237455]  ret_from_fork+0x10/0x34
[  521.241021] ---[ end trace 3907dc14c3eab63e ]---
```

graysky2, 2022-02-17:
@johnp789 - Can you try the iperf3 test and see if you get the same?

graysky2, 2022-02-18:
I confirmed this on a 2nd, newer RPi4B (hardware revision 1.4).
```
[  265.029040] ------------[ cut here ]------------
[  265.033684] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 0 timed out
[  265.040451] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:467 dev_watchdog+0x2c8/0x2d0
[  265.048706] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet brcmfmac wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack libchacha20poly1305 libblake2s iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211 cdc_ncm cdc_ether ax88179_178a xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables usbnet usbhid slhc sch_cake rtl8150 r8152 poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha libblake2s_generic hid_generic crc_ccitt compat
[  265.048835]  brcmutil sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact snd_bcm2835(C) hid evdev ifb ip6_udp_tunnel udp_tunnel veth snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 vfat fat nls_iso8859_1 nls_cp437
[  265.169349] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C        5.10.101 #0
[  265.176821] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  265.182646] pstate: 60400005 (nZCv daif +PAN -UAO -TCO BTYPE=--)
[  265.188643] pc : dev_watchdog+0x2c8/0x2d0
[  265.192643] lr : dev_watchdog+0x2c8/0x2d0
[  265.196642] sp : ffffffc010b33d70
[  265.199946] x29: ffffffc010b33d70 x28: 0000000000000140 
[  265.205250] x27: 00000000ffffffff x26: 0000000000000000 
[  265.210554] x25: 0000000000000001 x24: 0000000000000000 
[  265.215858] x23: ffffff8040c40480 x22: 0000000000000001 
[  265.221161] x21: ffffffc0109b6000 x20: ffffff8040c40000 
[  265.226464] x19: 0000000000000000 x18: 000000000000011f 
[  265.231767] x17: 0000000000000000 x16: 000000000000000c 
[  265.237070] x15: ffffffc0109c77d0 x14: 000000000000035d 
[  265.242376] x13: 000000000000011f x12: ffffffc010b33a48 
[  265.247687] x11: ffffffc010a1f7d0 x10: 00000000fffff000 
[  265.252999] x9 : ffffffc010a1f7d0 x8 : 0000000000000000 
[  265.258310] x7 : ffffffc0109c77d0 x6 : 0000000000000002 
[  265.263620] x5 : 0000000000000000 x4 : 0000000000000000 
[  265.268931] x3 : ffffffc0109b8f8c x2 : 0000000000000004 
[  265.274242] x1 : 0000000000000005 x0 : 000000000000003c 
[  265.279554] Call trace:
[  265.281998]  dev_watchdog+0x2c8/0x2d0
[  265.285662]  call_timer_fn.constprop.0+0x24/0x80
[  265.290276]  __run_timers.part.0+0x240/0x280
[  265.294543]  run_timer_softirq+0x3c/0x74
[  265.298462]  _stext+0x124/0x290
[  265.301602]  irq_exit+0xa0/0xf0
[  265.304742]  __handle_domain_irq+0x80/0xe0
[  265.308840]  gic_handle_irq+0x78/0xa0
[  265.312498]  el1_irq+0xcc/0x180
[  265.315637]  arch_cpu_idle+0x18/0x30
[  265.319210]  default_idle_call+0x20/0x70
[  265.323132]  do_idle+0x21c/0x280
[  265.326356]  cpu_startup_entry+0x28/0x60
[  265.330277]  secondary_start_kernel+0x128/0x150
[  265.334802] ---[ end trace 6a3a60879d0d2b7c ]---
```

johnp789, 2022-02-21:
I've done an update in the meantime.  

```
$ uname -a
Linux pi4-e8d2e7ed 5.16.3-1-aarch64-ARCH #1 SMP Thu Jan 27 16:41:47 UTC 2022 aarch64 GNU/Linux
```

I ran `iperf3 -n 100G -c $PI4_IP` twice, back to back, and there were no `NETDEV WATCHDOG` errors logged.

graysky2, 2022-02-23:
@wtschueller - See the work-around @wulfy23 proposed [here](https://github.com/openwrt/openwrt/issues/9305#issuecomment-1047344495).  Does it work for you?

wtschueller, 2022-02-23:
@graysky2 I do not see checksum errors in dmesg. See below.

Here is what I have tried already:

**kernel 5.10.17-v8 aarch64:**

Basically same behaviour. Wireshark shows some transfers from RPi4 to PC, PC sends ack and that's it. PC repeats the ack several times an then give up. RPi4 does not even respond to arp requests anymore.

However, I need to `ethtool -K eth0 tx off` because hardware checksumming seems to corrupt outgoing jumbo frames. It must be switched on by default somewhere between 5.4 and 5.10, could be [this commit](https://github.com/raspberrypi/linux/commit/ae895c49905cd99aca24d23361b72ed559b30f4f).

Then, I switch off everything `ethtool -k` advertises (including rx crc). No change.

Reducing txqueuelen with `ifconfig` down to 1 lets iperf3 survive a few seconds at MTU=2036.
Could be because throughput also gets reduced.

Disabling IPv6 in cmdline.txt, no change.

Changing number of txqueues:
```
bcmgenet_hw_params[] = {
...
[GENET_V5] = {
    .tx_queues = 0,
    .tx_bds_per_q = 0,
```
no change.

Benchmarking with udp traffic is less clear. Especially 8k packets seem to work for longer time. Less cpu load could be one reason.

**kernel 5.16.10-v8 aarch64:**

Done the sweep test only. Could not see dmesg updates because I compiled the kernel image only, no modules. Therefore, no wlan0 and eth0 got stuck. I would assume same behaviour.

wulfy23, 2022-02-24:
cheers... afaik rc checksumming off is effective for the 'tailroom' failure mode...
(sparse connectionless rcu hang or whatever is going on there)

this iperf stress testing must be triggering another codepath or whatnot... but afaik... it looks like a better place to troubleshoot from...

wtschueller, 2022-03-02:
I found an even faster way to trigger the transmit queue time out: UDP packet sweeping like this

```
for ((MTU=2024; MTU>8192; ((MTU=MTU-4)) )) do sudo ifconfig eth0 mtu ${MTU} ; ifconfig eth0 ; dd if=/dev/zero bs=65536 count=10 |netcat -u 169.254.1.1 65000 -q 1; done
```

Data is sent in chunks of 8k and fragmented according to MTU. I added logging to see what will go on.

```
[  +0,000391] bcmgenet fd580000.ethernet: bcmgenet_isr0: full_status=0x0007c000, masked=0x00010000, dma_ctrl=0x00020001
[  +0,000012] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1234 c_index=1235 txbds_ready=1
[  +0,000005] bcmgenet fd580000.ethernet: ring.cbs[210].dma_len=2106, dma_addr=0x0000000110b86002
[  +0,000145] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   1, free_bds=255->256
[  +0,000009] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1235 c_index=1236 txbds_ready=1
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[211].dma_len=2106, dma_addr=0x0000000110b87002
[  +0,000140] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   1, free_bds=254->255
[  +0,000009] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1236 c_index=1239 txbds_ready=3
[  +0,000005] bcmgenet fd580000.ethernet: ring.cbs[212].dma_len=2106, dma_addr=0x0000000110b85002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[213].dma_len=2106, dma_addr=0x0000000110b84002
[  +0,000140] bcmgenet fd580000.ethernet: ring.cbs[214].dma_len= 266, dma_addr=0x0000000103a15802
[  +0,000007] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   3, free_bds=253->256
[  +0,000009] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1239 c_index=1239 txbds_ready=0
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   0, free_bds=256->256
[  +0,188908] bcmgenet fd580000.ethernet: bcmgenet_isr0: full_status=0x0086c000, masked=0x00800000, dma_ctrl=0x00020001
[  +0,000099] bcmgenet fd580000.ethernet: bcmgenet_isr0: full_status=0x0086c000, masked=0x00800000, dma_ctrl=0x00020001
[  +0,835193] bcmgenet fd580000.ethernet: bcmgenet_isr0: full_status=0x0007c000, masked=0x00010000, dma_ctrl=0x00020001
[  +0,000022] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1239 c_index=1243 txbds_ready=4
[  +0,000007] bcmgenet fd580000.ethernet: ring.cbs[215].dma_len=2114, dma_addr=0x0000000110b84002
[  +0,000013] bcmgenet fd580000.ethernet: ring.cbs[216].dma_len=2114, dma_addr=0x0000000110b85002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[217].dma_len=2114, dma_addr=0x0000000110b87002
[  +0,000007] bcmgenet fd580000.ethernet: ring.cbs[218].dma_len=2114, dma_addr=0x0000000110b86002
[  +0,000007] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   4, free_bds=251->255
[  +0,000010] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1243 c_index=1245 txbds_ready=2
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[219].dma_len= 234, dma_addr=0x0000000103a15002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[220].dma_len=2114, dma_addr=0x0000000112fb1002
[  +0,000008] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   2, free_bds=253->255
[  +0,000008] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1245 c_index=1247 txbds_ready=2
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[221].dma_len=2114, dma_addr=0x000000011cee6002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[222].dma_len=2114, dma_addr=0x000000011cee4002
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   2, free_bds=253->255
[  +0,000009] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1247 c_index=1249 txbds_ready=2
[  +0,000005] bcmgenet fd580000.ethernet: ring.cbs[223].dma_len=2114, dma_addr=0x000000011cee5002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[224].dma_len= 234, dma_addr=0x0000000103a14802
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   2, free_bds=254->256
[  +0,000008] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1249 c_index=1249 txbds_ready=0
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   0, free_bds=256->256
[  +0,000008] bcmgenet fd580000.ethernet: bcmgenet_isr0: full_status=0x0007c000, masked=0x00010000, dma_ctrl=0x00020001
[  +0,000009] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1249 c_index=1251 txbds_ready=2
[  +0,000005] bcmgenet fd580000.ethernet: ring.cbs[225].dma_len=2114, dma_addr=0x000000011cee7002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[226].dma_len=2114, dma_addr=0x000000011cee5002
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   2, free_bds=253->255
[  +0,000008] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1251 c_index=1253 txbds_ready=2
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[227].dma_len=2114, dma_addr=0x000000011cee4002
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[228].dma_len=2114, dma_addr=0x000000011cee6002
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   2, free_bds=253->255
[  +0,000008] bcmgenet fd580000.ethernet: __bcmgenet_tx_reclaim ring=16 old_c_index=1253 c_index=1254 txbds_ready=1
[  +0,000006] bcmgenet fd580000.ethernet: ring.cbs[229].dma_len= 234, dma_addr=0x0000000103a14402
[  +0,000006] bcmgenet fd580000.ethernet: bcmgenet_tx_poll work_done=   1, free_bds=255->256
[... cut ...]
Surprisingly, the time out occurs right at the very first packet of size 2050+64=2114 (`ring.cbs[113].dma_len=2114`).
This time it was received by wireshark.

Reverting parts of [net: bcmgenet: Implement TX coalescing control knobs](https://github.com/raspberrypi/linux/commit/2f9130709d2c19e2eef336bda0fb0c807cbc1f64) and [bcmgenet: Better coalescing parameter defaults](https://github.com/raspberrypi/linux/commit/6482fde0f4e2331f5b2511602d055e49633236d5) does not help (nor changing the coalesce setting using `ethtool`.

wtschueller, 2022-03-16:
@pelwell - Could you have a look please? I saw your commit [phy: broadcom: Add bcm54213pe configuration](https://github.com/raspberrypi/linux/commit/a49a9425133aac8c2995ff0eeead99ef868863af). There is a function [bcm_phy_enable_jumbo()](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/net/phy/bcm-phy-lib.c#L612), so I guess there could be additional settings for stable use of jumbo frames on RPi4.

pelwell, 2022-03-17:
Adding a call to existing function for a different but compatible PHY requires very little knowledge (provided it works, which it did).

wtschueller, 2022-03-17:
@pelwell Thank you. Ok, I blindly assumed that you have access to the documentation. Maybe you could point me in the right direction.

I hope that if we know what goes wrong than we can develop a better time out handler that brings eth0 back to life.
As a side effect, I get jumbo frames stable.

wtschueller, 2022-04-25:
Still no real progress. 06d836801cd82ded282aaf9e888ff9e7e4a88b91 does not help in my case.

segabor, 2022-09-04:
Hi there,
My RPi 4 with the latest kernel `5.15.0-1013-raspi` became totally unusable. Logs are full of `bcmgenet_txmit: tx ring 0 full when queue 1 awake`
I have to turn the switch off and on to make it work for a couple of minutes.
I'm going to revert to an older kernel hoping it brings back stability.

mjasny, 2022-09-28:
Try to disable the eee setting on eth0 (“Energy Efficient Ethernet”). After I disabled I'm not experiencing these tx ring errors.

`sudo ethtool --set-eee eth0 eee off`

`sudo ethtool --show-eee eth0`

segabor, 2022-09-28:
Thanks @mjasny, I'll give it a try

zenlord, 2023-03-14:
There have been reports that the workaround using ethtool is reverted automatically and thus disables the workaround.

The devs have created a new config.txt parameter to switch off EEE in the driver. More info: https://github.com/raspberrypi/linux/issues/4289#issuecomment-1351697589