raspberrypi/linux #4486: KMS: blank screen after TV is brought out of standby on modes that need scrambling

HiassofT, 2021-07-29:
**Describe the bug**
When running modetest or kodi with a mode that needs scrambling (eg 3840x2160-50/60) my LG 55C8 reports no signal after switching it into standby and back on.

**To reproduce**
- add `hdmi_enable_4kp60=1` to config.txt, boot into RPiOS lite framebuffer console
- run `modetest -M vc4 -s 32:3840x2160-50`
- put TV into standby and back on

**Expected behaviour**
TV shows picture

**Actual behaviour**
TV displays "no signal"

**System**
RPiOS lite with rpi-update kernel/firmware https://github.com/Hexxeh/rpi-firmware/commit/b986fda889737f6ce0a4acda89d447f78d0a1a73
```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux
root@raspberrypi:~# vcgencmd version
Jul 21 2021 16:21:46
Copyright (c) 2012 Broadcom
version 6a796bb0062a6c75191c57cba1c13f9300076d02 (clean) (release) (start)
root@raspberrypi:~# cat /boot/config.txt
# Hias

dtoverlay=disable-bt
dtoverlay=vc4-kms-v3d
hdmi_enable_4kp60=1
#disable_fw_kms_setup=1
#hdmi_force_hotplug=1
```

Also verified with current rpi-5.10.y commit 6495fa3396fb387d3871ced42a30281fe7be5352 plus #4371

**Logs**
dmesg from rpi-update kernel: http://ix.io/3uoI
dmesg from rpi-5.10.y plus #4371: http://ix.io/3uoJ

**Additional context**
Manually initializing the scrambler with `i2cset -y 20 0x54 0x20 0x3` when TV shows "no signal" brings the signal back.

Signal also comes back when exiting modetest and it switches from 3840x2160-50 back to 3840x2160-60

We also got a bug report of a LibreELEC user where he noticed the same with a Panasonic AX800 TV: https://forum.libreelec.tv/thread/24279-pi4-no-screen-after-turning-tv-off-and-on-in-4k-50hz-or-4k-60hz/

ping @mripard

popcornmix, 2021-07-29:
Wasn't this reported as fixed with https://github.com/raspberrypi/linux/pull/4412 ?
Was it fixed then, and has subsequently regressed, or was it not actually fixed then?

HiassofT, 2021-07-29:
I successfully tested it back then (IIRC it worked only in combination with the "firmware-display-done" KMS commit though) but can't get it to work anymore, even with some older versions which I think I used for testing back then.

Not 100% sure, could be a race / timing issue and me just being lucky back then or some fault on my side when testing (can't think of which ATM tough).

mripard, 2021-08-19:
I just pushed a branch that fixes this issue on my LG TV

PR #4538 (2021-08-20): HDMI Hotplug Detection Fix
Hi,

Here's a fix for a hotplug detection issue that ultimately resulted in a black screen when using a mode using the scrambler, an application that doesn't handle hotplug events and an hotplug event.

Fixes #4486
closed 2021-08-20 by pelwell