raspberrypi/linux #4488: SPI Ethernet (enc28j60) strange MTU behavior when added to a bridge

AndysRobotics, 2021-07-29:
**Describe the bug**
When adding enc28j60 to a pi and creating a bridge, their appears to be a discrepancy of 4 bytes when traffic is crossing the bridge and SPI eth port. This causes full size packets to get dropped that are traversing the bridge due to being oversized.
This is only an issue when the SPI ethernet port is added to a bridge, when running independently it works as expected.

**To reproduce**
Hardware required
- 1x Raspberry pi for host device
- 2x Raspberry pi's for client devices (or 2 PCs)
- 1x USB ethernet port
- 1x ENC28J60 hat/module

Using the image (2021-05-07-raspios-buster-armhf-lite) enable enc28j60 driver and wire into SPI port.
Add USB eth port (such as Rankie gigabit ethernet adaptor) to help test bridge

To enable enc28j60 add following to `/boot/config.txt`
```
[pi4]
dtoverlay=spi0-1cs,cs0_pin=8
dtoverlay=enc28j60,int_pin=22,speed=12500000
[all]
```

To enable bridge `sudo apt-get install bridge-utils` and add following to `/etc/network/interfaces.d/br0`
```
auto br0
iface br0 inet static
        bridge_ports eth0 eth1 eth2
        bridge_stp off
        bridge_waitport 0
        bridge_fd 0
        address 10.0.1.106
        netmask 255.255.255.0
        gateway 10.0.1.1
```

Disable DHCP client on all ports add `denyinterfaces eth0 eth1 eth2` to top of `/etc/dhcpcd.conf` to prevent ethernet interfaces obtaining an address. Then reboot pi.

Connect host pi (with bridge and additional ports) to network via built in port.
Add second pi (client) with default image and dhcp to USB ethernet port.
Add third pi (client) with default image and dhcp to SPI ethernet port.

Ping all devices from another device connected to same network with varying packet sizes (1468, 1472, 1473) with don't defrag set. `ping 10.0.1.xxx -t -l 1468 -f` from windows or `ping 10.0.1.xxx -s 1468 -M do` from linux

**Expected behaviour**
Host to have static address assigned to bridge br0. Clients to pick up DHCP.
All devices to respond with correct sizes when 1468 or 1472 sized packet is sent and error with fragment warning when 1473 bytes sent.

**Actual behaviour**
Host pi and the client pi on USB port both behave as expected.
Pi on SPI port gets request timed out on 1469-1472 bytes and then fragment warning when 1473 bytes sent
This causes anything that needs to transfer larger than 1468 bytes at once to fail, so sftp and web hosting with larger files to fail from the pi connected to the SPI port.

**Additional context**
On the host pi, eth0 is built in port, eth1 is USB port and eth2 is SPI port.
Swapping the two clients around keeps the issue with the SPI port only. The host can ping the client on the SPI port at 1472 and the client on the SPI port can ping the host at 1472, the issue only occurs when the client has to cross the bridge.

Setting the MTU on the client connected in the SPI port to 1496 allows pings up to 1468 and will warn about ping 1469 and larger. This indicates that path discovery is working but we have devices that cannot change their MTU so this isn't a solution.

Changing the MTU value (up or down) on the bridge interface or on any of the routes/interfaces on the host pi has no effect. This indicates the issue is with traffic inbound to the SPI ethernet port.

On our production we are using 2x enc28j60 on SPI0 and SPI1 so iptables or other forms of bridging is not available. The issue is present when using 1 or 2 enc28j60 modules.

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
[... 600 lines cut ...]
[    7.069481] 8021q: 802.1Q VLAN Support v1.8
[    7.284233] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.615972] enc28j60 spi0.0 eth2: promiscuous mode
[    7.684276] br0: port 2(eth1) entered disabled state
[    7.684646] br0: port 3(eth2) entered disabled state
[   10.794502] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   10.794567] br0: port 1(eth0) entered blocking state
[   10.794586] br0: port 1(eth0) entered forwarding state
[   10.794831] IPv6: ADDRCONF(NETDEV_CHANGE): br0: link becomes ready
[   13.751174] Bluetooth: Core ver 2.22
[   13.751288] NET: Registered protocol family 31
[   13.751304] Bluetooth: HCI device and connection manager initialized
[   13.751335] Bluetooth: HCI socket layer initialized
[   13.751358] Bluetooth: L2CAP socket layer initialized
[   13.751392] Bluetooth: SCO socket layer initialized
[   13.760931] Bluetooth: HCI UART driver ver 2.3
[   13.760954] Bluetooth: HCI UART protocol H4 registered
[   13.761074] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.761389] Bluetooth: HCI UART protocol Broadcom registered
[   13.938493] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.938505] Bluetooth: BNEP filters: protocol multicast
[   13.938520] Bluetooth: BNEP socket layer initialized
[   18.154199] Bluetooth: hci0: command 0x1003 tx timeout
[   20.234203] Bluetooth: hci0: command 0x1001 tx timeout
[   22.314204] Bluetooth: hci0: command 0x1009 tx timeout

AndysRobotics, 2021-08-02:
Further testing

Pinging from my PC (Windows) with wireshark has revealed that the bridge/SPI Ethernet is definitely adding 4 additional bytes to the payload.

Image has 3 wireshark traces. The client is moved from the SPI to the USB and the ping responses are different.
When pinging the host or the client in the USB ethernet port the ping response is 60 bytes (14 payload + ICMP/network frames) and ends with 4 lots of 0x00 0x00 0x00 0x00. However when the client is connected to the SPI ethernet port, there is 4 random bytes (different on every single ping response) appended to the end.

Also, if the network is connected to the SPI ethernet port, you can ping the host or client and not get the additional bytes of data.

Ping 
PC (10.0.1.20)->Network->SPIEth->Br0->Host (10.0.1.106) No extra bytes
PC (10.0.1.20)->Network->SPIEth->Br0->Client (10.0.1.135) No extra bytes
PC (10.0.1.20)->Network->PiGigEth->Br0->Host (10.0.1.106) No extra bytes
PC (10.0.1.20)->Network->PiGigEth->Br0->USBEth->Client (10.0.1.135) No extra bytes
PC (10.0.1.20)->Network->PiGigEth->Br0->SPIEth->Client (10.0.1.135) **4 extra bytes (random)**

The 4 extra bytes only seem to occur when traffic crosses the bridge from the SPIEth indicating a bug in the bridge, SPI driver or interaction between the 2.

These 4 bytes do not match any hardware or IP addresses and appear to be completely random between pings.
I have concerns these could be memory overflow and could potentially contain part of sensitive data.

[image]

AndysRobotics, 2021-08-04:
Wireshark shows the 4 bytes as Eth.FCS (frame check sequence - 4 byte CRC)

I have compiled the drive for the ENC28J60 and disabled the FCS on the physical chip - TXCRCEN and CRCEN flags in MACON3 and ERXFCON registers as per the ENC28J60 datasheet.

The behavior does not change with this change.

I have found this email archive with a similar issue (between WLAN and ETH)
https://lists.linuxfoundation.org/pipermail/bridge/2004-August/004159.html

When you ping the client from my PC with a packet size of 1468 the total size of the response is 1514. This is still 4 bytes short of the max Eth frame size of 1518 so the checksum being present is not causing the bug, if anything it looks like the fact that the checksum is not present on the other Ethernet ports is the only reason they are working. 

I am not sure if this is an issue with the way the ethernet driver presents itself to the kernel/bridge or with the bridge itself. I have no idea where to take this bug now?

pelwell, 2021-08-04:
I don't think this is a Pi-specific problem. You might have more luck emailing the Bridge list where you found the archived issue.

dArKsTr666, 2021-09-02:
#