raspberrypi/linux #4489: events_freezable mmc_rescan after apt upgrade

PeterPablo, 2021-07-29:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
After sudo apt full-upgrade a kernel panic occured after reboot.

**To reproduce**
Upgrade from to.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
solid operation like before

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Pi4B 2GB
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
`dmesg`
```
sudo dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.52-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1440 SMP Tue Jul 27 09:55:21 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000bc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002bffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000002bffffff]
[    0.000000] On node 0 totalpages: 180224
[    0.000000]   DMA zone: 1584 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 180224 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 178640
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@50,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:00:97:FD vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=66606d14-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x00000000075d0000-0x000000000b5d0000] (64MB)
[    0.000000] Memory: 103360K/720896K available (10240K kernel code, 1364K rwdata, 3188K rodata, 2048K init, 891K bss, 93248K reserved, 524288K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34303 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[... 628 lines cut ...]
[   15.670925] Bluetooth: HCI UART protocol Broadcom registered
[   15.892082] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.892091] Bluetooth: BNEP filters: protocol multicast
[   15.892104] Bluetooth: BNEP socket layer initialized
[   33.463384] Bluetooth: RFCOMM TTY layer initialized
[   33.463401] Bluetooth: RFCOMM socket layer initialized
[   33.463422] Bluetooth: RFCOMM ver 1.11
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.68.104  netmask 255.255.255.0  broadcast 192.168.68.255
        ether dc:a6:32:00:97:fe  txqueuelen 1000  (Ethernet)
        RX packets 336  bytes 81803 (79.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 427  bytes 100706 (98.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

throttled=0x0
 19:18:11 up 0 min,  1 user,  load average: 2.66, 0.61, 0.20
Linux horst 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux
temp=53.5'C
health
```

May be this was a one off after the upgrade, but if more people are affected this might warrant further analysis. You may close, if you deem the report not relevant.

Best regards.

popcornmix, 2021-07-29:
This is fine. Quite a bit has changed in the latest firmware/kernel update, so there could be something here.
If you can see this again (and ideally find how to reproduce it), then that would be useful.

I see you had an OOM failure with kodi running, and I believe you are running with kms driver?
We have changed kms driver to default to 512M rather than 256M of CMA.

If you are able to reproduce this issue, then using:
```
dtoverlay=vc4-kms-v3d,cma-256
```
rather than what I assume is `dtoverlay=vc4-kms-v3d` may be interesting.
Posting your config.txt and cmdline.txt is always useful in case there are other non-default settings.

PeterPablo, 2021-07-29:
`raspinfo` was not installed. If my memory serves me right, I upgraded the distribution from stretch a long time ago.

```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-fkms-v3d
gpu_mem=320
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
pi3-disable-bt

pi@horst:~ $ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=66606d14-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

I have only seen this issue once and this was right after the upgrade 30minutes ago. I will keep an eye on the situation. I have not wilfully played with kms vs. fkms, as the system was working very well for me. Right now I'm running buster 32bit of raspbian/Raspberry Pi OS.

PeterPablo, 2021-07-29:
... so I see quite some old cruft in my config.txt (possibly from the times where I had the SD card plugged in the previous Pi3B). If suggested, I can take out the dtoverlay=vc4-fkms-v3d, gpu_mem=320 settings and see how the system behaves.

popcornmix, 2021-07-29:
I see:
```
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000bc00000, size 512 MiB
```
which combined with `gpu_mem=320` doesn't leave a lot of space in the bottom 1GB.

You almost certainly don't want gpu_mem=320 on the 5.10 kernel.
hevc decoder and 3d don't use gpu_mem.
Only h264 decode and encode (from camera) use gpu_mem and they are generally happy with the default (76M).
Just remove the gpu_mem line.

PeterPablo, 2021-07-29:
Thank you for the fast turn-around!
First, I disabled both settings, so I'm more relying on the defaults:
```
# more RAM and fake KMS (for Kodi)
#dtoverlay=vc4-fkms-v3d
#gpu_mem=320
```
This made Kodi process to not come up properly, with `kodi.log` containing.
```
2021-07-29 22:01:46.439 T:2972888160   ERROR: CWinSystemGbm::InitWindowSystem - failed to initialize Atomic DRM
2021-07-29 22:01:47.431 T:2972888160   ERROR: CWinSystemGbm::InitWindowSystem - failed to initialize Legacy DRM
2021-07-29 22:01:48.318 T:2972888160   ERROR: CWinSystemGbm::InitWindowSystem - failed to initialize off screen DRM
2021-07-29 22:01:48.318 T:2972888160   FATAL: CApplication::Create: Unable to init windowing system
```.

As (implicitly) suggested above, I explicitly set the default (?) `dtoverlay=vc4-kms-v3d` and rebooted. Now the kodi process at least shows up in htop (and kodi.log does not contain these error/fatal messages). It seems to me that the idle CPU usage with `kms` is considerably lower compared to `fkms` (0%, where it used to consume some 20-30% while simply displaying the GUI or big pictures screensaver).
I have not yet checked that everything is working smoothly on the TV (will do so in due course), but assume so in good faith. I was merely connected from ssh. Thus I'm closing this issue. 
Thank you again!

PeterPablo, 2021-07-29:
After a few minutes of uptime I checked `dmesg` and it now contains for `dtoverlay=vc4-kms-v3d`:
```
pi@horst:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.52-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1440 SMP Tue Jul 27 09:55:21 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000003b3fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 46080 pages, LIFO batch:15
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240960
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@50,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:00:97:FD vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=66606d14-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000000ac00000-0x000000000ec00000] (64MB)
[    0.000000] Memory: 351020K/970752K available (10240K kernel code, 1364K rwdata, 3188K rodata, 2048K init, 891K bss, 95444K reserved, 524288K cma-reserved, 184320K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34303 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=0
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000097] bcm2835: system timer (irq = 26)
[    0.000760] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000782] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000805] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000823] Switching to timer-based delay loop, resolution 18ns
[    0.001087] Console: colour dummy device 80x30
[    0.001848] printk: console [tty1] enabled
[    0.001916] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001973] pid_max: default: 32768 minimum: 301
[    0.002152] LSM: Security Framework initializing
[    0.002361] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002406] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003933] cgroup: Disabling memory control group subsystem
[    0.004191] CPU: Testing write buffer coherency: ok
[    0.004690] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005957] Setting up static identity map for 0x200000 - 0x20003c
[    0.006177] rcu: Hierarchical SRCU implementation.
[    0.007145] smp: Bringing up secondary CPUs ...
[    0.008400] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.009797] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 438 lines cut ...]
[  551.927504] cec-vc4: message 10 timed out
[  554.087557] cec-vc4: message 10 timed out
[... cut ...]
Is this possibly related to #4320?

Now, I rebooted with 
```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-kms-v3d,cma-256
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
pi3-disable-bt
```
... this now leads to the unfortunate surprise of me not being able to connect via ssh (which has always worked reliably). Ok, time to power up the TV then.

I'll report later/tomorrow.

PeterPablo, 2021-07-29:
I seemingly was not patient enough for the reboot to happen. Possibly it took longer as a process hung (related to the new errors logged in dmesg?!). In any case, this new `dmesg` output still contains two types of errors
```
[   27.656483] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   38.235881] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[   48.475690] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[   58.715855] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[   68.956037] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:70:plane-3] flip_done timed out
```
whereas the `drm_kms_helper` ones repeat with a 10s interval.  The hifiberry errors do not occur further down.

For now I return to 
```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
pi3-disable-bt
```

In case me testing some other settings related to `kms`, I'm happy to do so. Kindly instruct.

PeterPablo, 2021-07-30:
I grabbed output of `raspinfo`: http://paste.debian.net/1206035/
Two things I'd like to note: 
* After all, this RPi4 has 1GB of RAM it seems
* My /boot partition is (legacy) small, stemming from my installation being migrated over the years.

popcornmix, 2021-07-30:
Can you replace `dtoverlay=vc4-kms-v3d,cma-256` with:
`dtoverlay=vc4-fkms-v3d,cma-256`
and ensure gpu_mem line is removed.

PeterPablo, 2021-07-30:
Sure thing. I had tried that line yesterday, without `,cma-256`. Doing as requested yields after reboot:
... a non reachable pi (ssh). I'll have to check why.

PeterPablo, 2021-07-30:
The stuck system did not show any output on my connected TV. A forced power-cycle brings the system up again without errors. A consequent reboot also went fine. Playing a video in kodi also is fine.
Please find below the raspinfo output for the reqested config.txt
```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-fkms-v3d,cma-256
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
pi3-disable-bt
```

http://paste.debian.net/1206037/

popcornmix, 2021-07-30:
> Sure thing. I had tried that line yesterday, without `,cma-256`. Doing as requested yields after reboot:
> ... a non reachable pi (ssh). I'll have to check why.

If the previous boot was using kms and had hit "flip_done timed out" then `sudo reboot` doesn't work (it may eventually, but I think shutting down has some very long timeout).

PeterPablo, 2021-07-30:
well, no. it actually was already back to fkms, though without the gpu_mem
line. I had waited around ten minutes before power cycling.

so now I'm running fkms with the option specified and no gpu_mem line
present.

In case there is merit in further analyzing the kms failure we can leave
this ticket open, otherwise I can close it.
My system is back to error free operation.

popcornmix, 2021-07-30:
Could you create a new issue referencing the kms "flip_done timed out" issue?
It may be something others hit and it will be easier to find a dedicated issue compared to a buried comment in an issue with an unrelated title.
I think we can close this one as you've found the right solution.
We're considering whether we can do something to improve behaviour when large gpu and cma regions are active.

PeterPablo, 2021-07-30:
Thank you again. As suggested, I created a new issue #4493 for the other observed issue.

closed 2021-07-30 by PeterPablo