raspberrypi/linux #4490: shdci prepare_data and swiotlb buffer is full crash system during dd operation on external usb ssd storage

reyman, 2021-07-30:
**Describe the bug**

I have usb-storage & xhci_hcd & swiotlb buffer problem filling systemd journal with SSD on USB, causing crash of my system during any DD operations on SSD disk 

This is my first bug report on kernel, also crossposted here (https://www.reddit.com/r/NixOS/comments/omctp3/rpi4_nixos_native_2105_and_ssd_by_usb3/), here on nixos github (https://github.com/NixOS/nixpkgs/issues/130499), and finally on the kernel usb mailling liste. They said it's related to **shdci** code.

I boot from a SD with nixos on RPI4B 8gb (firmware full upgraded to latest), and try to configure/format/erase an external SSD (CT1000MX500SSD1)

Because my startech adapter (USB312SAT3CB 3.1 firmware upgraded) not working with UAS i use some kernel quirks to boot and go back on usb-storage for the SSD. The linked explanation for common RPI4/UAS/Sata adapters problems is here : https://jamesachambers.com/new-raspberry-pi-4-bootloader-usb-network-boot-guide/ 

> "usb-storage.quirks=174c:55aa:u"
> "usbcore.quirks=174c:55aa:u"

During any dd=/ operation on SSD on USB 3 port, my logs finish full of errors of type :

```
juil. 29 14:38:52 Nostromo-sd kernel: xhci_hcd 0000:01:00.0: swiotlb buffer is full (sz: 512 bytes), total 32768 (slots), used 32768 (slots)
```

and 

```
juil. 29 09:25:07 Nostromo-sd kernel: ------------[ cut here ]------------
juil. 29 09:25:07 Nostromo-sd kernel: WARNING: CPU: 1 PID: 180 at drivers/mmc/host/sdhci.c:1142 sdhci_prepare_data.isra.0+0x298/0x700
juil. 29 09:25:07 Nostromo-sd kernel: Modules linked in: dm_crypt cbc aes_neon_bs aes_neon_blk encrypted_keys trusted asn1_encoder 8021q garp mrp>
juil. 29 09:25:07 Nostromo-sd kernel: CPU: 1 PID: 180 Comm: kworker/1:1H Tainted: G        WC        5.13.4 #1-NixOS
juil. 29 09:25:07 Nostromo-sd kernel: Hardware name: Raspberry Pi 4 Model B (DT)
juil. 29 09:25:07 Nostromo-sd kernel: Workqueue: kblockd blk_mq_run_work_fn
juil. 29 09:25:07 Nostromo-sd kernel: pstate: a00000c5 (NzCv daIF -PAN -UAO -TCO BTYPE=--)
juil. 29 09:25:07 Nostromo-sd kernel: pc : sdhci_prepare_data.isra.0+0x298/0x700
juil. 29 09:25:07 Nostromo-sd kernel: lr : sdhci_prepare_data.isra.0+0x1b0/0x700
juil. 29 09:25:07 Nostromo-sd kernel: sp : ffff8000130b38b0
juil. 29 09:25:07 Nostromo-sd kernel: x29: ffff8000130b38b0 x28: ffff8000130b3cd8 x27: ffff00010306d148
juil. 29 09:25:07 Nostromo-sd kernel: x26: 0000000000000000 x25: 0000000000000000 x24: 000000000000000b
juil. 29 09:25:07 Nostromo-sd kernel: x23: ffff00010306d2d0 x22: ffff0001031c1d00 x21: ffff0001031c1d00
juil. 29 09:25:07 Nostromo-sd kernel: x20: ffff00010306d340 x19: ffff000101e79580 x18: 0000000000000000
juil. 29 09:25:07 Nostromo-sd kernel: x17: ffff0001ff015380 x16: 0000000000000000 x15: 0000000000016563
juil. 29 09:25:07 Nostromo-sd kernel: x14: 0000000000008000 x13: 0000000000000000 x12: 0000000000000002
juil. 29 09:25:07 Nostromo-sd kernel: x11: 0000000000000000 x10: ffff0001f62b9024 x9 : ffff800010149620
juil. 29 09:25:07 Nostromo-sd kernel: x8 : 00000000000000c0 x7 : ffff0001031c1d00 x6 : ffff800012b2e000
juil. 29 09:25:07 Nostromo-sd kernel: x5 : ffff0001f62b9024 x4 : 0000000000000003 x3 : 0000000000000000
juil. 29 09:25:07 Nostromo-sd kernel: x2 : cee7e42d2b457d00 x1 : 0000000000000000 x0 : 00000000ffffffe4
juil. 29 09:25:07 Nostromo-sd kernel: Call trace:
juil. 29 09:25:07 Nostromo-sd kernel:  sdhci_prepare_data.isra.0+0x298/0x700
juil. 29 09:25:07 Nostromo-sd kernel:  sdhci_send_command+0x444/0x878
juil. 29 09:25:07 Nostromo-sd kernel:  sdhci_send_command_retry+0x50/0x190
juil. 29 09:25:07 Nostromo-sd kernel:  sdhci_request+0xa0/0x128
juil. 29 09:25:07 Nostromo-sd kernel:  __mmc_start_request+0x84/0x1a8
juil. 29 09:25:07 Nostromo-sd kernel:  mmc_start_request+0x9c/0xc8
juil. 29 09:25:07 Nostromo-sd kernel:  mmc_blk_mq_issue_rq+0x330/0x988
juil. 29 09:25:07 Nostromo-sd kernel:  mmc_mq_queue_rq+0x14c/0x358
juil. 29 09:25:07 Nostromo-sd kernel:  blk_mq_dispatch_rq_list+0x124/0x8b8
juil. 29 09:25:07 Nostromo-sd kernel:  __blk_mq_do_dispatch_sched+0x230/0x2b0
juil. 29 09:25:07 Nostromo-sd kernel:  __blk_mq_sched_dispatch_requests+0x1b0/0x1f8
juil. 29 09:25:07 Nostromo-sd kernel:  blk_mq_sched_dispatch_requests+0x40/0x80
juil. 29 09:25:07 Nostromo-sd kernel:  __blk_mq_run_hw_queue+0x58/0x98
juil. 29 09:25:07 Nostromo-sd kernel:  blk_mq_run_work_fn+0x28/0x38
juil. 29 09:25:07 Nostromo-sd kernel:  process_one_work+0x1d0/0x478
juil. 29 09:25:07 Nostromo-sd kernel:  worker_thread+0x148/0x4e8
juil. 29 09:25:07 Nostromo-sd kernel:  kthread+0x128/0x138
juil. 29 09:25:07 Nostromo-sd kernel:  ret_from_fork+0x10/0x30
juil. 29 09:25:07 Nostromo-sd kernel: ---[ end trace 430fa301f491b192 ]---
```

**To reproduce**

- Branch any SSD to USB3
- Run long dd command on it  

**Expected behaviour**
[... 847 lines cut ...]
	Flags: bus master, fast devsel, latency 0, IRQ 63
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff [size=4K]
	Memory behind bridge: f8000000-f80fffff [size=1M]
	Prefetchable memory behind bridge: [disabled]
	Capabilities: [48] Power Management version 3
	Capabilities: [ac] Express Root Port (Slot-), MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240] L1 PM Substates
	Kernel driver in use: pcieport

01:00.0 USB controller: VIA Technologies, Inc. VL805/806 xHCI USB 3.0 Controller (rev 01) (prog-if 30 [XHCI])
	Subsystem: VIA Technologies, Inc. VL805/806 xHCI USB 3.0 Controller
	Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@1,0/usb@1,0
	Flags: bus master, fast devsel, latency 0, IRQ 64
	Memory at 600000000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: [80] Power Management version 3
	Capabilities: [90] MSI: Enable+ Count=1/4 Maskable- 64bit+
	Capabilities: [c4] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci

```

pelwell, 2021-07-30:
@mgrouch This one.

pelwell, 2021-07-30:
What's in your config.txt? I'm particularly interested in `gpu_mem` settings and if one of the `vc4_kms_v3d` or `vc4_fkms_v3d` overlays is being used.
1. If there's a `gpu_mem=...`, comment it out (`# gpu_mem=...`).
2. If you are using one of the vc4 overlays, try appending a CMA setting of 256MB, i.e. `dtoverlay=vc4_kms_v3d,cma-256` or `dtoverlay=vc4_fkms_v3d,cma-256`.

pelwell, 2021-07-30:
The contents of `/boot/cmdline.txt` might also be relevant.

mgrouch, 2021-07-30:
My boot configs

[cmdline.txt](https://github.com/raspberrypi/linux/files/6908705/cmdline.txt)
[config.txt](https://github.com/raspberrypi/linux/files/6908706/config.txt)

popcornmix, 2021-07-30:
Remove this:
`gpu_mem=384`
On 5.10 kernel there's very little reason for gpu_mem to be higher than default (and it will reduce usable memory).

mgrouch, 2021-07-30:
adding cma-256 fixed it

dtoverlay=vc4_fkms_v3d,cma-256

reyman, 2021-07-30:
My config.txt on nixos (mounted using `mount /dev/mmcblk1p1 /mnt`)
 
```
nostromo@Nostromo-sd:/]$ cat /mnt/config.txt 
[pi3]
kernel=u-boot-rpi3.bin

[pi4]
kernel=u-boot-rpi4.bin
enable_gic=1
armstub=armstub8-gic.bin

# Otherwise the resolution will be weird in most cases, compared to
# what the pi3 firmware does by default.
disable_overscan=1

[all]
# Boot in 64-bit mode.
arm_64bit=1

# U-Boot needs this to work, regardless of whether UART is actually used or not.
# Look in arch/arm/mach-bcm283x/Kconfig in the U-Boot tree to see if this is still
# a requirement in the future.
enable_uart=1

# Prevent the firmware from smashing the framebuffer setup done by the mainline kernel
# when attempting to show low-voltage or overtemperature warnings.
avoid_warnings=1

```

For cmdline, i suppose the equivalent on nixos is located into ` /nix/store/qhciy3ckbydm4bhx0jlg4yra33mmpp0p-nixos-system-Nostromo-sd-21.11pre304429.4a440fd9223/kernel-params` : 

`usb-storage.quirks=174c:55aa:u usbcore.quirks=174c:55aa:u console=ttySO,115200n8 console=tty1MA0n115200n8 console=tty0 loglevel=4
`

mgrouch, 2021-08-02:
Do I now always need to add 

dtoverlay=cma-256

just to make it boot?

Thanks

mgrouch, 2021-08-02:
Does arm64 bit version need the same change?

Thanks

popcornmix, 2021-08-02:
> Do I now always need to add
> 
> dtoverlay=cma-256

You should remove the deprecated gpu_mem settings. It's very likely not required and is just wasting memory.