raspberrypi/linux #4493: flip_done timed out on 5.10.52-v7l+ #1440 with vc4-kms-v3d

PeterPablo, 2021-07-30:
Applying `dtoverlay=vc4-kms-v3d` on an up-to-date Raspberry Pi OS and system yields `flip_done` error messages every 10s. Note that kodi is started on boot.
```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.52-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1440 SMP Tue Jul 27 09:55:21 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000003b3fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 46080 pages, LIFO batch:15
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240960
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@50,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:00:97:FD vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=66606d14-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000000ac00000-0x000000000ec00000] (64MB)
[    0.000000] Memory: 351020K/970752K available (10240K kernel code, 1364K rwdata, 3188K rodata, 2048K init, 891K bss, 95444K reserved, 524288K cma-reserved, 184320K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34303 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=0
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000097] bcm2835: system timer (irq = 26)
[    0.000760] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000782] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000805] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000823] Switching to timer-based delay loop, resolution 18ns
[    0.001087] Console: colour dummy device 80x30
[    0.001848] printk: console [tty1] enabled
[    0.001916] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001973] pid_max: default: 32768 minimum: 301
[    0.002152] LSM: Security Framework initializing
[    0.002361] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002406] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003933] cgroup: Disabling memory control group subsystem
[    0.004191] CPU: Testing write buffer coherency: ok
[    0.004690] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005957] Setting up static identity map for 0x200000 - 0x20003c
[    0.006177] rcu: Hierarchical SRCU implementation.
[    0.007145] smp: Bringing up secondary CPUs ...
[    0.008400] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.009797] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 439 lines cut ...]
[  554.087557] cec-vc4: message 10 timed out
[  554.727652] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  556.247575] cec-vc4: message 10 timed out

```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-kms-v3d,cma-256
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
pi3-disable-bt
```

Once I return to `fkms` the system and reboot `dmesg` stays clean. Ref. `raspinfo`: 
http://paste.debian.net/1206037/

Is this possibly related to #4320?

Best regards
  Peter

pelwell, 2021-07-30:
```
Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
```

It appears that, for some reason, the `cma-256` parameter didn't work as expected.

[ And `pi3-disable-bt` should just be `disable-bt` these days ]

popcornmix, 2021-07-30:
If you stop kodi starting from boot do you still see the flip_done message?
Do you see it when manually starting kodi later?
Does it occur as soon as kodi starts, or when video playback starts?
What hdmi resolution / refresh rate is being used?

PeterPablo, 2021-07-31:
Phil, I might have mixed up my test results from #4489 and thus redid the test:
```
pi@horst:~ $ cat /boot/config.txt | grep -v "^#" | grep -v "^[[:space:]]*$"
dtoverlay=hifiberry-dacplus
dtoverlay=vc4-kms-v3d,cma-256
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=31
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
dtparam=audio=off
[pi4]
disable-bt
```
http://paste.debian.net/1206148/ (this is with kodi running)

```
pi@horst:~ $ sudo systemctl disable kodi
Removed /etc/systemd/system/multi-user.target.wants/kodi.service.
pi@horst:~ $ sudo reboot
```
After the reboot without kodi startup, indeed the error messages of the HifiBerry DAC and flip_done do not show up:
http://paste.debian.net/1206149/

Once I manually start kodi, the errors show up in `dmesg` :
```
[   15.424412] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.424811] Bluetooth: HCI UART protocol Broadcom registered
[   15.632989] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.632998] Bluetooth: BNEP filters: protocol multicast
[   15.633012] Bluetooth: BNEP socket layer initialized
[   97.082040] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   97.099126] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   97.100461] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   97.101679] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   97.102138] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   97.102567] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[  107.355941] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  117.596090] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  127.836389] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[  138.086592] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:70:plane-3] flip_done timed out
```

# Details about kodi
It is installed from apt.
```
pi@horst:~ $ dpkg -s kodi
Package: kodi
Status: install ok installed
Priority: optional
Section: video
Installed-Size: 38801
Maintainer: <>
Architecture: all
Source: kodi
Version: 2:18.7-1~buster
```
The pipplware repository is the source (I think):
`http://pipplware.pplware.pt/pipplware/dists/buster/main/binary `

My kodi installation does bring up a bigpicture screensaver after some timeout. I would need to check the exact addon name, in case this is relevant to you. Kindly instruct what further information you need.
Best regards
  Peter

popcornmix, 2021-07-31:
I'd like to know the resolution kodi is switching to. If hdmi output stops immediately you may not be able to see that on display.
Do you see kodi's gui briefly before the 'flip_done' error?

Can you paste output of:
```
sudo cat /sys/kernel/debug/dri/0/state
```
(if that does exist try `/sys/kernel/debug/dri/1/state` )

PeterPablo, 2021-07-31:
On the not rebooted system, below you find the desired dri state:
```
pi@horst:~ $ sudo cat /sys/kernel/debug/dri/0/state | pastebinit 
/usr/bin/pastebinit:42: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5
  release = platform.linux_distribution()[0].lower()
/usr/bin/pastebinit:413: DeprecationWarning: pasteURLopener style of invoking requests is deprecated. Use newer urlopen functions/methods
  url_opener = pasteURLopener()
http://paste.debian.net/1206150/
```

I now did a reboot while having the TV switched on (a ~8 year old Grundig model). It happily boots to console, as kodi service is still disabled. Once starting kodi from console (via ssh), the TV immediately goes first black and one second later to blue, indicating "no signal".

Now I for testing purposes commented the `hdmi_group` and `hdmi_mode` setting and tried these more default settings. As reported in the other issue #4489 , the `sudo reboot` is not honored and I have to power-cycle.
The behaviour is unchanged (console displayed on TV fine, starting kodi vis ssh is leading to immediate signal loss).

pelwell, 2021-07-31:
Thanks for the updated log - that does show the correct CMA size.

popcornmix, 2021-07-31:
The dri status looks okay. It thinks it is driving
```mode: "1920x1080": 50 148500 1920 2448 2492 2640 1080 1084 1089 1125 0x68 0x5```
with only the console (fbcon) on display (so kodi hasn't started). 
1920x1080 RG16
Can you post your edid? (e.g. `base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid`) and kodi.log.

popcornmix, 2021-07-31:
Also did this setup work on a previous kernel? Do you know what the last working kernel was?
I assume switching to the fkms driver does work okay?

PeterPablo, 2021-07-31:
According to apt, this was the conducted upgrade of firmware/kernel:
```
Preparing to unpack .../0-libraspberrypi-doc_1%3a1.20210727-1_armhf.deb ...
Unpacking libraspberrypi-doc (1:1.20210727-1) over (1:1.20210527-1) ...
Preparing to unpack .../1-libraspberrypi-dev_1%3a1.20210727-1_armhf.deb ...
Unpacking libraspberrypi-dev (1:1.20210727-1) over (1:1.20210527-1) ...
Preparing to unpack .../2-raspberrypi-kernel_1%3a1.20210727-1_armhf.deb ...
...
```
Certainly, the kernel deployed by the 1.20210527 package could be looked up - to me it is not obvious where.

Before reporting the issue I was operating on
```
dtoverlay=vc4-fkms-v3d
gpu_mem=320
```
without noticing issues.

Now on 5.10.52-v7l+, I gave `vc4-kms-v3d` a go, with the above described issues. I am not ware of having tried kms in the past so can not reference a kernel, where this has worded fine.

# fkms information
Beware that the following output is from me now running again on `dtoverlay=vc4-fkms-v3d,cma-256`: 
card1 was not available, but card0 is
```
pi@horst:~ $ base64 /sys/devices/platform/gpu/drm/card0/card0-HDMI-A-1/edid 
AP///////wAeVUhEAQAAACUWAQOAoFp4KuxroFZHnCUQSEonzwABAQEBgUCBgIGPiwABAalAAjqA
0HI4LUAQLEWAoFoAAAAeAjqAGHE4LUBYLEUAoFoAAAAeAAAA/QAxTA9RCAAKICAgICAgAAAA/ABH
UlVORElHIFdVWEdBAYYCAyJyTxQFEwQSAxECFgcVBgGfkCYVB1AJBwdmAwwAEACAGyFQoFEAHjDQ
iGUEoFoAAAAcDh8AgFEAHjDAgEcEoFoAAAAcoCiAgEFgKTDAeCQIoFoAAAAcjy940FEaJ0DokAQI
oJAAAAAcmimg0FGEIjDomJYEoJAAAAAcAAAAdw==
```

kodi.log (until startup finished):
```
2021-07-31 14:58:27.333 T:2973142112  NOTICE: -----------------------------------------------------------------------
2021-07-31 14:58:27.334 T:2973142112  NOTICE: Starting Kodi (18.7 Git:20200528-2c5354e70a). Platform: Linux ARM 32-bit
2021-07-31 14:58:27.334 T:2973142112  NOTICE: Using Release Kodi x32 build
2021-07-31 14:58:27.334 T:2973142112  NOTICE: Kodi compiled 2020-06-12 by GCC 8.3.0 for Linux ARM 32-bit version 4.18.20 (266772)
2021-07-31 14:58:27.334 T:2973142112  NOTICE: Running on Raspbian GNU/Linux 10 (buster), kernel: Linux ARM 32-bit version 5.10.52-v7l+
2021-07-31 14:58:27.344 T:2973142112  NOTICE: FFmpeg version/source: 4.0.4-Kodi
2021-07-31 14:58:27.344 T:2973142112  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 4 cores available
2021-07-31 14:58:27.344 T:2973142112  NOTICE: ARM Features: Neon enabled
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://xbmcbinaddons/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi/addons
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://masterprofile/ is mapped to: /home/kodi/.kodi/userdata
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://envhome/ is mapped to: /home/kodi
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://home/ is mapped to: /home/kodi/.kodi
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://temp/ is mapped to: /home/kodi/.kodi/temp
2021-07-31 14:58:27.344 T:2973142112  NOTICE: special://logpath/ is mapped to: /home/kodi/.kodi/temp
2021-07-31 14:58:27.344 T:2973142112  NOTICE: The executable running is: /usr/lib/arm-linux-gnueabihf/kodi/kodi.bin_v8
2021-07-31 14:58:27.344 T:2973142112  NOTICE: Local hostname: horst
2021-07-31 14:58:27.344 T:2973142112  NOTICE: Log File is located: /home/kodi/.kodi/temp/kodi.log
2021-07-31 14:58:27.344 T:2973142112  NOTICE: -----------------------------------------------------------------------
2021-07-31 14:58:27.353 T:2973142112    INFO: loading settings
2021-07-31 14:58:27.358 T:2973142112  NOTICE: special://profile/ is mapped to: special://masterprofile/
2021-07-31 14:58:27.473 T:2973142112   DEBUG: CSkinSettings: no <skinsettings> tag found
2021-07-31 14:58:27.475 T:2973142112  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
2021-07-31 14:58:27.475 T:2973142112  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <?xml version="1.0" encoding="UTF-8" ?>
                                            <advancedsettings>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <video>
                                                <busydialogdelayms>750</busydialogdelayms>
                                              </video>
                                              <samba>
                                                <clienttimeout>30</clienttimeout>
                                              </samba>
                                              <cache>
[... 334 lines cut ...]
                                                result = func(*args)
                                              File "/usr/lib/python2.7/urllib2.py", line 1241, in https_open
                                                context=self._context)
                                              File "/usr/lib/python2.7/urllib2.py", line 1198, in do_open
                                                raise URLError(err)
                                            URLError: <urlopen error [Errno -3] Temporary failure in name resolution>
                                            -->End of Python script error report<--
[... cut ...]
iGUEoFoAAAAcDh8AgFEAHjDAgEcEoFoAAAAcoCiAgEFgKTDAeCQIoFoAAAAcjy940FEaJ0DokAQI
oJAAAAAcmimg0FGEIjDomJYEoJAAAAAcAAAAdw==
```
This is identical to the fkms edid (expected, but I guess you never know.

`kodi.log`
Is this variant of kodi.log of interest? If I find some more time I'll return to the problematic configuration and obtain it. I'd like to not do it right now, as I need to again do the power cycling. I had looked at it the other day, but it is overwritten now. It was not looking helpful to me. It errored out, but how I don't recall for certain.

popcornmix, 2021-08-01:
kodi.log with kms and debug logging enabled would be useful (enable debug logging, restart kodi then grab kodi.log when you've got flip_done error).

PeterPablo, 2021-08-08:
Finally I got the chance to do the test with kms and debug logging activated in kodi. Honestly, I'm not sure if the test went fine. I can confirm that once I manually initiated the kodi start, the display lost signal, but the kodi.log does not seem to contain enlightening output:
```
pi@horst:~ $ cat /home/kodi/.kodi/temp/kodi.log
2021-08-08 09:46:46.398 T:2972658784  NOTICE: -----------------------------------------------------------------------
2021-08-08 09:46:46.398 T:2972658784  NOTICE: Starting Kodi (18.7 Git:20200528-2c5354e70a). Platform: Linux ARM 32-bit
2021-08-08 09:46:46.398 T:2972658784  NOTICE: Using Release Kodi x32 build
2021-08-08 09:46:46.398 T:2972658784  NOTICE: Kodi compiled 2020-06-12 by GCC 8.3.0 for Linux ARM 32-bit version 4.18.20 (266772)
2021-08-08 09:46:46.398 T:2972658784  NOTICE: Running on Raspbian GNU/Linux 10 (buster), kernel: Linux ARM 32-bit version 5.10.52-v7l+
2021-08-08 09:46:46.407 T:2972658784  NOTICE: FFmpeg version/source: 4.0.4-Kodi
2021-08-08 09:46:46.407 T:2972658784  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 4 cores available
2021-08-08 09:46:46.407 T:2972658784  NOTICE: ARM Features: Neon enabled
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://xbmcbinaddons/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi/addons
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://masterprofile/ is mapped to: /home/kodi/.kodi/userdata
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://envhome/ is mapped to: /home/kodi
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://home/ is mapped to: /home/kodi/.kodi
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://temp/ is mapped to: /home/kodi/.kodi/temp
2021-08-08 09:46:46.407 T:2972658784  NOTICE: special://logpath/ is mapped to: /home/kodi/.kodi/temp
2021-08-08 09:46:46.407 T:2972658784  NOTICE: The executable running is: /usr/lib/arm-linux-gnueabihf/kodi/kodi.bin_v8
2021-08-08 09:46:46.407 T:2972658784  NOTICE: Local hostname: horst
2021-08-08 09:46:46.407 T:2972658784  NOTICE: Log File is located: /home/kodi/.kodi/temp/kodi.log
2021-08-08 09:46:46.407 T:2972658784  NOTICE: -----------------------------------------------------------------------
2021-08-08 09:46:46.414 T:2972658784    INFO: loading settings
2021-08-08 09:46:46.419 T:2972658784  NOTICE: special://profile/ is mapped to: special://masterprofile/
2021-08-08 09:46:46.439 T:2972658784   DEBUG: CSkinSettings: no <skinsettings> tag found
2021-08-08 09:46:46.442 T:2972658784  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
2021-08-08 09:46:46.442 T:2972658784  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <?xml version="1.0" encoding="UTF-8" ?>
                                            <advancedsettings>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <video>
                                                <busydialogdelayms>750</busydialogdelayms>
                                              </video>
                                              <samba>
                                                <clienttimeout>30</clienttimeout>
                                              </samba>
                                              <cache>
                                                <memorysize>31457280</memorysize>
                                              </cache>
                                            </advancedsettings>
2021-08-08 09:46:46.442 T:2972658784 WARNING: CSettingsManager: missing version attribute
2021-08-08 09:46:46.443 T:2972658784  NOTICE: No settings file to load (special://masterprofile/advancedsettings.xml)
2021-08-08 09:46:46.443 T:2972658784  NOTICE: Default Video Player: VideoPlayer
2021-08-08 09:46:46.443 T:2972658784  NOTICE: Default Audio Player: paplayer
2021-08-08 09:46:46.443 T:2972658784  NOTICE: Disabled debug logging due to GUI setting. Level 0.
2021-08-08 09:46:46.443 T:2972658784  NOTICE: Log level changed to "LOG_LEVEL_NORMAL"
2021-08-08 09:46:46.584 T:2972658784  NOTICE: PulseAudio: Server not running
2021-08-08 09:46:46.641 T:2972658784  NOTICE: Running database version Addons27
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: audioencoder.kodi.builtin.aac v1.0.0 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: audioencoder.kodi.builtin.wma v1.0.0 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: game.controller.default v1.0.8 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: game.controller.snes v1.0.8 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: inputstream.adaptive v2.4.4 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: inputstream.rtmp v2.0.6 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: kodi.binary.global.audioengine v1.0.1 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: kodi.binary.global.filesystem v1.0.2 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: kodi.binary.global.general v1.0.3 installed
2021-08-08 09:46:46.931 T:2972658784  NOTICE: ADDON: kodi.binary.global.gui v5.12.0 installed
[... cut ...]
                                            hdmi_edid_filename:1=
                                            hdmi_timings:0=
                                            hdmi_timings:1=

```
It seems that me enabling debug logging did not stick (`2021-08-08 09:46:46.443 T:2972658784  NOTICE: Disabled debug logging due to GUI setting. Level 0.`). I will try to redo the test and report back. For now sticking to fkms works around the issue nicely. As I understand, kms is the intended path forward, so I'll do my best to resolve this potential issue.

Axionkt, 2021-08-20:
I'm having the same issue on 2 Pi4's. Here is the rest of kodi.log after waiting a few minutes when the display output turns off kodi loads for a minutes then it's back to "no signal" but the kodi process is still running:
I removed a good part of the log as I seem to have some an unclean database throwing a lot of warnings.
Unfortunately I did not test the kms driver before now so I can't say when the issue appeared.

I think you can safely ignore the 4096x2160 resolution as I'm normally using a 1080p TV and the problem is the same.

```
                                             hdmi_timings:1=
2021-08-20 17:02:57.015 T:2972707936    INFO: RetroPlayer[PROCESS]: Registering process control for GBM
2021-08-20 17:02:57.015 T:2972707936    INFO: RetroPlayer[RENDER]: Registering renderer factory for OpenGLES
2021-08-20 17:03:57.063 T:2930766032   DEBUG: CAESinkPi:Drain delay:99ms now:0ms
2021-08-20 17:04:07.063 T:2930766032   DEBUG: CAESinkPi:Deinitialize
2021-08-20 17:04:07.063 T:2930766032   DEBUG: CAESinkPi:SetAudioProps hdmi_stream_channels 0 hdmi_channel_map 00000000
2021-08-20 17:04:07.077 T:2930766032   DEBUG: COMXCoreComponent::Deinitialize : OMX.broadcom.audio_render handle 0xae1030e0
2021-08-20 17:07:54.426 T:2972707936   DEBUG: CDRMUtils::FindConnector - found connector: 32
2021-08-20 17:07:54.427 T:2972707936   DEBUG: CDRMUtils::OpenDrm - opened device: /dev/dri/card1 using module: vc4
2021-08-20 17:07:54.531 T:2972707936   DEBUG: CDRMUtils::FindConnector - found connector: 32
2021-08-20 17:07:54.531 T:2972707936   DEBUG: CDRMUtils::FindEncoder - found encoder: 31
2021-08-20 17:07:54.531 T:2972707936   DEBUG: CDRMUtils::FindCrtc - found crtc: 76
2021-08-20 17:07:54.532 T:2972707936   DEBUG: CDRMUtils::FindPlane - found video plane 70
2021-08-20 17:07:54.535 T:2972707936 WARNING: CDRMUtils::FindPlane - could not find plane
2021-08-20 17:07:54.535 T:2972707936   DEBUG: CDRMUtils::FindPlane - found gui plane 91
2021-08-20 17:07:54.540 T:2972707936   DEBUG: CDRMAtomic::InitDrm - initialized atomic DRM
2021-08-20 17:07:56.136 T:2972707936   DEBUG: CWinSystemGbm::InitWindowSystem - initialized DRM
2021-08-20 17:07:56.136 T:2972707936  NOTICE: Found resolution 1920x1080 with 4096x2160 @ 24.000000 Hz
2021-08-20 17:07:56.136 T:2972707936  NOTICE: Found resolution 1920x1080 with 4096x2160 @ 23.976025 Hz
2021-08-20 17:07:56.136 T:2972707936  NOTICE: Found resolution 1920x1080 with 3840x2160 @ 30.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 3840x2160 @ 29.970032 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 3840x2160 @ 25.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 3840x2160 @ 24.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 3840x2160 @ 23.976025 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 60.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 59.940063 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080i @ 60.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080i @ 59.940063 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 50.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080i @ 50.000000 Hz
2021-08-20 17:07:56.137 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 30.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 29.970032 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 24.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1920x1080 with 1920x1080 @ 23.976025 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1680x1050 with 1680x1050 @ 60.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1600x900 with 1600x900 @ 60.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1280x1024 with 1280x1024 @ 60.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1152x864 with 1152x864 @ 75.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 60.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 59.940063 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 50.000000 Hz
2021-08-20 17:07:56.138 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 30.000000 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 29.970032 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 24.000000 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 1280x720 with 1280x720 @ 23.976025 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 1024x768 with 1024x768 @ 60.000000 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 800x600 with 800x600 @ 60.000000 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 720x576 with 720x576 @ 50.000000 Hz
2021-08-20 17:07:56.139 T:2972707936  NOTICE: Found resolution 720x576 with 720x576i @ 50.000000 Hz
[... cut ...]
2021-08-20 17:55:13.123 T:803     DEBUG <general>: CMultiPathDirectory::MergeItems, items = 0
2021-08-20 17:55:13.137 T:813     DEBUG <general>: -->Python Interpreter Initialized<--
2021-08-20 17:55:13.137 T:813     DEBUG <general>:
2021-08-20 17:55:14.706 T:822     DEBUG <general>: OpenConnection - connection to the CEC adapter opened
2021-08-20 17:55:14.706 T:822     DEBUG <general>: SetConfigurationFromLibCEC - settings updated by libCEC
2021-08-20 17:55:14.707 T:829     DEBUG <general>: Thread CECAdapterUpdate start, auto delete: false
'''

popcornmix, 2021-08-20:
> This shows no apparent problem:

Can you clarify if you meant that there is no problem (i.e. display is working), or "no apparent problem" in log (but display is broken).

Axionkt, 2021-08-20:
Sorry if that wasn't clear.

**In the first log** from Raspberry Pi OS running Kodi I experience the same problem as PeterPablo.
I get [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
messages in dmesg
However after waiting a few minutes (around 2021-08-20 17:07:54.426 T:2972707936 in the log) the display starts working for around 1min (not sure exactly), as in I get the kodi interface and it seems to be scanning my library, but then stops working again and never comes back on.
The first log should show this full cycle.

**The second log** is just running the latest LibreELEC to compare which works fine with approximately the same config.txt
With this everything seems to be working, sound display, CEC etc.

popcornmix, 2021-08-20:
@Axionkt can you report `uname -a` and `vcgencmd versions` for RPiOS and LibreELEC.

Axionkt, 2021-08-20:
@popcornmix Sure. I'm running Kodi 18.7 on Raspberry Pi OS
I don't know if it makes a difference but on Raspberry Pi OS I tried 19.1 from upstream and also gbm_matrix from your repo with the same result except that I see the kodi splash screen before the display crashes.

### Raspbian GNU/Linux 10 (buster)
<details><summary>Raspberry Pi OS vcgencmd version</summary>
<p>

```Aug  3 2021 18:14:56
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)
```
</p>
</details>
<details><summary>Raspberry Pi OS vcgencmd bootloader_version</summary>
<p>

```Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1629367548
capabilities 0x0000001f
```
</p>
</details>
<details><summary>Raspberry Pi OS config.txt</summary>
<p>

```disable_overscan=1
dtparam=audio=on
[pi4]
max_framebuffers=2
gpu_mem=76
hdmi_ignore_cec_init=1
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
disable_fw_kms_setup=1
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
```
</p>
</details>
<details><summary>Raspberry Pi OS uname -a</summary>
<p>

```Linux RPi4-2 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
```
</p>
</details>
</details>
<details><summary>Raspberry Pi OS dmesg</summary>
<p>

```[Fri Aug 20 18:39:11 2021] Booting Linux on physical CPU 0x0
[Fri Aug 20 18:39:11 2021] Linux version 5.10.52-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1441 SMP Tue Aug 3 18:11:56 BST 2021
[Fri Aug 20 18:39:11 2021] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[Fri Aug 20 18:39:11 2021] CPU: div instructions available: patching division code
[Fri Aug 20 18:39:11 2021] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[Fri Aug 20 18:39:11 2021] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[Fri Aug 20 18:39:11 2021] random: fast init done
[Fri Aug 20 18:39:11 2021] Memory policy: Data cache writealloc
[Fri Aug 20 18:39:11 2021] Reserved memory: created CMA memory pool at 0x000000000e400000, size 512 MiB
[Fri Aug 20 18:39:11 2021] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[Fri Aug 20 18:39:11 2021] Zone ranges:
[Fri Aug 20 18:39:11 2021]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[Fri Aug 20 18:39:11 2021]   Normal   empty
[Fri Aug 20 18:39:11 2021]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[Fri Aug 20 18:39:11 2021] Movable zone start for each node
[Fri Aug 20 18:39:11 2021] Early memory node ranges
[Fri Aug 20 18:39:11 2021]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[Fri Aug 20 18:39:11 2021]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[... 1316 lines cut ...]
[   17.835765] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   17.863974] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[   18.261331] uart-pl011 fe201000.serial: no DMA platform data
[   18.375163] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   18.375690] bcmgenet fd580000.ethernet eth0: Link is Down
[   18.630679] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   22.425390] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   22.425442] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   22.488684] 8021q: 802.1Q VLAN Support v1.8
[   25.095148] Bluetooth: Core ver 2.22
[   25.095221] NET: Registered protocol family 31
[   25.095228] Bluetooth: HCI device and connection manager initialized
[   25.095246] Bluetooth: HCI socket layer initialized
[   25.095256] Bluetooth: L2CAP socket layer initialized
[   25.095272] Bluetooth: SCO socket layer initialized
[   25.097821] Bluetooth: HCI UART driver ver 2.3
[   25.097831] Bluetooth: HCI UART protocol H4 registered
[   25.097886] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   25.098047] Bluetooth: HCI UART protocol Broadcom registered
[   25.205069] cryptd: max_cpu_qlen set to 1000
[   33.309128] hdmi-audio-codec hdmi-audio-codec.1.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   33.309756] hdmi-audio-codec hdmi-audio-codec.1.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
```
</p>
</details>

Axionkt, 2021-08-24:
I tried to pinpoint when the issue appeared using rpi-update.

Kodi works fine with this version:
[V] Linux RPi4-2 5.10.46-v7l+ #1430 SMP Mon Jun 28 17:14:36 BST 2021 armv7l GNU/Linux [48aef33c6139b7a38013c741f18bf4a1851a1b6d](https://github.com/Hexxeh/rpi-firmware/commit/48aef33c6139b7a38013c741f18bf4a1851a1b6d)

<details><summary>dmesg</summary>
<p>

```[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.46-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1430 SMP Mon Jun 28 17:14:36 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 458752
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 457024
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:E4:A8:3F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=0907ec34-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000009a00000-0x000000000da00000] (64MB)
[    0.000000] Memory: 1207680K/1835008K available (10240K kernel code, 1364K rwdata, 3188K rodata, 2048K init, 891K bss, 103040K reserved, 524288K cma-reserved, 1048576K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34291 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000099] bcm2835: system timer (irq = 26)
[    0.000754] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000776] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000800] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000818] Switching to timer-based delay loop, resolution 18ns
[    0.001085] Console: colour dummy device 80x30
[    0.001851] printk: console [tty1] enabled
[    0.001919] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001973] pid_max: default: 32768 minimum: 301
[    0.002155] LSM: Security Framework initializing
[    0.002361] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002407] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003913] cgroup: Disabling memory control group subsystem
[    0.004168] CPU: Testing write buffer coherency: ok
[... 754 lines cut ...]
[  441.916616] cec-vc4: message 10 timed out
[  444.076629] cec-vc4: message 10 timed out
[  446.236670] cec-vc4: message 10 timed out
[  448.396671] cec-vc4: message 10 timed out
[  450.556654] cec-vc4: message 10 timed out
[  452.716702] cec-vc4: message 10 timed out
[  454.876735] cec-vc4: message 10 timed out
[... cut ...]
[  478.396796] cec-vc4: message 10 timed out
[  498.396991] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  509.277015] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  519.517081] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:91:plane-6] flip_done timed out
```
</p>
</details>

If it helps, as soon as I have enough time I'll try the modifications 1 by 1 between these 2 commits.

Axionkt, 2021-11-23:
So, I can't tell what exactly worked around the issue but compiling Kodi 19.3 from the popcornmix bullseye branch and starting it with the same parameters as before works.
With Kernel 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux.

mripard, 2021-11-23:
This has probably been fixed by the firmware update mentioned in #4540

PeterPablo, 2022-02-22:
In the meanwhile I'm in the process of migrating to a freshly setup bullseye Raspberry Pi OS image (32bit). Here, running under kms, the apt supplied kodi works flawlessly. Thus I'll close this issue.

closed 2022-02-22 by PeterPablo