raspberrypi/linux #4494: Reboots on high USB disk activity with July 22, 2021 kernel, not with July 22, 2021 5.10.52 kernel

satmandu, 2021-07-30:
This is on a heavily modified RaspiOS 64-bit system updated to Debian Bullseye, with xfs and zfs volumes mounted on USB drives, running docker on a 8gb RPI4B system.

**Describe the bug**
Build a multi-GB docker image, and the machine locks up and reboots when writing the image to disk.

This only happens with the newer kernel. The older kernel was rock solid. Reverting to it resolves the issue.

**System**
Raspinfo:
https://paste.debian.net/1206106/
* Which model of Raspberry Pi? RPI4B

**Logs**
Nothing shows up before reboot.

pelwell, 2021-07-31:
Can you clarify the good and bad kernel versions? One of the two appears to be build 1439, but what about the other?

Are there any clues in /var/log/syslog from before the crash?

satmandu, 2021-08-01:
Good kernel:. https://github.com/Hexxeh/rpi-firmware/commit/1fad1d2beceaaf96e84e3f5caa79f52b87332e89

Bad kernel: 
https://github.com/Hexxeh/rpi-firmware/commit/b986fda889737f6ce0a4acda89d447f78d0a1a73

I will report back if I see anything in the logs. dmesg was showing nothing useful before the crash. I was connected via Ethernet and the device just stopped responding before the crash.

popcornmix, 2021-08-01:
Can you try the latest kernel with (in config.txt):
```
dtoverlay=vc4-fkms-v3d
```
changed to 
```
dtoverlay=vc4-fkms-v3d,cma-256
```

satmandu, 2021-08-01:
> Can you try the latest kernel with (in config.txt):
> ```
> dtoverlay=vc4-fkms-v3d
> ```
> changed to 
> ```
> dtoverlay=vc4-fkms-v3d,cma-256
> ```

Trying this now.

satmandu, 2021-08-01:
No instability 5 hours in!

Connecting via ssh did feel sluggish with this kernel earlier, but if feels fine now.

I'll close this and reopen if there are any further issues.

Thanks!

closed 2021-08-01 by satmandu